openscad/openscad #1849: Create portable image for Linux

t-paul, 2016-11-01:
The generic Linux binary on the download page is quite outdated now, mostly because it's very hard to build generic / portable binaries that work on different systems / distributions.

The current workaround is to build on [OBS](https://build.opensuse.org/package/show/home:t-paul/OpenSCAD) for a number of distributions that are available there.

Lately there's a number of contenders trying to solve the issue of getting desktop application faster to the end user.

* [Flat-pack](http://flatpak.org/) (previously xdg-app) - based on shared runtimes like Gnome or KDE
* [Snap](http://snapcraft.io/) - self contained apps, but needs a system service? driven by Ubuntu
* [AppImage](http://appimage.org/) - self contained apps, similar to dmg on macOS?
* others?

Todo:
- [ ] Decide if we want to support one (or more?) of the available options

donbright, 2016-11-01:
That OBS setup is extremely impressive. Can it be integrated better into the OpenSCAD download page?  

Regarding portable linux binary packagers, i am extremely skeptical because of history. OpenSCAD's own chrpath based linux portable binary build, would crash a lot on different distros and kernels.  These are nasty crashes nearly impossible to diagnose. gdb and backtrace you will not find a happy trail of bread crumbs to the bug's cottage. Web search for 'portable linux binary' leads to lots of problem reports. Example, search 'Appimage crash':

http://www.openshotvideo.com/2016/08/openshot-21-released.html

in other words, even if these systems are used, in my humble opinion extensive testing must happen for there to be any guarantee that the package will work properly on a given system for the users. however it has the advantage of 'one build on one system', so if the tests come back positive, then perhaps it may be suitable.

probonopd, 2016-11-01:
An AppImage should be entirely doable, especially given that this project is already building on Travis CI. See [linuxdeployqt](https://github.com/probonopd/linuxdeployqt) and especially the [Travis CI examples](https://github.com/probonopd/linuxdeployqt#projects-using-linuxdeployqt) linked there. Basically `linuxdeployqt -appimage` should be able to do it automatically for you (but it hasn't been tested on such a large and complex application like OpenSCAD so far - so be prepared to file some issues on it).

If you don't want to use linuxdeployqt, then you can use [this more manual approach](https://github.com/search?utf8=%E2%9C%93&q=%22Package+the+binaries+built+on+Travis-CI+as+an+AppImage%22&type=Code&ref=searchresults).

donbright, 2016-11-03:
Thanks, that is very impressive. Is there a "linter" for AppImage that will prevent the types of problems seen with OpenShot?

probonopd, 2016-11-03:
There s a very basic one at https://github.com/probonopd/AppImages/blob/master/appdir-lint.sh but it probably wouldn't be able t recognize this type of issues yet. So nothing really replaces testing on different distributions.

kintel, 2016-11-03:
@donbright For Mac OS X bundles, I wrote a script for sifting through all dependencies checking that they were built and linked correctly. Not perfect but it does catch common build errors:
https://github.com/openscad/openscad/blob/master/scripts/macosx-sanity-check.py

probonopd, 2016-11-04:
That is very cool @donbright. From the result, how is this different from running `ldd` (Linux) or `otool -L` (macOS) on all binaries in the bundle and searching for "not found"?

kintel, 2016-11-04:
@probonopd The script essentially walks all the dependencies using `otool -L`, making sure that they are referenced and located in the correct location.
In addition, there are platform specific issues we look for (on Mac):

* Mixing of C++ library versions (Common issue on Mac)
* Wrong target SDKs (OS X does run-time resolution of functions in libraries based on target version, so you can build for older libs using the latest SDK. We try to detect this without having to physically test on all previous versions of OS X).
* Mixing of CPU architectures in fat binaries (also used to be a common issue on Mac, but now it's mostly 64-bit Intel)

probonopd, 2016-11-13:
@t-paul does my example AppImage work for you?

t-paul, 2016-11-13:
@probonopd thanks a lot for your support. I didn't have the time to test that image yet. I had one working locally based on our release build script, so in general the packaging should be compatible with our internal file loading stuff (settings, examples, ...). I'll give yours a try, hopefully some free time coming up soon. On what system / environment are those images built?

As for the type of packaging, I'm not a fan of repackaging the debs, it might be the simplest solution, but feels like it would be quite hard to resolve issues if someone has problems.

Using Travis sounds like an interesting approach, the only drawback there is we would be stuck with all dependencies from the older system they run. But then as long as they support external package archives, that could be solved, I guess.

probonopd, 2016-11-13:
> Using Travis sounds like an interesting approach, the only drawback there is we would be stuck with all dependencies from the older system they run.

Which is _good_ - you _want_ to build on old systems, so that the resulting binaries can run on this and on newer systems. In my experience, Travis is excellent for building AppImages.

(The example linked above merely repackages trusty ingredients into an AppImage.)

kintel, 2016-11-13:
@probonopd I might have misunderstood something, but isn't one of the major advantages of smth. like AppImage that we could start using more modern technologies while targeting older distros?
Things I'd like to use: Proper C++11 (i.e. post-gcc-4.6), newer CGAL with important bugfixes and features, more recent Qt.

..which I guess would be possible to do on an old system, just somewhat of a pain to set up all the compiler env and dependencies?

t-paul, 2016-11-13:
Yep, it's two different things to consider. It's good to build on an older system so we get a reasonable compatible glibc and other system stuff. But for the libraries we would bundle, especially CGAL and OpenCSG, we really want the latest version to get bug fixes.

probonopd, 2016-11-13:
Using recent compilers and recent Qt on a mature build system should not be a problem.

kintel, 2016-11-13:
@probonopd I tried it on Ubuntu 12.04 x64:
```
$ ./OpenSCAD.AppImage path/to/file.scad -o out.stl
```
->
```
Can't open input file 'path/to/file.scad'!
```

The same cmd-line works with a locally built openscad binary.
Not sure where to go from there. Any ideas?

probonopd, 2016-11-14:
Please retry with an absolute path as the argument. If that works I may have a solution.

kintel, 2016-11-17:
@probonopd Absolute path works. The output file needs to be absolute as well.

probonopd, 2016-11-17:
Try `union: true` as the second line of your recipe and see if relative paths work then.

probonopd, 2017-01-15:
Example files were missing in the welcome screen. `strace` reveals that OpenSCAD uses `chdir()`. In combination with the workaround that AppImage uses for making compiled binaries relocateable, this failed, since it apparently results in OpenSCAD looking for examples/ in `$APPDIR/usr/bin`, as `strace` also reveals. Fixed the example AppImage by using a symlink.

probonopd, 2017-01-15:
Also uploaded __OpenSCAD version 2017.01.09.nightly (git acb5331) __.
Please try it out and let me know how well it works for you. Some fine-tuning might be needed, which is expected.

```
wget -O OpenSCAD-20170109T193006.gitacb5331.glibc2.14-x86_64.AppImage "https://bintray.com/probono/AppImages/download_file?file_path=OpenSCAD-20170109T193006.gitacb5331.glibc2.14-x86_64.AppImage"
chmod a+x OpenSCAD-20170109T193006.gitacb5331.glibc2.14-x86_64.AppImage
./OpenSCAD-20170109T193006.gitacb5331.glibc2.14-x86_64.AppImage
```

t-paul, 2017-01-15:
Works fine on Debian/Testing (soon to be released Debian9).

donbright, 2017-01-25:
basic testing on ubuntu 16.04 looks ok

how would one run the regression test suite using appimage though? very very interesting system.

probonopd, 2017-01-25:
> how would one run the regression test suite using appimage though? 
Not sure I understand the question - just like with the installed app?

kintel, 2017-01-25:
Since the regressions tests are black box tests, we could even modify it to be able to run against any binary.

probonopd, 2017-03-17:
So, where do we want to take it from here?

kintel, 2017-03-18:
It sounds like a good start would be getting this into either Travis or OBS for development snapshots would be a good way forward. I'm not sure which is the better option. Travis can push to our file server, not sure if OBS can do that (@t-paul?), but OSB can also build release builds while some work is needed for Travis to manage that.

t-paul, 2017-03-18:
No, OBS builds have no internet connection, only services to import stuff upfront. It probably could publish the packages as *.deb to retrieve and unpack.

probonopd, 2017-08-09:
@t-paul I have started integrating AppImage building into the Travis CI pipeline.

- https://github.com/probonopd/openscad/blob/appimage/.travis.yml
- https://github.com/probonopd/openscad/blob/appimage/scripts/travis-ci.sh

I want to use Qt5 because a) the only way to get it ever working properly is to exercise it, and what would be better suitable for that than continuous builds, b) we have Qt 5.9 now which should have additional bugfixes, c) it's the inevitable future, and d) the main reason, [linuxdeployqt](https://github.com/probonopd/linuxdeployqt) does not handle Qt 4. 

However, I am failing at getting Scintilla to be recognized. Can you have a try please? I am pretty convinced that once we get this to build with Qt5 on Travis CI, I can make an AppImage of it rather easily.

probonopd, 2017-08-09:
Edited above

openelearning, 2017-10-13:
Hello,
For an open source and open hardware project for schools, I'm trying to make OpenScad 2015 or higher running on H3 nano-computer (Orange Pi PC+) with armbian ubuntu xenial legacy, that run armhf. Openscad is not compiled for armhf because of bugs (cf. https://forum.armbian.com/index.php?/topic/5356-how-to-make-available-openscad-2015-in-oppc-with-armbian-desktop-legacy/&tab=comments#comment-41129 ). I have seen that there is appImage that could run for arm. Somebody knows if I try to learn how to make an appimage, this could work on H3 nano-computer (armhf) even if there is bug for traditional compilation ? (does appimage could be a workaround ?)

I have also tried to clone @t-paul OBS config and adding arm, but there is problems. I'm newbie. Linuxdeployqt could also work for this goal ?
Thanks

probonopd, 2017-10-14:
>  I'm newbie.

Then it's better to ask the original authors of an application to provide an official AppImage rather than trying to do it yourself.

t-paul, 2017-10-14:
Any pointers where that could be built? We don't have any infrastructure to provide ARM builds at this time. I can have a look if OBS can do OpenSCAD ARM builds. Although that may take a while due to lack of spare time.

Raspbian actually has OpenSCAD in the [repository](http://archive.raspbian.org/raspbian/pool/main/o/openscad/), so it might be possible in general.

openelearning, 2017-10-14:
> Then it's better to ask the original authors of an application to provide an official AppImage rather than trying to do it yourself.
Ok thanks for your reply. Maybe the best way is to open a new thread on the openscad forum, no ?

>Any pointers where that could be built? We don't have any infrastructure to provide ARM builds at this time. I can have a look if OBS can do OpenSCAD ARM builds.
Thanks a lot, you could save the project :) About arm build, we can have this information, I don't know if it's could be helpful : 
* How to run x86 AppImages on ARM using QEMU https://github.com/AppImage/AppImageKit/issues/121
* Provide binaries for ARM https://github.com/AppImage/AppImageKit/issues/111

> Raspbian actually has OpenSCAD in the repository, so it might be possible in general.
Oh yes you're right, and it's armhf, perfect, I will test it on the OPPC+

openelearning, 2017-10-14:
> I will test it on the OPPC+
When I try to install it on armbian ubuntu xenial desktop legacy there is a status error with gdeby : Dependency is not satisfiable:libboostfilesystem1.55.0

openelearning, 2017-10-14:
I don't know if this can have an impact on AppImage, but there is a reply from Chrysn that have participate on talking about the openscad bug on armfh : https://forum.armbian.com/index.php?/topic/5356-how-to-make-available-openscad-2015-in-oppc-with-armbian-desktop-legacy/&tab=comments#comment-41422

t-paul, 2017-10-30:
OBS does now support building AppImages. For now this is still Qt4, I have to investigate Qt5 builds. It seems QScintilla is only available in OpenSUSE 42.3 which would be a bit against the AppImage idea to not build on the latest version to improve compatibility.

I've configured the nightly builds (= git master branch) and here's the result:

[OpenSCAD-latest-x86_64.AppImage](https://download.opensuse.org/repositories/home:/t-paul/AppImage/OpenSCAD-latest-x86_64.AppImage) - Intel 64-bit, no 32-bit at this time, it seems the AppImage support for 32-bit does not work at this point
[OpenSCAD-latest-armv7l.AppImage](https://download.opensuse.org/repositories/home:/t-paul/AppImage.arm/OpenSCAD-latest-armv7l.AppImage) - Starts on RaspberryPI, but rendering glitches a lot, it seems to run on Mesa 3.0
[OpenSCAD-latest-aarch64.AppImage](https://download.opensuse.org/repositories/home:/t-paul/AppImage.arm/OpenSCAD-latest-aarch64.AppImage) - Totally untested I don't have any 64 bit ARM board

openelearning, 2017-10-30:
Wow... that's just fantastic you did i t! Works very well in x86 and... armv7 on OrangePiPC+ ! 

I will try to test it on arm 64 with OrangePiWin+ but today I failed to use an OS on the screens that I have. I will try other OS on OrangePiWin+ and if I can display it on a screen, I will give feedback about the OpenSCAD-Nightly-latest-aarch64.AppImage.

Thanks a lot, that's really so cool that OpenScad can now run on H3 boards :)

probonopd, 2017-10-30:
> no 32-bit at this time, it seems the AppImage support for 32-bit does not work at this point

Can you elaborate?

t-paul, 2017-10-31:
I don't know what's going on in the background, it's just complaining about lots of missing packages https://build.opensuse.org/package/show/home:t-paul/OpenSCAD-Release and setting the build to "unresolvable".

Config includes the graphics repo which provides `libopencsg-devel`.
```
  <repository name="AppImage.arm">
    <path project="graphics" repository="openSUSE_Leap_42.2"/>
    <path project="OBS:AppImage" repository="AppImage.arm"/>
    <arch>armv7l</arch>
    <arch>aarch64</arch>
  </repository>
  <repository name="AppImage">
    <path project="graphics" repository="openSUSE_Leap_42.2"/>
    <path project="OBS:AppImage" repository="AppImage"/>
    <arch>x86_64</arch>
    <arch>i586</arch>
  </repository>
```

probonopd, 2017-10-31:
@adrianschroeter any idea what might be causing this?

[image]

https://build.opensuse.org/package/show/home:t-paul/OpenSCAD-Release

probonopd, 2017-10-31:
@t-paul looks like for _aarch64_ (ARM 64-bit) compilation fails with

> [ 1419s] /usr/lib64/gcc/aarch64-suse-linux/4.8/../../../../aarch64-suse-linux/bin/ld: objects/cgalutils.o: undefined reference to symbol '_ZN5boost7atomics6detail8lockpool12get_lock_forEPVKv'

Any idea what might cause this?

openelearning, 2017-10-31:
Maybe it's linked to the cgal problems described here : https://forum.armbian.com/topic/5356-solved-how-to-make-available-openscad-2015-in-oppc-with-armbian-desktop-legacy/?tab=comments#comment-41422 ?

t-paul, 2017-10-31:
Not sure about the build error yet, that's probably something in the (old) release code of OpenSCAD itself (e.g. need to link against boost atomic explicitly). It's working in the branch I tried to build in https://build.opensuse.org/package/show/home:t-paul/OpenSCAD-Branch-InputDriver

probonopd, 2017-11-24:
According to
https://abi-laboratory.pro/index.php?view=navigator&symbol=_ZN5boost7atomics6detail8lockpool12get_lock_forEPVKv#result
the symbol `_ZN5boost7atomics6detail8lockpool12get_lock_forEPVKv` was removed from `libboost_atomic.so` in version 1.56.

Overall, Boot's ABI stability appears to be less than stellar: https://abi-laboratory.pro/tracker/timeline/boost/

donbright, 2017-12-11:
@probonopd yes. OpenSCAD has a long history of workarounds for boost's API stability problems , for example src/boosty.h , but of course openscad release code from 2015 can't anticipate what boost from 2017 will break in it's API. I don't know much about OpenSUSE's build system or AppImage but the traditional kludge here would be to force it to build against an older version of boost if possible.

t-paul, 2018-04-06:
I'll close this as we have AppImage on OBS now. Sadly just Intel/64-bit right now, I can't get any of the other builds (Intel/32-bit, ArmV7, AArch64) to work anymore.

closed 2018-04-06 by t-paul

probonopd, 2018-04-06:
@adrianschroeter do you have an idea why the other architectures may fail?

probonopd, 2018-04-08:
Not good. Can you `strace` it?

t-paul, 2018-04-08:
In one of the AppImage I found some Qt4 libs even though it's compiled with Qt5 but that does not seem to be the actual problem. Fixing that using `-qmake=<path>` option it's still getting the BusError.

The `strace` output does not seem to help much
```
mprotect(0x7f366c4ca000, 8192, PROT_READ) = 0
mprotect(0x7f366c6f4000, 8192, PROT_READ) = 0
mprotect(0xb36000, 57344, PROT_READ)    = 0
mprotect(0x7f366c924000, 4096, PROT_READ) = 0
set_tid_address(0x7f366c907b10)         = 29481
set_robust_list(0x7f366c907b20, 24)     = 0
rt_sigaction(SIGRTMIN, {sa_handler=0x7f3667668d60, sa_mask=[], sa_flags=SA_RESTORER|SA_SIGINFO, sa_restorer=0x7f3667672c10}, NULL, 8) = 0
rt_sigaction(SIGRT_1, {sa_handler=0x7f3667668df0, sa_mask=[], sa_flags=SA_RESTORER|SA_RESTART|SA_SIGINFO, sa_restorer=0x7f3667672c10}, NULL, 8) = 0
rt_sigprocmask(SIG_UNBLOCK, [RTMIN RT_1], NULL, 8) = 0
getrlimit(RLIMIT_STACK, {rlim_cur=8192*1024, rlim_max=RLIM64_INFINITY}) = 0
--- SIGBUS {si_signo=SIGBUS, si_code=SI_KERNEL, si_addr=NULL} ---
+++ killed by SIGBUS +++
Bus error
```

probonopd, 2018-04-09:
Maybe running `LD_DEBUG=libs ./the.AppImage` shows something interesting?

probonopd, 2018-04-09:
Please make sure that __none__ of the libraries listed on https://github.com/AppImage/AppImages/blob/master/excludelist are included in the AppImage.

t-paul, 2018-04-09:
how? that is OBS doing tha packaging.

probonopd, 2018-04-09:
To be sure that this is actually the reason, please run your AppImage with `--appimage-extract`, then

```
# Source helper functions
wget -q https://github.com/probonopd/AppImages/raw/master/functions.sh -O ./functions.sh
. ./functions.sh

cd squashfs-root/

# Delete blacklisted libraries that should not go into the AppImage
delete_blacklisted

# Check if it runs
./AppRun
```

t-paul, 2018-04-09:
Yep, that helped. Nice!

How do I integrate that in the OBS build which just calls linuxdeployqt. Is there some flag to pass/enable that blacklist?

probonopd, 2018-04-09:
Since you are using `linuxdeployqt`, it _should_ actually do this automatically (and it used to). Possibly this is a bug in `linuxdeployqt` recently introduced by changes in how the `linuxdeployqt` gets its excludeilst? cc @TheAssassin @patrickelectric

t-paul, 2018-04-09:
What does it mean "dynamically generated from the AppImage repository"? Are we talking network access?

probonopd, 2018-04-09:
No, as this is done at build time...

t-paul, 2018-04-09:
Even at build time, if it's network access, that's a no-go at OBS as the `linuxdeployqt` is built on OBS aswell, I think? OBS does not have any network access once the service loader is done preparing. The full built can only use the files downloaded upfront by the service thingy.

probonopd, 2018-04-09:
You might have a _great_ point there... cc @patrickelectric @TheAssassin

t-paul, 2018-04-09:
Yep, actually it fails with `wget` not found, but even with that present, that would not work I think. https://build.opensuse.org/package/live_build_log/OBS:AppImage/linuxdeployqt/toolchain/x86_64

probonopd, 2018-04-09:
Thanks @t-paul, opened https://github.com/probonopd/linuxdeployqt/issues/272