openscad/openscad #1853: STL export, wrong facet normal values

indazoo, 2016-11-04:
Hi

I experienced hole errors in netfabb and meshlab when loading the STL files from OpenScad.
First I thought the error must lie between my ears but in this case it seems not to be the case.

In a STL file, for each triangle there are three vertexes and for these a norm vector.
After repairing my STL file in Netfabb I realised the norm vector changed for a lot of values between these two STL files. The vertexes stayed the same.
So below you find some OpenScad code from which I believe does calculate the norm vector correctly.
I took the triangle vertexes from OpenScad's ascii export, used them as points in OpenScad and then calculated the normal vectors.
The norm vectors found in OpenScad's STL file were not(!) corresponding with the calculated  vertexes. The netfabb output showed way better acurracy. The differences are quite significant.
I used version 2016.10.04 (git 7e0935d)  and first 2015.04.25 (git 2b33922) .

case1_pt1 = [2.40086, -3.19936, 10];
case1_pt2 = [2.40086, -3.19936, 9.96858];
case1_pt3 = [2.45734, -3.15223, 10];
OpenScad_pt_1_stl_export = [0.640593, -0.76788, 0];
Netfab_corrected_pt_1 = [0.640692, -0.767798, 0.000000];
echo("***************");
echo("**  CASE 1   **");
echo("OpenScad norm vector (ascii stl export) = ",  OpenScad_pt_1_stl_export);
echo("Calculated from pts = ", triangle_normal_stl(case1_pt1, case1_pt2, case1_pt3) );
echo("Netfabb repaired norm vector (ascii stl export)= ", Netfab_corrected_pt_1);

case2_pt1 = [2.21668, -3.32328, 10];
case2_pt2 = [2.05917, -3.42926, 9.9189];
case2_pt3 = [2.40086, -3.19936, 10];
OpenScad_pt_2_stl_export = [0.558244, -0.829677, -7.70895e-016];
Netfab_corrected_pt_2 = [0.558230, -0.829686, 0.000039];
echo("****************");
echo("**  CASE 2    **");
echo("OpenScad norm vector (ascii stl export) = ",  OpenScad_pt_2_stl_export);
echo("Calculated from pts = ", triangle_normal_stl(case2_pt1, case2_pt2, case2_pt3) );
echo("Netfabb repaired norm vector (ascii stl export) = ", Netfab_corrected_pt_2);

function triangle_normal_stl(p1, p2, p3) =
	let(tria_norm = triangle_normal(p1, p2, p3),
			k=sqrt(1/(pow(tria_norm.x,2)+pow(tria_norm.y,2)+pow(tria_norm.z,2)))
			)
	[
		k*tria_norm.x, k*tria_norm.y, k*tria_norm.z
	];

function triangle_normal(p1, p2, p3) =
	let(U = p2 - p1, V=p3 - p1)
	[

	U.y*V.z - U.z*V.y,
	U.z*V.x - U.x*V.z,
	U.x*V.y - U.y*V.x
];

t-paul, 2016-11-04:
I don't think there's much way for actually *wrong* calculation. I think the difference comes from [calculating](https://github.com/openscad/openscad/blob/master/src/export_stl.cc#L61) the normal based on the original vertex values which are then truncated to single precision when writing the output file.
Maybe it's an option to recalculate the normal after the truncation happened.

indazoo, 2016-11-04:
Yes, this should be done like you said. The stl file should be correct in itself.

indazoo, 2016-11-11:
I wanted to put a bounty on this. But yesterday and today bountysource.com is down. Anybody knows if BountySource with Github still works ?

t-paul, 2016-11-11:
Hmm, the integration looks a bit broken, but there was an update 10 days ago which was posted via mail. Also https://www.bountysource.com/teams/openscad works for me.

We may need to update our repo config.

kintel, 2016-11-12:
@indazoo Looks like it's just an issue with the bountysource github issue integration: https://github.com/bountysource/core/issues/1062

The rest of bountysource is officially ok.

indazoo, 2016-11-12:
Yes, as of today bountysource.com is up again. But it was not for at least two days. Since it is my first use of bountysource I was a bit sceptic. You know, money and websites which are down....hmmm
I try to do it today or tomorrow.

kintel, 2016-11-12:
I believe this normal vector issue is due to us writing double precision vertices to the STL file while Netfabb casts them to single precision on input.

I need to validate that this is the case, but if I'm correct, we're technically writing the correct normal.
The real questions is then whether we should force STL output to be single precision.

nophead, 2016-11-12:
I have to ask, why does it matter if they are slightly wrong? Most programs
ignore them as they are redundant information.

kintel, 2016-11-12:
@nophead I agree, with the two following caveats:

1) There's the report that MeshLab reports a mesh as self-intersecting based on the normal being slightly off. It might just be a bug in MeshLab but it's worth investigating.

2) There a chance that a triangle flips during downconverting to floating point vertices. In that case it could be interesting to have an authoritative normal vector for repair purposes. Likely only useful for us though as that's a pretty messed up workflow.

indazoo, 2016-11-13:
@nophead 
I may be wrong, more so because I had to refresh my vector skills with this project. But I calculated the normals in the sample above and they were not correct. 
The consuming application of the exported STL reads these inaccurate values. Why do you want to blame these third party products?  Forcing them to accept incorrect values by saying "this data is redundant" rather than just make a small additional calculation? We talk here about a few lines of code. I don't have the build chain on my system. But this would have been done in minutes for someone who can open their OpenScad build environment and knows which files to change. But no, we discuss ...
I'll only place a bounty (50$, yes it is small, my wife knows what I spend ....) under the condition that the change will be "corrrect normal vector matching the vertexes". In the ideal case, the ascii export format will be the same as Meshlab's so we can easily make diffs.
This is not a decision I can make.

MichaelAtOz, 2016-11-13:
> 1) There's the report that MeshLab reports a mesh as self-intersecting based on the normal being slightly off. It might just be a bug in MeshLab but it's worth investigating.

This is a recent stl which had significant issues with Shapeways checking software. Basically SW didn't 'see' the deep red collar (the top half of the band). I suspect the pale red may be the 'slightly off' bits?? I don't know what the darker red v's deep red means.

[image]

Turned out, my mistake, was the collar was a very thin (0.05) angled tube, but IIRC it did intersect at the top & bottom with the major cylinder/lower-angled-bit. So the union should have joined up a hollow solid. This could be another, 'very small numbers' issue, but it highlights it is not just a MeshLab fussiness issue.

nophead, 2016-11-13:
@indazoo,
 Yes they might be wrong but only by a minute amount. Floating point is never exact anyway, so I am surprised any program makes topological decisions using the precise value of the normals. 

I say they are redundant because they can easily be calculated from the vertices so its stupid of STL to include them. I think a lot of programs ignore them and calculate the normal themselves if they need it. This will be because other file formats don't include them and also because they are notorious for being wrong or missing in STL files.

kintel, 2016-11-13:
@indazoo I'm trying to reproduce.. In your example above, you're claiming:
`OpenScad_pt_1_stl_export = [0.640593, -0.76788, 0];` and
`OpenScad_pt_2_stl_export = [0.558244, -0.829677, -7.70895e-016];`

Where do these numbers come from?

If I let OpenSCAD build a polyhedron and export, i.e.:
```
polyhedron(points=[case1_pt1,case1_pt2,case1_pt3,case2_pt1,case2_pt2,case2_pt3], faces=[[0,1,2],[3,4,5]]);
```

..I get almost the same normals as you listed for Netfabb:
```
  facet normal -0.640692 0.767798 0
  facet normal -0.55823 0.829686 -4.20592e-05
```

(note the second normal's Z component which differs from Netfabb)

indazoo, 2016-11-13:
Well, I cannot tell you what exact object i created. I'm working on a helix-like module for threads.
I have some test cases which i go through. Sometimes I tweak the numbers by 1/pow(2,50) or so. 
You say "almost". Meshlab complains about very small differences. I cannot say more. You guys decide whether OpenScad exports norm vectors matching the vertexs "exactly" or not. 
You are free! I suppose at least!
I'm too! No exact norm vectors no money! Well, this is not a good argument with such a low amount....:-)
It just would be great if I can push F6, export the STL and netfabb does not show me non-existant holes and meshlab would not make the whole screen red.
After this very important blabla here some investigations:
I ran one of my test cases through. It created the thread and I exported it as STL and opened it in Netfabb. Netfabb showed a hole !
Then I took the echoed data from OpenScad's output and went to another OpenScad instance and created there the same object with this copy pasted text. The exported STL file was completely accepted by netfabb.

kintel, 2016-11-13:
@indazoo While a fix is potentially easy, I need to write a test case where I can repeatedly reproduce the problem. The code you posted initially isn't such a test case, so I'm just asking for a way to reproduce this with a minimal example.

indazoo, 2016-11-14:
Here we go:
Attached you find the OpenScad code. It produces a three start metric thread. Of course a whole bunch of other thread types (see line 380 an later) to play with it. It has some old unused code in it and also some fixes are needed. I wanted to remove that old stuff and make all test before publish it. But it works ok. Nice experience in a functional language for me.
If you want to output the faces and points for the polyhedron then go to line 5250 and uncomment there the echo statements.
[fasteners.zip](https://github.com/openscad/openscad/files/588463/fasteners.zip)

indazoo, 2016-11-19:
There seems to be a bounty on this issue now:
https://www.bountysource.com/issues/38950877-stl-export-wrong-facet-normal-values

kintel, 2016-11-19:
@indazoo Thanks for the design, although as far from "minimal" as it can get ;)

It doesn't makes sense to me that MeshLab or Netfabb would claim there are holes in a mesh just because normal vectors have rounding errors.

Anyway, I had a look at it, exported to STL, imported into MeshLab. I cannot get it to show any errors.

Are you certain that your workflow issues are related to normal vectors? Could you look at e.g. #1867 and see if the fix for that issue improves your workflow? If not, we still need a reproducible test case for this.

indazoo, 2016-11-19:
Here is a new version:
[threads.zip](https://github.com/openscad/openscad/files/601308/threads.zip)
I fixed a bug where the cross point calculation (z=0, z=length) created identical cross points. This resulted in polygons with multiple identical points which confused CGAL.
Warning:There are still CGAL issues if you run test_threads().

indazoo, 2016-11-19:
Oh, I did not read your comment about the sample..
My workflow :
1. Run threads.scad (F6)
2. Export stl in OpenScad  (metric thread)
3. Open Netfabb Basic
4. Load the exported STL file
5. Select the part in Netfabb
6. In Netfabb Menu save as STL (ascii) under a different name
7. Diff the two files
8. You see a lot of different normal vectors.

MichaelAtOz, 2016-11-19:
Note: When you open a stl in Meshlab, the checkbox 'unify duplicated vertices' is checked by default.
If you wish to look at faults in the stl, uncheck this box. 
See #1854 for two example views, with unchecked & checked.

indazoo, 2016-11-19:
Thanks for the tip Michael.
If you guys prefer Meshlab, this is OK for me. I could switch to it. But there is the problem.
Meshlab is even more accurate. So, if one filters for faults the hole part is red. Defenetly NOT helpful in a situation I try to debug. I assume, with correct normal vectors, a lot of red faces will go away and also "unify duplicate vertices" will have to correct only the "real" errors. Which, with the same output format could be easily detected by a diff tool.
So, if OpenScad would output in Meshlab format....I'll offer several virtual bows before the coder master....;-)

nophead, 2016-11-19:
I don't think correcting small errors in the normals will make any difference at all unless the sign changes.

indazoo, 2016-11-19:
But I do.

nophead, 2016-11-19:
Well all my objects generated by OpenSCAD are fine in Netfabb, despite they fact they must have the same small errors in the normals. Holes in a mesh mean some triangle edges are not shared between exactly two triangles. Nothing to do with normals.

kintel, 2016-11-19:
@MichaelAtOz I don't think unchecking "unify duplicated vertices" will do much good in MeshLab. In STL, all shared vertices are duplicated by definition, so you need to merge them in order to get a valid mesh. If you don't, MeshLab will, IMO, just report on any random calculation depending floating point inaccuracies in their code.

kintel, 2016-11-19:
@indazoo Just to clarify any confusion: Are you saying that the only issue you encounter is that you can detect changes in normal vectors after exporting from Netfabb? ..and that this doesn't otherwise break your workflow?

indazoo, 2016-11-19:
You probably misunderstood something. Like so often, there are two issues at the same time. I want to find in my code the problems/errors (issue 1) and to solve this I would be happy if a secondary program Meshlab/Netfabb only shows the real errors and not inaccuracies (issue 2).
This github issue is about to fix "issue 2" and has nothing to do with the holes per se. Maybe my description of the problem was misleading. I don't expect that fixing this issue will solve all my problems. For example, the last thing I fixed was because some points had a z-value of 4.456E-16 and not zero.
However, the STL output is in itself obviously wrong and fixing this would help me. The change will not help you, because you never had problems. But it also will not create problems for you, because you were happy with slightly off normal vectors.
Even if all red facets in Meshlab are based on my wrong polyhedron, even then it would be a change to the better.

nophead, 2016-11-19:
Yes it won't cause any problems but it won't fix anything either because they simply aren't used to determine non-manifold errors. I doubt they are used at all because they convey no useful information. The topology is completely defined by the vertices.

indazoo, 2016-11-19:
OK you are right.
Can now someone with a drive to excellence fix the issue or not?

MichaelAtOz, 2016-11-20:
> Well all my objects generated by OpenSCAD are fine in Netfabb

Do people in general just rely on the automatic checks when loading a STL, or examine the "+" repair view? Often degenerate faces show which the automatic check doesn't trigger on.

What about executing the Repair/Detect-self-intersections? Not automatic either.

Previously (not for a while now) I just assumed the automatic checks on loading the STL was enough...not so.

(just sayin for the wider audience)

kintel, 2016-11-20:
@indazoo Thanks for the clarification. Tagged as Low Hanging.

Still don't see how this aids much with debugging, but your wish is clear. One possible test: Importing and directly exporting any STL file which is already exported from OpenSCAD should result in exactly the same STL file.

indazoo, 2016-11-20:
@kintel 

> 
Just to clarify any confusion: Are you saying that the only issue you encounter is that you can detect changes in normal vectors after exporting from Netfabb? ..and that this doesn't otherwise break your workflow?
> 
Now at this stage of the discussion what I'm saying and what I'm thinking are different things.
How I see the problem:
I suspect (as much nophead "doubts") that Meshlab shows intersecting triangles as red warning when filtering for it which norm vectors are not correct based on the fact, the values of the vertices do not change and the number of points do not change, only the normal vectors change.

All I ask for, is your help to fix the norm vectors so I can spot the changes/errors better if Netfabb/Meshlab repaired the part. So it is an exclusion of a possible error.

To be honest, I'm surprised, that fixing an obvious issue (the norm vectors are obviously wrong) creates such a strong opposition. Even more it does not impact core code in OpenScad at all. I don't ask for an echo in functions which would have saved a lot of time or anything exotic stuff.

As I typed this, I saw your marking to "Low hanging" and thank you for that.

indazoo, 2016-11-29:
Hi I just published a version which has no CGAL errors any more. Anyone interested in a OpenScad helix-like function with a custom 2D profile as input or has an affinity to threads can look here:
https://github.com/indazoo/MCAD/tree/master/fasteners
Files needed: helix.scad, threads.sad and optionally thread_tests.scad.
I'll put it on Thingiverse later.

If anybody can activate the bounty on this issue so it is visible in this projects the issues overview I would appreciate that. I found only a method on bountysource to publish the bounty to my linked GitHub projects not your project. 
https://www.bountysource.com/issues/38950877-stl-export-wrong-facet-normal-values

t-paul, 2016-11-29:
Ok, I've linked it manually as it seems the automatic update is still not working.

thehans, 2017-10-18:
I've taken a stab at this.  I tested as @kintel suggested, 
> Importing and directly exporting any STL file which is already exported from OpenSCAD should result in exactly the same STL file.

The object I tested for this was:
```
sphere(100, $fn=100);
```
(It passed the test)

indazoo, 2018-09-11:
I'l check tonight with the above mentioned ec4cdec version and maybe latest version.

t-paul, 2018-09-11:
If it works as intended, we can close the issue and @thehans can then claim the bounty. Every backer has 2 weeks to accept or reject the claim.

indazoo, 2018-09-17:
hi kintel, there seems to be a misunderstanding/confusion of "scientific format", "floating point format" and decimals on my side. 
However, the STL file format article says:

> where each n or v is a floating-point number in sign-mantissa-"e"-sign-exponent format, e.g., "2.648000e-002"<

We all know, Wikipedia has many false claims and is heavily influenced. Maybe this is just what the editor  of the Wikipeadia article knew?
I can accept that you don't want to change that only for me. You surely have much else coding to do. I think it should be mantissa/exponent format, but of course maybe my liking is heavily biased by my wants? ;-)
Do we wait for a reaction of "thehans" or should I close the issue? I will pay him even the thread is closed. But he must claim it. Also, you may close the issue if you want.

indazoo, 2018-09-17:
Oh, and do you know the identity of thehans? Some evil person could claim it on Bountysource. Maybe I claim it via my wife's email? ;-)))

kintel, 2018-09-17:
The wikipedia source is a book from 1993: https://en.wikipedia.org/wiki/STL_(file_format)#cite_note-burns-6

I think the STL file format has evolved naturally from there. e.g. I regularly see STL files where the entire models is _not_ constrained to the positive octant. The one thing to keep an eye out for is that readers may interpret numbers as single-precision floats, so writing floats with higher precision may cause slight deformation or topology changes for tiny details.

kintel, 2018-09-17:
@thehans is relatively active here - let's just wait for him to surface

thehans, 2018-09-17:
Hi, I have just registered for bountysource (username also "thehans" ), authenticated through my github account.
This is my first time claiming a bounty.  Closing now to try to move this forward.

closed 2018-09-17 by thehans