openscad/openscad #1854: CGAL error with second object

indazoo, 2016-11-04:
Hi
Below is my part as I create it as polyhedron. It renders fine (F6) and can be exported to STL. Netfabb is happy with it.
But as soon I add another part, the CGAL error kicks in.
What shall I do? Any tips welcome.

Output: 
```
Saved backup file: C:/Users/User/Documents/OpenSCAD/backups/unsaved-backup-Hp148460.scad
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /data/OpenSCAD/libraries-mingw64-master/mxe-w64/usr/x86_64-w64-mingw32.static/include/CGAL/Nef_S2/SM_const_decorator.h Line: 326 
Geometries in cache: 27
Geometry cache size in bytes: 251728
CGAL Polyhedrons in cache: 4
CGAL cache size in bytes: 11104
Total rendering time: 0 hours, 0 minutes, 0 seconds
   Top level object is a 3D object:
   Simple:        yes
   Vertices:        8
   Halfedges:      24
   Edges:          12
   Halffacets:     12
   Facets:          6
   Volumes:         2
Rendering finished.
```

Code:

```
translate([10,10,10])
cube([2,2,2]);

polyhedron(points = [[0, 0, -4], [2, 0, -4], [4.46705, 0, -4], [4.46705, 0, -4], [5.54958, 0, -3.375], [4.46705, 0, -3.375], [5.54958, 0, -3.175], [4.46705, 0, -3.175], [4.46705, 0, -2.55], [4.46705, 0, -2.55], [4.46705, 0, -2], [4.46705, 0, -2], [4.46705, 0, -1.375], [4.46705, 0, -1.375], [4.46705, 0, -1.175], [4.46705, 0, -1.175], [4.46705, 0, -0.55], [4.46705, 0, -0.55], [4.46705, 0, 0], [4.46705, 0, 0], [4.46705, 0, 0.625], [4.46705, 0, 0.625], [4.46705, 0, 0.825], [4.46705, 0, 0.825], [4.46705, 0, 1.45], [4.46705, 0, 1.45], [4.46705, 0, 2], [4.46705, 0, 2], [4.46705, 0, 2.625], [4.46705, 0, 2.625], [4.46705, 0, 2.825], [4.46705, 0, 2.825], [4.46705, 0, 3.45], [4.46705, 0, 3.45], [4.46705, 0, 4], [4.46705, 0, 4], [2, 0, 0], [0, 0, 0], [0, 0, -3.71429], [1.24698, 1.56366, -3.71429], [2.78516, 3.49248, -3.71429], [2.78516, 3.49248, -3.71429], [3.46011, 4.33884, -3.08929], [2.78516, 3.49248, -3.08929], [3.46011, 4.33884, -2.88929], [2.78516, 3.49248, -2.88929], [2.78516, 3.49248, -2.26429], [2.78516, 3.49248, -2.26429], [2.78516, 3.49248, -1.71429], [2.78516, 3.49248, -1.71429], [2.78516, 3.49248, -1.08929], [2.78516, 3.49248, -1.08929], [2.78516, 3.49248, -0.889286], [2.78516, 3.49248, -0.889286], [2.78516, 3.49248, -0.264286], [2.78516, 3.49248, -0.264286], [2.78516, 3.49248, 0.285714], [2.78516, 3.49248, 0.285714], [2.78516, 3.49248, 0.910714], [2.78516, 3.49248, 0.910714], [2.78516, 3.49248, 1.11071], [2.78516, 3.49248, 1.11071], [2.78516, 3.49248, 1.73571], [2.78516, 3.49248, 1.73571], [2.78516, 3.49248, 2.28571], [2.78516, 3.49248, 2.28571], [1.24698, 1.56366, 0], [0, 0, 0], [0, 0, -3.42857], [-0.445042, 1.94986, -3.42857], [-0.994012, 4.35505, -3.42857], [-0.994012, 4.35505, -3.42857], [-1.2349, 5.41044, -2.80357], [-0.994012, 4.35505, -2.80357], [-1.2349, 5.41044, -2.60357], [-0.994012, 4.35505, -2.60357], [-0.994012, 4.35505, -1.97857], [-0.994012, 4.35505, -1.97857], [-0.994012, 4.35505, -1.42857], [-0.994012, 4.35505, -1.42857], [-0.994012, 4.35505, -0.803571], [-0.994012, 4.35505, -0.803571], [-0.994012, 4.35505, -0.603571], [-0.994012, 4.35505, -0.603571], [-0.994012, 4.35505, 0.0214286], [-0.994012, 4.35505, 0.0214286], [-0.994012, 4.35505, 0.571429], [-0.994012, 4.35505, 0.571429], [-0.994012, 4.35505, 1.19643], [-0.994012, 4.35505, 1.19643], [-0.994012, 4.35505, 1.39643], [-0.994012, 4.35505, 1.39643], [-0.994012, 4.35505, 2.02143], [-0.994012, 4.35505, 2.02143], [-0.994012, 4.35505, 2.57143], [-0.994012, 4.35505, 2.57143], [-0.445042, 1.94986, 0], [0, 0, 0], [0, 0, -3.14286], [-1.80194, 0.867767, -3.14286], [-4.02467, 1.93818, -3.14286], [-4.02467, 1.93818, -3.14286], [-5, 2.40787, -2.51786], [-4.02467, 1.93818, -2.51786], [-5, 2.40787, -2.31786], [-4.02467, 1.93818, -2.31786], [-4.02467, 1.93818, -1.69286], [-4.02467, 1.93818, -1.69286], [-4.02467, 1.93818, -1.14286], [-4.02467, 1.93818, -1.14286], [-4.02467, 1.93818, -0.517857], [-4.02467, 1.93818, -0.517857], [-4.02467, 1.93818, -0.317857], [-4.02467, 1.93818, -0.317857], [-4.02467, 1.93818, 0.307143], [-4.02467, 1.93818, 0.307143], [-4.02467, 1.93818, 0.857143], [-4.02467, 1.93818, 0.857143], [-4.02467, 1.93818, 1.48214], [-4.02467, 1.93818, 1.48214], [-4.02467, 1.93818, 1.68214], [-4.02467, 1.93818, 1.68214], [-4.02467, 1.93818, 2.30714], [-4.02467, 1.93818, 2.30714], [-4.02467, 1.93818, 2.85714], [-4.02467, 1.93818, 2.85714], [-1.80194, 0.867767, 0], [0, 0, 0], [0, 0, -2.85714], [-1.80194, -0.867767, -2.85714], [-4.02467, -1.93818, -2.85714], [-4.02467, -1.93818, -2.85714], [-5, -2.40787, -2.23214], [-4.02467, -1.93818, -2.23214], [-5, -2.40787, -2.03214], [-4.02467, -1.93818, -2.03214], [-4.02467, -1.93818, -1.40714], [-4.02467, -1.93818, -1.40714], [-4.02467, -1.93818, -0.857143], [-4.02467, -1.93818, -0.857143], [-4.02467, -1.93818, -0.232143], [-4.02467, -1.93818, -0.232143], [-4.02467, -1.93818, -0.0321429], [-4.02467, -1.93818, -0.0321429], [-4.02467, -1.93818, 0.592857], [-4.02467, -1.93818, 0.592857], [-4.02467, -1.93818, 1.14286], [-4.02467, -1.93818, 1.14286], [-4.02467, -1.93818, 1.76786], [-4.02467, -1.93818, 1.76786], [-4.02467, -1.93818, 1.96786], [-4.02467, -1.93818, 1.96786], [-4.02467, -1.93818, 2.59286], [-4.02467, -1.93818, 2.59286], [-4.02467, -1.93818, 3.14286], [-4.02467, -1.93818, 3.14286], [-1.80194, -0.867767, 0], [0, 0, 0], [0, 0, -2.57143], [-0.445042, -1.94986, -2.57143], [-0.994012, -4.35505, -2.57143], [-0.994012, -4.35505, -2.57143], [-1.2349, -5.41044, -1.94643], [-0.994012, -4.35505, -1.94643], [-1.2349, -5.41044, -1.74643], [-0.994012, -4.35505, -1.74643], [-0.994012, -4.35505, -1.12143], [-0.994012, -4.35505, -1.12143], [-0.994012, -4.35505, -0.571429], [-0.994012, -4.35505, -0.571429], [-0.994012, -4.35505, 0.0535714], [-0.994012, -4.35505, 0.0535714], [-0.994012, -4.35505, 0.253571], [-0.994012, -4.35505, 0.253571], [-0.994012, -4.35505, 0.878571], [-0.994012, -4.35505, 0.878571], [-0.994012, -4.35505, 1.42857], [-0.994012, -4.35505, 1.42857], [-0.994012, -4.35505, 2.05357], [-0.994012, -4.35505, 2.05357], [-0.994012, -4.35505, 2.25357], [-0.994012, -4.35505, 2.25357], [-0.994012, -4.35505, 2.87857], [-0.994012, -4.35505, 2.87857], [-0.994012, -4.35505, 3.42857], [-0.994012, -4.35505, 3.42857], [-0.445042, -1.94986, 0], [0, 0, 0], [0, 0, -2.28571], [1.24698, -1.56366, -2.28571], [2.78516, -3.49248, -2.28571], [2.78516, -3.49248, -2.28571], [3.46011, -4.33884, -1.66071], [2.78516, -3.49248, -1.66071], [3.46011, -4.33884, -1.46071], [2.78516, -3.49248, -1.46071], [2.78516, -3.49248, -0.835714], [2.78516, -3.49248, -0.835714], [2.78516, -3.49248, -0.285714], [2.78516, -3.49248, -0.285714], [2.78516, -3.49248, 0.339286], [2.78516, -3.49248, 0.339286], [2.78516, -3.49248, 0.539286], [2.78516, -3.49248, 0.539286], [2.78516, -3.49248, 1.16429], [2.78516, -3.49248, 1.16429], [2.78516, -3.49248, 1.71429], [2.78516, -3.49248, 1.71429], [2.78516, -3.49248, 2.33929], [2.78516, -3.49248, 2.33929], [2.78516, -3.49248, 2.53929], [2.78516, -3.49248, 2.53929], [2.78516, -3.49248, 3.16429], [2.78516, -3.49248, 3.16429], [2.78516, -3.49248, 3.71429], [2.78516, -3.49248, 3.71429], [1.24698, -1.56366, 0], [0, 0, 0], [4.46705, 0, 0], [3.36016, 2.29847, 0], [2.78516, 3.49248, 0], [1.59004, 3.76526, 0], [-0.710574, 4.29036, 0], [-0.90509, 4.33476, 0], [-0.994012, 4.35505, 0], [-3.00257, 2.75328, 0], [-4.02467, 1.93818, 0], [-4.02467, 0.585254, 0], [-4.02467, -1.93818, 0], [-3.91771, -2.02348, 0], [-3.68372, -2.21008, 0], [-2.5762, -3.0933, 0], [-1.56226, -3.90189, 0], [-1.17229, -4.21288, 0], [-0.994012, -4.35505, 0], [1.37723, -3.81383, 0], [2.78516, -3.49248, 0], [3.31281, -2.3968, 0], [4.46705, 0, -4]], 
faces = [[2, 4, 42], [2, 42, 40], [4, 6, 44], [4, 44, 42], [6, 8, 46], [6, 46, 44], [8, 10, 48], [8, 48, 46], [10, 12, 50], [10, 50, 48], [12, 14, 52], [12, 52, 50], [14, 16, 54], [14, 54, 52], [1, 11, 9], [1, 9, 7], [1, 7, 5], [1, 5, 3], [4, 2, 5], [7, 4, 5], [6, 4, 7], [9, 6, 7], [66, 36, 1], [66, 1, 39], [40, 42, 72], [40, 72, 70], [42, 44, 74], [42, 74, 72], [44, 46, 76], [44, 76, 74], [46, 48, 78], [46, 78, 76], [48, 50, 80], [48, 80, 78], [50, 52, 82], [50, 82, 80], [96, 66, 39], [96, 39, 69], [70, 72, 102], [70, 102, 100], [72, 74, 104], [72, 104, 102], [74, 76, 106], [74, 106, 104], [76, 78, 108], [76, 108, 106], [78, 80, 110], [78, 110, 108], [80, 82, 112], [80, 112, 110], [126, 96, 69], [126, 69, 99], [100, 102, 132], [100, 132, 130], [102, 104, 134], [102, 13 [... cut ...] [168, 236, 198], [237, 198, 236], [218, 198, 237], [198, 218, 18], [36, 66, 220, 219, 218], [66, 96, 224, 223, 222, 221, 220], [96, 126, 226, 225, 224], [126, 156, 228, 227, 226], [156, 186, 234, 233, 232, 231, 230, 229, 228], [186, 216, 236, 235, 234], [216, 36, 218, 237, 236], [1, 2, 40, 1, 40, 39], [39, 40, 70, 39, 70, 69], [69, 70, 100, 69, 100, 99], [99, 100, 130, 99, 130, 129], [129, 130, 160, 129, 160, 159], [159, 160, 190, 159, 190, 189], [189, 190, 10, 189, 10, 1]], 
convexity = 1);
```

kintel, 2016-11-04:
Do you have a version of that in STL or smth.?
I usually run designs through the 3D printer toolbox in Blender or MeshLab, as Netfabb tends to not catch all issues.

Btw., which Netfabb did you test with? Autodesk Netfabb, Microsoft's Netfabb or some old Netfabb version?

indazoo, 2016-11-05:
I was not aware of so much netfabbs...well mine version is netfabb Basic 5.2.1 (January 2015).

You asked for the STL output. Well, I don't trust/rely very much anymore on the messages from third party products als long OpenScad's output is so inaccurate, see : [https://github.com/openscad/openscad/issues/1853] How do I know whether these filters/repairs in other programs are really needed or not?

The fact is, the code creates the polyhedron as shown above. It renders correctly in OpenScad. No error messages. Then I add another object. It fails. This has nothing to do with other programs. Nothing.
OpenScad should tell me if my halffacets are not on a plane or something else. For example, the output in OpenScad does not allow me to see a difference between z = 10 and z = 10.0000000000000000000045. I may be wrong, but the accuracy between CGAL andOpenScad code is not the same ( I believe I read it somewhere in the comments on another issue here).
Or, it should tell me other errors I did. Maybe I'm too much a "User".
It would be nice, if I don't have to trash my OpenScad script. But in this shape I cannot publish it.

Maybe, if the halffacets are the cause, you can help me with a OpenScad script which fills a polygon whose points are on a plane with triangles. But on the other hand, Mehslab shows also many holes and intersections on triangles not only on the halffactes. How can this be? I have just points connected together.

kintel, 2016-11-05:
An STL would still be helpful. I can manually re-create an STL from your code and run it through Blender myself, but having the STL would save some manual labor and this make it faster for us to help out.

MichaelAtOz, 2016-11-05:
Netfabb doesn't show self intersecting faces, which your poly has:
[image]
(I deselected the unify faces option on loading it)

Below is with unify faces selected, still a little problem.
[image]

indazoo, 2016-11-11:
Thanks a lot for this analysis!
If I knew what self intersecting faces are....how can a triangle intersect itself? Head scratching...

MichaelAtOz, 2016-11-12:
Not intersecting itself, one face intersecting another, rather than sharing edges.
See the images [here](http://blender.stackexchange.com/questions/9175/how-can-i-join-the-geometry-of-intersecting-faces-planes-and-then-remove-parts).

@kintel in my travels I came across [this](http://doc.cgal.org/latest/Polygon_mesh_processing/#title18)
>Self intersections can be detected from a triangle mesh, by calling the predicate CGAL::Polygon_mesh_processing::does_self_intersect(). Additionally, the function CGAL::Polygon_mesh_processing::self_intersections() records all pairs of intersecting triangles.

Perhaps OpenSCAD can test, post render or pre export?

MichaelAtOz, 2016-11-12:
@indazoo also see [this](http://forum.openscad.org/strange-polygon-behaviour-in-small-scale-td17390.html), as your poly has small numbers is could be the issue.

kintel, 2016-11-12:
@indazoo I guess netfabb is more forgiving, but glancing quickly at your indices, I see this: `[159, 160, 190, 159, 190, 189]`. Note the duplicate vertices. I think this is what constitutes the self-intersecting issue.

indazoo, 2016-11-12:
@kintel
Yes, my algorithm flattened one vector with two triangles in it one time too much, resulting in a face vector with six elements. Unfortunately it was not visible in OS preview (F5).
I request (very kindly) for people like me a warning message if faces of the array supplied to polyhedron() function have integrity issues. For me, to prevent annoying you too much I have the below shown OpenScad polyhedron integrity check code. It would be very cool If someone could integrate in OS a warning of this kind if the users push F5. 

After fixing my code, CGAL has no longer an issue with a second object. Cool!
But, Meshlab still complains about self intersecting faces....

@MichaelAtOz 
Thanks for the scale tip. This will give me some work (code changes) so I cannot yet post a answer if this helped. But it could. I'll try.

Tests: I loaded the new STL into Meshlab. Then ok'ed the "unify" message box. After that, there are still the two intersecting polygons as found before by MichaelAtOz. Then I saved the STL so I had a unified version.
I kdiffed the two. Both files have the same number of lines. This means no facet has been unified.
It's an accuracy issue. What is incorrect? I don't know. I checked the corners of the two remaining triangles and found no error. The points have the same values (at least on the screen).
Attached the STL files. One from OpenScad export and on from Meshlab unified.

[Untitled.zip](https://github.com/openscad/openscad/files/587179/Untitled.zip)

```
faces = [[2, 4, 42], [2, 42, 40], [4, 6, 44], [4, 44, 42], [6, 8, 46], [6, 46, 44], [8, 10, 48], [8, 48, 46], [10, 12, 50], [10, 50, 48], [12, 14, 52], [12, 52, 50], [14, 16, 54], [14, 54, 52], [1, 11, 9], [1, 9, 7], [1, 7, 5], [1, 5, 3], [4, 2, 5], [7, 4, 5], [6, 4, 7], [9, 6, 7], [66, 36, 1], [66, 1, 39], [40, 42, 72], [40, 72, 70], [42, 44, 74], [42, 74, 72], [44, 46, 76], [44, 76, 74], [46, 48, 78], [46, 78, 76], [48, 50, 80], [48, 80, 78], [50, 52, 82], [50, 82, 80], [96, 66, 39], [96, 39, 69], [70, 72, 102], [70, 102, 100], [72, 74, 104], [72, 104, 102], [74, 76, 106], [74, 106, 104], [76, 78, 108], [76, 108, 106], [78, 80, 110], [78, 110, 108], [80, 82, 112], [80, 112, 110], [126, 96, 69], [126, 69, 99], [100, 102, 132], [100, 132, 130], [102, 104, 134], [102, 134, 132], [104, 106, 136], [104, 136, 134], [106, 108, 138], [106, 138, 136], [108, 110, 140], [108, 140, 138], [110, 112, 142], [110, 142, 140], [156, 126, 99], [156, 99, 129], [130, 132, 162], [130, 162, 160], [132, 134, 164], [132, 164, 162], [134, 136, 166], [134, 166, 164], [136, 138, 168], [136, 168, 166], [186, 156, 129], [186, 129, 159], [160, 162, 192], [160, 192, 190], [162, 164, 194], [162, 194, 192], [164, 166, 196], [164, 196, 194], [166, 168, 198], [166, 198, 196], [216, 186, 159], [216, 159, 189], [190, 192, 12], [190, 12, 10], [192, 194, 14], [192, 14, 12], [194, 196, 16], [194, 16, 14], [196, 198, 18], [196, 18, 16], [36, 216, 189], [36, 189, 1], [16, 18, 218], [219, 16, 218], [220, 16, 219], [16, 220, 54], [221, 54, 220], [222, 54, 221], [52, 54, 222], [223, 52, 222], [224, 52, 223], [52, 224, 82], [225, 82, 224], [226, 82, 225], [82, 226, 112], [227, 112, 226], [228, 112, 227], [112, 228, 142], [229, 142, 228], [230, 142, 229], [140, 142, 230], [231, 140, 230], [232, 140, 231], [138, 140, 232], [233, 138, 232], [234, 138, 233], [138, 234, 168], [235, 168, 234], [236, 168, 235], [168, 236, 198], [237, 198, 236], [218, 198, 237], [198, 218, 18], [36, 66, 220, 219, 218], [66, 96, 224, 223, 222, 221, 220], [96, 126, 226, 225, 224], [126, 156, 228, 227, 226], [156, 186, 234, 233, 232, 231, 230, 229, 228], [186, 216, 236, 235, 234], [216, 36, 218, 237, 236], [1, 2, 40], [1, 40, 39], [39, 40, 70], [39, 70, 69], [69, 70, 100], [69, 100, 99], [99, 100, 130], [99, 130, 129], [129, 130, 160], [129, 160, 159], [159, 160, 190], [159, 190, 189], [189, 190, 10], [189, 10, 1]];

points = [[0, 0, -4], [2, 0, -4], [4.46705, 0, -4], [4.46705, 0, -4], [5.54958, 0, -3.375], [4.46705, 0, -3.375], [5.54958, 0, -3.175], [4.46705, 0, -3.175], [4.46705, 0, -2.55], [4.46705, 0, -2.55], [4.46705, 0, -2], [4.46705, 0, -2], [4.46705, 0, -1.375], [4.46705, 0, -1.375], [4.46705, 0, -1.175], [4.46705, 0, -1.175], [4.46705, 0, -0.55], [4.46705, 0, -0.55], [4.46705, 0, 0], [4.46705, 0, 0], [4.46705, 0, 0.625], [4.46705, 0, 0.625], [4.46705, 0, 0.825], [4.46705, 0, 0.825], [4.46705, 0, 1.45], [4.46705, 0, 1.45], [4.46705, 0, 2], [4.46705, 0, 2], [4.46705, 0, 2.625], [4.46705, 0, 2.625], [4.46705, 0, 2.825], [4.46705, 0, 2.825], [4.46705, 0, 3.45], [4.46705, 0, 3.45], [4.46705, 0, 4], [4.46705, 0, 4], [2, 0, 0], [0, 0, 0], [0, 0, -3.71429], [1.24698, 1.56366, -3.71429], [2.78516, 3.49248, -3.71429], [2.78516, 3.49248, -3.71429], [3.46011, 4.33884, -3.08929], [2.78516, 3.49248, -3.08929], [3.46011, 4.33884, -2.88929], [2.78516, 3.49248, -2.88929], [2.78516, 3.49248, -2.26429], [2.78516, 3.49248, -2.26429], [2.78516, 3.49248, -1.71429], [2.78516, 3.49248, -1.71429], [2.78516, 3.49248, -1.08929], [2.78516, 3.49248, -1.08929], [2.78516, 3.49248, -0.889286], [2.78516, 3.49248, -0.889286], [2.78516, 3.49248, -0.264286], [2.78516, 3.49248, -0.264286], [2.78516, 3.49248, 0.285714], [2.78516, 3.49248, 0.285714], [2.78516, 3.49248, 0.910714], [2.78516, 3.49248, 0.910714], [2.78516, 3.49248, 1.11071], [2.78516, 3.49248, 1.11071], [2.78516, 3.49248, 1.73571], [2.78516, 3.49248, 1.73571], [2.78516, 3.49248, 2.28571], [2.78516, 3.49248, 2.28571], [1.24698, 1.56366, 0], [0, 0, 0], [0, 0, -3.428 [... cut ...]
            equal   = [ for (y = arr) if (y[0] == pivot) y ],
            greater = [ for (y = arr) if (y[0]  > pivot) y ]
      )
      concat( quicksort_faces(lesser), equal, quicksort_faces(greater) );

	if(len(duplicate_faces) > 0)
	{
		echo("!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!");
		echo("Duplicate faces", duplicate_faces);
	}
	if(len(self_intersecting_faces) > 0)
	{
		echo("!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!");
		echo("Self intersecting faces", self_intersecting_faces);
	}
```

indazoo, 2016-11-12:
Grmpf don't wanted to close...

indazoo, 2016-11-12:
I checked more things:
In Meshlab i deleted the two remaining erroneous facets and let Meshlab fill the two holes.
Then a diff over the files showed to facets changed (as expected). But only the norm vector of the faces was different in the fixed STL. The points were the same. So https://github.com/openscad/openscad/issues/1853   (STL export is incorrect) seems to get more important and should really be done.

The difference is: 
"facet normal 9.00968**9**e-001"   vs  "facet normal 9.00968**8**e-001"

```
  **********************************************************************
  Triangle A
  
  Meshlab Unified, shown as faces intersection:
  facet normal 9.009689e-001 4.338838e-001 0.000000e+000
    outer loop
      vertex  2.785160e+000 3.492480e+000 -1.089290e+000
      vertex  4.467050e+000 0.000000e+000 -1.375000e+000
      vertex  4.467050e+000 0.000000e+000 -2.000000e+000
    endloop
  endfacet

  Meshlab Unified, Meshlab deleted and hole fixed:
  facet normal 9.009688e-001 4.338838e-001 -0.000000e+000
    outer loop
      vertex  4.467050e+000 0.000000e+000 -2.000000e+000
      vertex  2.785160e+000 3.492480e+000 -1.089290e+000
      vertex  4.467050e+000 0.000000e+000 -1.375000e+000
    endloop
  endfacet

  **********************************************************************
  Triangle B
  
  Meshlab Unified, shown as faces intersection:
  facet normal 9.009689e-001 4.338838e-001 0.000000e+000
    outer loop
      vertex  2.785160e+000 3.492480e+000 -2.642860e-001
      vertex  4.467050e+000 0.000000e+000 -5.500000e-001
      vertex  4.467050e+000 0.000000e+000 -1.175000e+000
    endloop
  endfacet

  Meshlab Unified, Meshlab deleted and hole fixed:
  facet normal 9.009688e-001 4.338838e-001 -0.000000e+000
    outer loop
      vertex  4.467050e+000 0.000000e+000 -1.175000e+000
      vertex  2.785160e+000 3.492480e+000 -2.642860e-001
      vertex  4.467050e+000 0.000000e+000 -5.500000e-001
    endloop
  endfacet
```

kintel, 2016-11-12:
@indazoo In terms of integrity checking on input polyhedrons, perhaps we could add that to #1580 ?

indazoo, 2016-11-12:
I'm not upset if you close this issue and insert my wish somewhere else. 
You guys helped me to find an error in my code and if some time later a feature more will be implemented then what can I ask for more??
You could even move  #1853 (STL export is incorrect) to  #1580. This would fit even better. I'll put the bounty where the STL export with correct norm vector will be fixed. Without this, checking the output of OpenScad in other tools is in most cases misleading. I think (maybe  I'm wrong) a not small number of people may have dropped OpenScad because they could not create objects which are correct and got annoyed with correcting every object before printing or otherwise working with them.

janny101, 2018-10-16:
Hii...
    your post is very informative related to using CGAL.
                            Thank you for sharing the Useful post.

Any issue related to printer get help from this       
https://www.brotherprintersupportnumber.com/brother-customer-support/

kintel, 2024-03-27:
Status as of 2024-03-24: This now works when using the Manifold backend

stefano2734, 2026-09-20:
After 2 years a new test with actual CGAL 6.x can show a solved issue.

indazoo, 2026-09-24:
Hi guys, since we have AI now (time passes) so I let it check the vectors for errors and it showed multiple errors. First I assumed the AI hallucinates, unfortunately not, I checked with another AI, the error is between my ears. Embarassing:

Summary of issues                                          Issue Count / Details                                              Severity
-------------------------------------------------------------------------------------------------------------------------------------------
Boundary (open) edges                                  10                                                                            Critical – not a closed solid
Exact duplicate points                                    239	points → only 150 unique location                 High
Unused points                                                152 points never used by any face                          Medium
Degenerate (zero-area) faces                         2                                                                               High
Non-triangular faces                                      7 (5-gons to 9-gons)                                                Medium (allowed but risky)
Non-manifold edges (shared by >2 faces)    0                                                                               Good
Faces with < 3 vertices                                   0                                                                               Good

I believe, the only thing  my vectors are good for are as a test-case if ever someone builds in warnings for the user.
With current openscad version 2026.09.23 (git 28fe66bc) no error is shown.

So, since bountysource is down I don't know if you ever got the money I paid there.
Hmm, I make a second donation.
For me the case is closed although dreaming of a feature where polyhedron() checks the vectors and points out errors for the user.
A big thanks for everybody involved.
Have a nice day!

MichaelAtOz, 2026-09-25:
> since bountysource is down...Hmm, I make a second donation.

Down? DEAD. [In November 2023, the Bountysource parent company announced it had filed for bankruptcy.](https://en.wikipedia.org/wiki/Bountysource)