openscad/openscad #1857: Cursor-centric mousehweel zooming

thehans, 2016-11-06:
This is a feature request to alter the zoom behavior so that points under the mouse cursor stay under the mouse cursor when zooming.   

This is easier to explain and conceptualize when dealing with a strictly 2d application, such as html/js canvas, in which I have successfully implemented this feature before.  In the 2d case the cursor position always corresponds with a single point in "world" space.  The goal then, when cursor-centric zooming, is to keep that world space coordinate corresponding with the same screen-space coordinate during the zoom operation.

Zooming in this way makes it very convenient to zoom out, adjust cursor pos, and zoom in to a new area very quickly without the need to manually pan in between.

For 3d, things are more tricky, and I don't know a solution off the top of my head, since the cursor now represents a ray in 3d space.  I'm hoping maybe with some discussion an appropriate algorithm could be found.  I am fairly certain I have seen behavior like this in other 3d modeling applications.

thehans, 2016-11-06:
I just checked FreeCAD, and it does implement this sort of zooming.

nairaner, 2016-11-06:
If I am correct this is how zoom in SketchUp behaves

BenjamenMeyer, 2018-03-24:
@thehans perhaps a bug on some platforms?

I'm on Kubuntu 17.10 which provides version 2015.03-2 out-of-the-box and am finding it very difficult to work with things since I can only zoom in and out; I can't change where the focus of the zoom is located. So while trying to model large (14 foot long) work benches, buildings (16 feet by 30 feet) or walls (61 inches by 97 inches) it gets very difficult to see how components line up to make sure placement is correct.

May be I'm doing something wrong (hope so), but this bug report/feature request seems related.

thehans, 2018-03-24:
Not a bug, this is a feature request.  Yes the zoom currently works as you describe, no matter which platform.  It currently zooms in/out around the origin, or whatever point you have panned the camera to using right-click-drag.

BenjamenMeyer, 2018-03-25:
@thehans thanks; yes - I understood the original request to be; but from the responses got the impression it might have been viewed as a bug. Either way; I'd love it as the request is more intuitive than the right-click drag - I'll have to try that out. Thanks!

thehans, 2018-11-29:
Just worked out an implementation that seems to function well.  Just wondering if the old behavior should be kept as a configurable preference?
I also left the shift + right click zoom as is, since the cursor changes position as that is adjusted, it wouldn't make much sense.

thehans, 2018-12-07:
Closed by #2611

closed 2018-12-07 by thehans