openscad/openscad #1860: Accept assert() as stable

kintel, 2016-11-08:


MichaelPFrey, 2017-01-07:
I would like to add
- [ ] update syntax highlighting
- [ ] scadlexer.cpp
- [ ] highlighter.cc
to the list

I am working on the documentation:

https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#assert

t-paul, 2017-01-07:
Good point, thanks. Added to he main list.

MichaelPFrey, 2017-01-07:
color-schemes/editor/visualstudio.json has its own keyword list (unlike all the other color schemes).
Those keywords are effective (as in they are applied)

I do not know which of this two alternatives is more sensible:
- Remove the Keyword List from color-schemes/editor/visualstudio.json (prevents further inconsistency, but I do not know if there are side-effects and/other usages of this file)
- Update the Keyword List in color-schemes/editor/visualstudio.json (please check all keywords, for e.g "children" is also missing)

MichaelPFrey, 2017-01-11:
In OpenSCAD, I can overwrite build in functions. Example:
function sin() = true;
echo (sin(1));

but I can not overwrite assertion:
function assert() = true;
echo (assert(1));

closed 2018-10-07 by kintel

t-paul, 2018-10-07:
Added new issue #2506 as followup on the function override comment. I think that's not just specific to `assert()`.

commit 5101d7d5 (2018-10-07): Remove experimental flag of assert() (fixes #1860).
commit 2b39c986 (2018-10-07): Remove experimental flag of assert() (fixes #1860).