openscad/openscad #1862: Preview render error when subtracting from rotate_extrude()

jrslagle, 2016-11-12:
[image]

OpenCSG / Preview / F5, renders the polygons of positive volumes with missing pixels beside where wireframe lines would go. (Perhaps where anti-aliasing pixels would go?) You can see the background through these "snow flecks".

The bug persists regardless of whether [$fs, $fa, $fn] are set. However, doing rotate_extrude(convexity=1) gets rid of the snow but then has convexity issues.

This is the minimal code that reproduces the bug for me (in 2015.03). Essentially it's whenever I delete from a rotational extrusion.

difference() {
    rotate_extrude() {
        translate([1,0]) square(10);
    }
    cylinder(r=4,h=11);
}

[image]

MichaelAtOz, 2016-11-12:
This is what I get w 2015.03-2 (& 2014.03 with a convexity artifact)
[image]

Could be a video driver issue??

> However, doing rotate_extrude(convexity=1) gets rid of the snow but then has convexity issues.

Because that needs a convexity of 4. One for each surface a ray will cross.

kintel, 2016-11-12:
@jrslagle This is mostly likely an issue related to graphics card drivers. Could you do two things?

1) Test with a recent development snapshot to see if it's a solved issue
2) Paste the output of Help->Library Info here

jrslagle, 2016-12-03:
Sorry for taking so long.

I cloned the repo and am trying to compile from source. I'm at a snag trying to install qscintilla2. I'm running ArchLinux.

The check dependencies script says 
qscintilla2 2.7         unknown     NotOK

The link in README.md (https://riverbankcomputing.com/software/qscintilla/intro) directs me to Qscintilla, which is already at version 2.9.3  on my system. "warning: qscintilla-qt4-2.9.3-1 is up to date"

And trying to install "sudo pacman -S qscintilla2" doesn't seem to exist. "error: target not found: qscintilla2"

I synced with my package repository and did all of this again, and no luck.

I also tried bypassing this and compiling anyway. But:
/openscad/scintilla.pri:31: Cannot find feature qscintilla2
Project MESSAGE: Using local copy of qscintilla2.prf instead.
Project MESSAGE: Using qt5scintilla2 as library name
Project MESSAGE: Using C++11
sh: pkg-config: command not found

@kintel You mentioned testing with a recent "development snapshot." Did you mean something else?

kintel, 2016-12-04:
@jrslagle You could try the openscad-git package on Arch: https://aur.archlinux.org/packages/openscad-git/

In terms of building, it may look like you're missing pkg-config. Perhaps we've got a missing dependency check somewhere..

@keenerd any ideas what build or documentation issue this could be?

thehans, 2022-12-08:
Closing stale bug on 7yr old version of OpenSCAD.

closed 2022-12-08 by thehans