openscad/openscad #1864: Simple mesh validation on import/polyhedron

kintel, 2016-11-13:


MichaelAtOz, 2016-11-13:
Note my comment from elsewhere, may be useful, seems many STLs have self intersects.

kintel in my travels I came across [this](http://doc.cgal.org/latest/Polygon_mesh_processing/#title18)

>Self intersections can be detected from a triangle mesh, by calling the predicate CGAL::Polygon_mesh_processing::does_self_intersect(). Additionally, the function CGAL::Polygon_mesh_processing::self_intersections() records all pairs of intersecting triangles.

kintel, 2016-11-13:
@MichaelAtOz Thanks - we've avoided using the Polygon_mesh_processing thus far, as it requires a pretty recent version of CGAL, which has limited availability. Perhaps time to reconsider that..

kintel, 2024-03-27:
Closing this, as we currently do basic mesh repair on import when using the Manifold backend.

closed 2024-03-27 by kintel