openscad/openscad #1870: dependency missing in documentation/scripts libzip-dev

ert485, 2016-11-23:
I needed to run " sudo apt install libzip-dev " before running qmake. Error was:

> qmake openscad.pro
> Project MESSAGE: If you're building a development binary, consider adding CONFIG+=experimental
> Project MESSAGE: Using QOpenGLWidget
> Project MESSAGE: Using C++11
> Package libzip was not found in the pkg-config search path.
> Perhaps you should add the directory containing `libzip.pc'
> to the PKG_CONFIG_PATH environment variable
> No package 'libzip' found
> Package libzip was not found in the pkg-config search path.
> Perhaps you should add the directory containing `libzip.pc'
> to the PKG_CONFIG_PATH environment variable
> No package 'libzip' found

t-paul, 2016-11-23:
It's optional, but yes, it should be added to the documentation...

ert485, 2016-11-23:
Ah I was under the impression that it failed. Did I mention I'm new to this haha

kintel, 2016-11-23:
A nicer warning message wouldn't hurt either, e.g. "Optional libzip library not found - reading compressed AMF files disabled"

t-paul, 2016-11-23:
The warning is from pkg-config, not sure it's possible to change that without hacks (especially with qmake). I do want to add the overview output at the end though.

thehans, 2017-05-02:
Just came across this same issue, overview output would be nice as its really unclear that it apparently succeeded.

thehans, 2017-05-02:
I think it wouldn't hurt to add libzip-dev to scripts/uni-get-dependencies.sh, just to avoid this message.