openscad/openscad #1874: Define rules for internal mesh generation from polyhedron

t-paul, 2016-11-27:
This is from the forum discussion "[Clarifying behaviors](http://forum.openscad.org/Clarifying-behaviors-tp18492p19371.html)"

> **_Parkinbot:_**
> The following two polyhedra are identical and will be forever:
> 
> x = [[0,0,0], [0,10,0], [5,5,0], [2,2,2]];
> polyhedron(x, [[0,2,1], [0,1,3], [1,2,3], [0,3,2]] );
> 
> y = [[0,0,0],[5,5,0], [0,10,0],
>      [0,0,0],[5,5,0], [0,10,0],
>      [0,10,0], [5,5,0],[2,2,2],
>      [0,0,0],[2,2,2],[5,5,0],
>      [0,0,0], [2,2,2], [0,10,0]];
> polyhedron(y, [[0,1,2], [3,4,5], [6,7,8], [9,10,11]] ); 

> **_Ronaldo:_**
> // the following should render without any CGAL error or warnings
> p = [[0,0,0],[10,0,0],[0,10,0],[0,0,10],[10,0,0],[0,10,0],[0,0,10]];
> difference() {
>     polyhedron(points=p, faces=[[0,2,1], [0,1,3], [0,3,2], [4,5,6] ]);
>     translate([5,0,0]) cube(10);
> }

ToDo:
* [ ] Define rules to check against
* [ ] Create test cases to verify those rules

RonaldoCMP, 2016-11-27:
This may not be a rule but it is a good (Kintel's) recipe to start with:
> If a user-defined polyhedron describes a polygon soup, we will _attempt_ to reconstruct a manifold mesh from that polygon soup, in a similar way as from triangle soups in STL files

kintel, 2024-03-27:
Revisiting this:

> x = [[0,0,0], [0,10,0], [5,5,0], [2,2,2]];
> polyhedron(x, [[0,2,1], [0,1,3], [1,2,3], [0,3,2]] );

This is a well-defined polyhedron and describes a valid manifold topology and geometry. It's the ideal way of describing polyhedrons in OpenSCAD.

> y = [[0,0,0],[5,5,0], [0,10,0],
> [0,10,0], [5,5,0],[2,2,2],
> [0,0,0],[2,2,2],[5,5,0],
> [0,0,0], [2,2,2], [0,10,0]];
> polyhedron(y, [[0,1,2], [3,4,5], [6,7,8], [9,10,11]] );

This describes a valid geometry, but not a valid topology: Vertices are duplicated and we don't know that your intent is that all geometrically equal positions in space represent the same topological vertex. OpenSCAD will reconstruct a topology using various analysis and repair tools.

The reason this distinction is important is that it's possible to describe more complex topologies, like two objects touching each other in a vertex or edge, or self-touching objects, which can remain valid manifolds when handled correctly. For more complex examples, see https://github.com/openscad/openscad/tree/master/tests/data/scad/3D/misc