openscad/openscad #1875: Preview not working after rendering or error

sanchosk, 2016-11-29:
I build the latest version of OpenSCAD to verify that this is not caused by outdated version.
I am using Ubuntu Mate 16.10 and OpenSCAD 2016.11.29.

When I create a new file and start to edit, everything works - I can preview, render, modify, preview again, etc.
But as soon as I make a typo in editor and save (which is what I am doing to refresh the preview), everything is gone. I can fix the typo and save again, but the preview will no longer show.
When re-rendering, I can see the object again, but when I modify the source again and save (to preview), nothing shows in preview.
In console I can see
`Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 0 seconds`
And that's it.
When I exit OpenSCAD and restart with the same file, everything works again OK, no problem.
I tried to google and look at the open issues, but no luck. Perhaps this is just my problem?
Any advice?

sanchosk, 2016-11-29:
Sample code:
This code will NOT show preview after a typo:
```
$fn = 24;

difference() {
    cube([10,10,10], center = true);
    sphere(r = 6.5);
}
```

But when left out the difference
```
$fn = 24;

//difference() {
    cube([10,10,10], center = true);
    sphere(r = 6.5);
//}
```
the preview is back. Of course - re-enabling difference makes the preview go away again.

t-paul, 2016-11-29:
Where is the package coming from? This sounds like #1727 that needs an updated OpenCSG (version 1.4.2).

Please paste the Help->Library Info

sanchosk, 2016-11-29:
Library info:

OpenSCAD Version: 2016.11.29
System information: Linux 4.8.0-27-generic #29-Ubuntu SMP Thu Oct 20 21:03:13 UTC 2016 x86_64 Ubuntu 16.10 2 CPUs 3.76 GB RAM
Compiler: GCC "6.2.0 20161005" 64bit
Boost version: 1_61
Eigen version: 3.2.93
CGAL version, kernels: 4.8.1, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.0
Qt version: 5.6.1
QScintilla version: 2.9.3
MingW build: No
GLib version: 2.50.0
libzip version: 1.1.2
Application Path: /usr/local/bin
Documents Path: /home/sancho/.local/share
Resource Path: /usr/local/share/openscad
User Library Path: /home/sancho/.local/share/OpenSCAD/libraries
User Config Path: /home/sancho/.config/OpenSCAD
Backup Path: /home/sancho/.local/share/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/home/sancho/.local/share/OpenSCAD/libraries
/usr/local/share/openscad/libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.13.0
OpenGL Version: 3.0 Mesa 12.0.3
GL Renderer: Mesa DRI Intel(R) Ivybridge Mobile 
GL Vendor: Intel Open Source Technology Center
RGBA(8880), depth(0), stencil(0)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QOpenGLWidget
QSurfaceFormat: RGBA(8880), depth(0), stencil(0)

GL Extensions:
GL_3DFX_texture_compression_FXT1
GL_AMD_conservative_depth
GL_AMD_draw_buffers_blend
GL_AMD_performance_monitor
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_trinary_minmax
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_vertex_array_object
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
[... 171 lines cut ...]
GL_KHR_debug
GL_KHR_robustness
GL_MESA_pack_invert
GL_MESA_texture_signed_rgba
GL_MESA_window_pos
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays

I'll google how to update OpenCSG...

MichaelAtOz, 2016-11-29:
> and save (which is what I am doing to refresh the preview),

Save doesn't do a preview??? You do know F5 does preview?

sanchosk, 2016-11-29:
Sure I do :) But hitting save or F5 makes exactly the same result - no preview visible. Looks like old CSG lib, trying to re-build.

sanchosk, 2016-11-30:
Issue is fixed by installing newest version of OpenCSG (1.4.2). Closing, thanks for your help!

closed 2016-11-30 by sanchosk

matthew-viglione, 2018-01-08:
How do you update openCSG on it's own? I am having this same issue and followed openCSG's build instructions, but it either is not installing correctly or OpenSCAD isn't recognizing the update somehow.

kintel, 2018-01-09:
@matthew-viglione You need to build from source to get the correct version, as the OpenCSG version number is header-only. If you did that and still go the old version, that means that you still have an old OpenCSG installed.

matthew-viglione, 2018-01-09:
I built both OpenCSG and OpenSCAD from source. How do I uninstall the old version to force it to use the new one?

kintel, 2018-01-09:
@matthew-viglione Depends on your OS - usually just reverse how you installed it (i.e. if you used `apt-get` to install, use `apt-get` to remove).

AndKe, 2018-12-06:
I have this issue with snap installed daily, my package manager has nothing with *csg* installed. 
@matthew-viglione   - did you get rid of the problem ?