openscad/openscad #1877: qt4 download link not working with ./scripts/uni-build-dependencies.sh qt4

romeojulietthotel, 2016-12-02:
Neither of these below worked to fetch qt4 sources but I think the second one worked for qt5 when I tried building with qt5.  FWIW it fetched qt5.3.1

I played with a few different urls but they use mirrors so the url changes based on location and I don't know what url to fetch to get redirected to a mirror.

```
http://download.qt-project.org/qt4/source/qt-everywhere-opensource-src-$version.tar.gz

http://download.qt-project.org/official_releases/qt/$v/$version/single/qt-everywhere-opensource-src-$version.tar.gz

```

donbright, 2016-12-03:
the url for qt4 source is https://download.qt.io/archive/qt/4.8/4.8.6/qt-everywhere-opensource-src-4.8.6.tar.gz

(see https://download.qt.io/archive/qt/4.8/4.8.6/)

but please note per README.md

> "Note that huge dependencies like gcc, qt, or glib2 are not included here"

in other words, building your own qt4 with uni-build-dependencies.sh is not officially supported, its something for hackers to play around with.

kintel, 2016-12-03:
If we're truly don't want to attempt building qt in uni-build-dependencies.sh, we should probably just remove it. Better having no solution than a wrong solution :)

donbright, 2016-12-03:
that is ok with me, i used to use it on gcc compile farm that lacked qt,  but not so much anymore 

not sure if anyone else uses it ( @t-paul  ? )

t-paul, 2016-12-03:
I'm not using it and I guess for the newer build stuff we want to go with Qt5 even if the remaining glitches are a bit annoying.

romeojulietthotel, 2016-12-08:
@kintel and also better than blaming the reporter for using existing features of a wrong solution.

The best thing would be to document openscad_deps. Well if openscad_deps is officially supported. And how to tell what is official and what is not.... I would submit a doc PR but don't find the doc sources. 

> If you are building Openscad in a custom environment the typical "./configuration --prefix=somedir" can be set to your openscad_deps directory.

But I don't know if openscad_deps is officially supported and so on and so forth.

kintel, 2016-12-08:
`openscad_deps` is just a folder where the build script by default places built dependencies. This is really only used if you're using an old system and need access to more recent libraries. Building on old systems is not very fun, so it's a lot better to upgrade if you want to do any development.

The doc source _is_ the wikibooks, apart from the github README

romeojulietthotel, 2016-12-08:
There are many use cases for building non-std packages or out-of-tree or static or etc., etc. So I'm glad that the feature exists. But it sounds like it is not officially supported. Maybe it can be put into a contrib or unofficial directory by itself.

Regarding docs, if I interpret what you're saying correctly, there's no way to submit PR against docs. Or I am misunderstanding.

OBTW, FWIW the openscad I built against Qt5 was not functional. The interface was garbled.

kintel, 2016-12-08:
Docs: You can edit the docs on wikibooks - it's a public wiki. Most docs are user-contributed.

Qt5: Please open a separate issue with more info about your system

Support: This is an open source project run by a very small team of volunteers. Nothing is really "officially supported", we just do our best to help people. Any help to clean up, improve or document stuff is welcome.

romeojulietthotel, 2016-12-08:
Ok, thanks, I now understand the doc model. I usually read docs locally so I will check the wikibooks.

I am not planning on using Qt5 for now. And yes I understand the support model. My reported issues are my attempt to find answers but also trying to give observations that may help the project. Of course since I have used Openscad for one week I don't know what's helpful/useful to the project and what's not. I have a little better idea now thanks to the replies.

donbright, 2017-01-25:
the "official support" (as in, the instructions in the README) for the build-dependency linux build system was based on the idea that OpenSCAD had only about ten dependencies, so it was possible to maintain it properly. this was many years ago. the effort to port Marius' Mac OSX dependency build script to linux/bsd was made and it tested good on dozens of different versions of linux/bsd. (OSX has a history of this kind of dependency-build system because Apple doesn't necessarily ship a lot of the most up to date typical open source software infrastructure libraries that most open source projects depend on. Linux is different, it is kind of unusual for a project to build its own dependencies on linux, but the use of CGAL and OpenCSG kind of made it important because so many linux distros did not used to come with those libraries in new enough versions to work well with OpenSCAD)

as years went by the extra features to OpenSCAD have added more dependencies to OpenSCAD and new permutations of build configurations to the point where in my opinion the 'build dependency' script for linux/bsd should be dropped, as it now has begun to resemble a packaging system like pkgsrc and is not maintainable very well because of the diversity of linux and bsd systems where testing is necessary to make it 'just work'.

my most recent build system experiments were to move OpenSCAD's linux "i need to build my own dependencies and im not root" build to be based on the **linuxbrew** system which allows non-root users to custom build an entire package system dependency tree on their machine. It accomplishes the same goal - you get to build your own dependencies from source - but without the OpenSCAD crew having to maintain the building of the package source tree. 

@kintel has had OSX **homebrew** working on Mac for many years.... my idea is just to do the same on Linux and get rid of a lot of problems with dependency builds on linux - effectively outsourcing the problem of building a source package manager to the Brew Community. (i also experimented with pkgsrc but it doesn't seem to have quite the level of momentum as Brew). This plan would effectively change it so the linux version of 'build dependency' script would be eliminated and be replaced by a line or two of  linuxbrew commands. 

linuxbrew is based on Mac 'homebrew' and there are even plans in the "brew community" to merge linuxbrew into homebrew so that it will 'just work' even better than it does now. Recently, linuxbrew merged Qt into their tree which is a huge step towards getting OpenSCAD to 'just work' with linuxbrew. I have built OpenSCAD myself on my own modified version of linuxbrew, and it was very pleasant, there just needs to be that last few dozen hours of polish and integration upstream for it to be Shiny. 

linuxbrew might be slightly more difficult to 'customize' if you want to build custom dependency libraries because you have to learn how they do things and modify their ruby '.rb' build scripts if you want to easily use a custom version of something. however i have found this personally preferable to mucking about with endless different config and build systems from different projects. for example 'boost' uses bjam, cgal uses cmake, gmp/mpfr use config scripts, OpenCSG uses qmake, and GLEW has it's own hard coded Makefiles. So  being able to hack on a single ruby file per package might be a nice change of pace. 

that is my two cents on the build system on linux

romeojulietthotel, 2017-01-30:
Maybe http://appimage.org/  would suffice. One thing is clear and that is linux has too many distros with different build and packaging systems. It's not a sustainable model.

probonopd, 2017-02-05:
> Maybe http://appimage.org/ would suffice. 

AppImages for testing are here:
https://bintray.com/probono/AppImages/OpenSCAD/_latestVersion#files

Check https://github.com/openscad/openscad/issues/1849

romeojulietthotel, 2017-04-16:
...time passes...

I have to build from sources so I think this issue can be closed since the direction is moving away from the build scripts.

donbright, 2017-08-09:
... time keeps passing...

yeah looks like i was completely wrong about linuxbrew.  unfortunately linuxbrew has become a gigantic mess. 

Their install instructions have become more and more complicated, confusing, user-hostile, and less and less reliable. A bunch of the packages also print out 'caveats' or 'this doesnt work' warnings when you build them... 

I just tried downloading and "brew install glew", it depends on  'systemd' which, it reports, is broken. 

That might be a surmountable problem... i have submitted countless issues to projects in the past about things that didnt build.... except that now the linuxbrew process for reporting problems has itself become massively complicated. 

They tell you to run 'brew doctor' and 'fix all the warnings', then you run 'brew doctor' and 'brew doctor'  tells you to ignore the warnings. Then if you go to their 'bug report' website they tell you that they will delete "Without comment" any issue that hasnt filled out a confusing form of checkboxes, including the aforementioned "brew doctor" warning fixes. Brew Doctor wants you to delete everthing from /usr/local/ on linux, even though linuxbrew is not supposed to touch /usr/local/ on linux.  

All this to report that something doesnt build. On a fresh install.  It is like living inside a Douglas Adams novel. 

Back to our scripts...

romeojulietthotel, 2017-08-11:
...some jiffies pass

I see a lot of talk about Flatpak. Not that I like it or use it but it may provide a solution. Although it requires root.   nix is interesting and I played with it but I don't use it.

I seem to always come back to building and installing to `/usr/local `and I do so as a non-root user and a non-user user. Just a user that builds/installs and is separate from any other users. I think some call that separation of permissions. 

I have encountered cmake a lot more recently and I still don't like it but it sucks less than autoconf because usually a mortal can figure out what went wrong. It's probably the way to go. But it requires creating the correct cmake modules, most of which already exist somewhere and just need tuning for OpenSCAD.

donbright, 2017-08-17:
I'm going to go ahead and say that Nix is the future. 

I just built openscad on ubuntu 16, ubuntu 14, ubuntu 12, and ... ubuntu 10 using nix. 

It was exactly the same shell commands on all four platforms... to be specific its about four shell commands 

    curl https://nixos.org/nix/install | sh
    source ~/.nix-profile/etc/profile.d/nix.sh
    openscad/scripts/nixshell-run.sh (this basically calls nix-shell -p opencsg cgal qt5 boost, etc etc)
    qmake && make

Let me be clear. It was easier for me to build OpenSCAD using nix than using ubuntu's own package manager (because the names of packages change over time). I was able to use exactly the same installation commands to pull down dozens of dependency packages, without interfering with the OS in any way. 

There is only one problem, and that is that OpenGL on nix is a complete mess at the moment. So you can't actually run openscad after you build it. Not even software rendering works easily. 

https://github.com/NixOS/nixpkgs/issues/9415

romeojulietthotel, 2018-11-12:
Building from source has been working for me for a long time now. (looks like nix still has not fixed OpenGL)

closed 2018-11-12 by romeojulietthotel