openscad/openscad #1879: Artifacts appearing at regular intervals.

romeojulietthotel, 2016-12-02:
I built from current git using uni-build-dependencies.sh.  The artifact/glitch first appears in column 31 (if I counted right) and then repeats at regular intervals. For example if you change the col factor from 71 to 101 you will see three artifacts.  But maybe you won't and it's some issue with the ivybridge gpu here. 

Have I done something stupid or is it an off-by-one error possibly?

    module bread_board(width, depth, thickness){        
          rowheight = depth/9;
          colheight = width/71;
          rows = floor(depth/rowheight);
          cols = floor(width/colheight);
          holeradius = 1.1/2;
          holedepth = thickness/2;
        
            mirror([0,-1,0])
            difference(){
              translate([width/2,depth/2,-thickness/2]) 
                cube([width, depth, thickness], true);
              
              //union(){
                for (i = [colheight:colheight:width-colheight]){
                    for (j = [rowheight:rowheight:depth-rowheight]){
                        translate([i,j,-thickness*1.1])
                            //cylinder(h=1+holedepth/2, r=holeradius, center=false, $fn=17);
                            #cylinder(h=thickness+.5, r=holeradius, center=false, $fn=17);
                    }
               }
              //}
            }
        
    }
    bread_board(91,29,2);

donbright, 2016-12-03:
can you dump help/library info? it looks ok on openscad version 2015.03-1

[image]

```
OpenSCAD Version: 2015.03-1
System information: Linux 4.4.0-51-generic #72-Ubuntu SMP Thu Nov 24 18:29:54 UTC 2016 x86_64 Ubuntu 16.04.1 LTS 4 CPUs 7.71 GB RAM
Compiler: GCC "5.2.1 20151028" 64bit
Boost version: 1_58
Eigen version: 3.2.5
CGAL version, kernels: 4.7, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.0
Qt version: 5.5.1
QScintilla version: 2.9.1
MingW build: No
GLib version: 2.47.1
Application Path: /usr/bin
Documents Path: /home/don/.local/share
Resource Path: /usr/share/openscad
User Library Path: /home/don/.local/share/OpenSCAD/libraries
User Config Path: /home/don/.config/OpenSCAD
Backup Path: /home/don/.local/share/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/home/don/.local/share/OpenSCAD/libraries
/usr/share/openscad/libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.13.0
OpenGL Version: 3.0 Mesa 11.2.0
GL Renderer: Mesa DRI Intel(R) Broadwell GT1 
GL Vendor: Intel Open Source Technology Center
RGBA(8880), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Using QGLWidget

```

romeojulietthotel, 2016-12-08:
I built from git master, should I file issue reports against OpenSCAD Version: 2015.03-1. The image you show here may not reveal the artifacts. i.e. they could be lost in the noise of that particular image and/or view/angle/resolution/color choice.

t-paul, 2016-12-08:
Please add a screenshot of the actual issue then. It's not going to help guessing what artifacts we are talking about.

romeojulietthotel, 2016-12-08:
I switched to opencsg 1.4.2 and I don't see the artifacts but now openscad is extremely slow. It may be that the slowness was due to having goldwing enabled. (which was suggested in other issues) I will disable that and retry but I guess this issue is resolved and I'll open (or comment on existing) an issue.

kintel, 2016-12-08:
@romeojulietthotel Please open a separate issue for the slowness so we can report that upstream to OpenCSG.

closed 2016-12-11 by t-paul