openscad/openscad #1883: Build on Windows 10 error

mrJingGuo, 2016-12-06:
When I follow https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Building_on_Microsoft_Windows,

Version: OpenSCAD 2015.03-2
System: Windows 10 x64.

```
$ qmake openscad.pro CONFIG-=debug
Project MESSAGE: If you're building a development binary, consider adding CONFIG+=experimental
Project MESSAGE: If you're building a development binary, consider adding CONFIG+=experimental
Project MESSAGE: If you're building a development binary, consider adding CONFIG+=experimental

$ mingw32-make.exe

Makefile.Release:20936: recipe for target 'objects/func.o' failed
mingw32-make[1]: *** [objects/func.o] Error 1
mingw32-make[1]: Leaving directory 'C:/Msys64/home/Jing G/openscad2'
Makefile:34: recipe for target 'release' failed
mingw32-make: *** [release] Error 2
```

anyone have any idea how to fix it?

thanks

kintel, 2016-12-06:
Looks like the error message from the compiler was swallowed. Do you have a more complete error message?

mrJingGuo, 2016-12-06:
I try to install on my another computer Windows 7,

It shows error:
```
$ qmake openscad.pro CONFIG-=debug
Package libzip was not found in the pkg-config search path.
Perhaps you should add the directory containing `libzip.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libzip' found
Package libzip was not found in the pkg-config search path.
Perhaps you should add the directory containing `libzip.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libzip' found
Package libzip was not found in the pkg-config search path.
Perhaps you should add the directory containing `libzip.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libzip' found
Package libzip was not found in the pkg-config search path.
Perhaps you should add the directory containing `libzip.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libzip' found
Package libzip was not found in the pkg-config search path.
Perhaps you should add the directory containing `libzip.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libzip' found
Package libzip was not found in the pkg-config search path.
Perhaps you should add the directory containing `libzip.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libzip' found
Project MESSAGE: If you're building a development binary, consider adding CONFIG+=     experimental
Project MESSAGE: Using QGLWidget
Project MESSAGE: Using C++11
Project MESSAGE: If you're building a development binary, consider adding CONFIG+=     experimental
Project MESSAGE: Using QGLWidget
Project MESSAGE: Using C++11
WARNING: Failure to find: objects/openscad_win32_res.o
Project MESSAGE: If you're building a development binary, consider adding CONFIG+=     experimental
Project MESSAGE: Using QGLWidget
Project MESSAGE: Using C++11
WARNING: Failure to find: objects/openscad_win32_res.o
```

actually the my Windows 10 has same error too.

kintel, 2016-12-06:
Those are just warnings - it shouldn't affect the build.

mrJingGuo, 2016-12-06:
on my windows 7, after this, I run "mingw32-make.exe"

it shows:

```
compilation terminated.
Makefile.Release:36584: recipe for target 'objects/version_check.o' failed
mingw32-make[1]: *** [objects/version_check.o] Error 1
mingw32-make[1]: Leaving directory 'C:/Msys64/home/Mixshop/openscad'
Makefile:34: recipe for target 'release' failed
mingw32-make: *** [release] Error 2
```

kintel, 2016-12-06:
It just says `recipe for target 'objects/version_check.o' failed`, but the actual error message from the compiler isn't displayed. Is there a way you can get that error message?
(I don't have a Windows computer handy)

donbright, 2017-01-25:
Thanks for the bug report

I have built many many times on Windows using the MSYS2 system.  It may not always be guaranteed to work 100% because msys2 is a very dynamic project with very new dependency versions changing often, however I have had a very very high success rate in the past. In this particular case the culprit may be the wiki instructions themselves not OpenSCAD necessarily. 

The libzip stuff is from a recent dependency/feature added to OpenSCAD that the Wiki probably did not add yet. But as @kintel said that is a warning, it is possible to build without it.  However to avoid this, you can install libzip using 'pacman' msys package manager. 

Secondly, i would avoid this **mingw32-make.exe** stuff. on my MSYS2 system i notice that ming32-make is actually an older version of make... we want newer versions, newer versions are often better for this sort of thing. 

[image]

Now the last step i might suggest is to guarantee that one is using the proper MSYS2 shell. MSYS2 comes with three options,in the windows(TM) start menu on my machine,  MSYS2-MSYS, MSYS64 and MSYS32. 

I would recommend MSYS64 if you are on 64 bit windows. 

I would also recommend building out of source tree, to make debugging problems easier. It is very simple to type 'rm -rf ./builddir' than to try to clean up the junk left during a source tree build. I do something like this

       cd openscad
       mkdir build
       cd build
       qmake .. CONFIG-=debug
       make

nophead, 2017-01-30:
I don't have a make after downloading MSYS2 but mingw32-make.exe works for me. Odd as I am using MINGW64. I successfully built in Win7 64, despite a lot of warnings.

kintel, 2017-01-30:
We should clean up warnings. This may not be Windows-specific though - we probably should do some work on a recent Linux distro with gcc 6 first.

t-paul, 2018-11-06:
Windows/MSYS2 builds are now verified on [AppVeyor](https://ci.appveyor.com/project/kintel/openscad) (currently using the MSYS2 provided there + installing additional packages, no system/package upgrades)

Details:
https://github.com/openscad/openscad/blob/e1f043d2d038a2c1839ec39354383b5de2b12cdb/appveyor.yml#L13-L22

closed 2018-11-06 by t-paul