openscad/openscad #1884: Shortcut Keys not working.

amarjeetkapoor1, 2016-12-07:
Some Shortcut keys are not working for actions like saveAs, Reload, close, Comment, uncomment, past veiwport translation, All find commands.
I observed this on ubuntu.

MichaelAtOz, 2016-12-07:
In 2016.11.11 on Windows they are working (Ctrl-S, Ctrl-R etc.)

kintel, 2016-12-07:
@amarjeetkapoor1 Since when are they not working (OpenSCAD version, Linux version) ?

t-paul, 2016-12-07:
Could be also some Ubuntu/Desktop-Environment issue, I did not try all, but the shortcuts are working for me on Debian/testing running Gnome 3 with the latest OpenSCAD nightly build (Qt4).

amarjeetkapoor1, 2016-12-08:
I found it on both Ubuntu 14 LTS and 16 LTS both  with default environment (I think its unity). 
I found this on both 2015.03-1 version and also latest build with qt5.

t-paul, 2020-02-16:
Assuming that was specific to Unity (messing with Qt menu bars) which is pretty much dead now, so closing the issue.

closed 2020-02-16 by t-paul