openscad/openscad #1886: rotate_extrude and cylinders with $fn have sides which don't align

neiljp, 2016-12-07:
I couldn't find a reference to a bug similar to this. I'm running a nightly debian build 20161201T231538.git5192388-1734.1, after suggestion to do so to resolve a dxf/svg import issue via irc.

Having both a cylinder and doing a rotate_extrude on a 2D shape, each with $fn set to finite values, leads to edges (sides) which don't align properly due to different rotational offsets, as illustrated by the following:
```
coin_diameter=40;
coin_thickness=2;
ring_height=1;
slope_inset=1.5; // non-zero values add slope inside ring (eating into text_padding)
sides=7; // or 0 for default (~round)
side_rotation=90;
width_to_fit_text=35;

rotate_extrude_extra_angle=180/sides; // Necessary to ensure sides>0 has edges which align
// =========================================================================
rotate([0,0,side_rotation])
difference()
{
  cylinder(r=coin_diameter/2,h=ring_height+coin_thickness,$fn=sides);
  translate([0,0,coin_thickness]) 
    cylinder(r=width_to_fit_text/2,h=ring_height+1,$fn=sides);
}

if (slope_inset>0)
{
  overlap=1;
  rotate([0,0,side_rotation+rotate_extrude_extra_angle])
  rotate_extrude($fn=sides)
  polygon(points=[[width_to_fit_text/2-slope_inset,coin_thickness],
                  [width_to_fit_text/2,ring_height+coin_thickness],
                  [overlap+width_to_fit_text/2,ring_height+coin_thickness], // into ring
                  [width_to_fit_text/2,coin_thickness]]);
}
```

thehans, 2016-12-07:
Yes this is a particular inconsistency in the way that rotate_extrude works.  

In early code, before rotate_extrude had an angle parameter, it was implemented such that it started drawing the points from the negative x-axis, instead of positive x.  IMO, this behavior was a mistake from the start, but has been purposely kept for legacy support (so people's older scripts don't break).  So its actually rotated a complete 180 degrees, regardless of the number of sides;  and rotate_extrude_extra_angle=180 would also have worked in your example.

However, if an angle parameter other than 360 is given, the newer angle functionality starts the first point at the more logical positive x-axis, (where angle=0 by convention for cartesian system).
So you can potentially do a kludge like:  rotate_extrude($fa=359.999/sides, angle=359.999)  
and it would appear how you want without any rotate_extrude_extra_angle

The related code is here in case you are curious:
https://github.com/openscad/openscad/blob/4a0ac7b2fa6e8ca390e3b95e1c2d2541cf9d77af/src/GeometryEvaluator.cc#L868

kintel, 2016-12-07:
We might clean up some of these things if we do a major language overhaul in the future, but changing this now is likely to break existing designs.

neiljp, 2016-12-07:
Has there been a release with the angle=* feature? If not, could angle=360 work consistently and leave the no-parameter version as backward-compatible?

If an angle not 360 uses a different rotation... well I appreciate the desire to break designs, but that's worse IMO.

Could openscad determine the version used to (last) save the file and work accordingly, or something like that?

Re the label, while I appreciate it was a design decision and so arguably not a bug as such, perhaps also adding a reason such as 'backwards compatibility' or 'language inconsistency' would be good. Do you intend to leave this open?

Thanks for the feedback. Is this behaviour documented?

kintel, 2016-12-07:
There's only been development snapshots with the `angle=` feature, so it's still possible to change it.
I have to think about this.

We currently don't have a versioning scheme for .scad files, but we're looking into ways of solving this. One way is to do a major overhaul and thus force people to upgrade their syntax (where we could automatically flip rotate_extrudes with an odd number of segments). Another way is to introduce a version header.

I didn't close yet - before closing we should take note of this somewhere.

Documentation isn't clear on this. Do you want to take a stab at clarifying it? https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/2D_to_3D_Extrusion#Rotate_Extrude

neiljp, 2016-12-07:
Options I see (as a new user):
1) leave it as currently coded, with explicit documentation of the 180 degree offset in the docs
2) have that specifying angle=* switches the default rotation axis to that used elsewhere
3) have that specifying angle=* switches to the proper axis (as above), with angle=-360 (note minus) being the default, implying the reverse angle - this also would allow use of different orientations by users? (better/explicit/extensible version of (2))
4) introduce a new command which allows partial angles (ie accepts angle=*), with new default value behaviour that fits that elsewhere; potentially deprecate old command later, converting eg. rotate_extrude() to angular_extrude(-360), and allowing use of angular_extrude() and angular_extrude(360) to match angular behaviour of cylinders etc.

For me, (3) would be bearable, though still surprising (default of -360, vs 360); (4) would be best IMO.

No issue for tracking .scad versioning? (to cross-reference to this :))

Could look at docs at some point, but would be good to find out if there's a plan too :)
The wikibook is the accepted documentation?

Ma-XX-oN, 2016-12-07:
If I'm understanding this thread correctly, this is the result of backward compatibility issue?

To maintain legacy support, why not have another parameter that is a start angle (referenced against +ve X-axis).  This would default to 180 to make it backward compatible, but could be set to 0 to make it more intuitive.

BTW, what is meant by `angle=*`.  Is this a new syntax?

kintel, 2016-12-07:
I don't think there is an issue for tracking scad file versioning.
The wikibooks docs is the reference manual.

I'm uncertain about adding sugar to specify start angle. having rotation and sweep angle orthogonal may make sense as you could e.g. do `rotate(30) rotate_extrude(angle=60)`.

I agree that a new module may be the cleanest. For completeness, we might then want to look at naming scheme of `linear_extrude()` as well.

neiljp, 2016-12-07:
I only came up with it as a simple idea, but linear_ and angular_extrude seem like good parallels to me :)

indazoo, 2018-11-04:
Hi
I just stumbled over this also. Come on guys, the last official release is 2015! Rotate_extrude not even knew the angle parameter , I think. So you only "break" scripts with full turns. So "only" small overlapping errors for uneven(!) $fn values will appear in a future major release.
On the other side, if you leave it this way, and release it, then everybody using rotate_extrude will fall into this "feature" and have to adjust his design so it looks good. This feature is especially annoying if you do parametric models with an adjustable opening from 0 to 360 degree. Then, your thingy looks different with 0  degree in contrast when the user inputs 1 degree.

rotate_extrude suffers from these problems:
1. as mentioned above the different start point in contrast to cylinder()
2. the "fix" with using angles other than 0 and 360 to use another starting point
3. the more problematic one: If you do a cylinder with $fn=13 and then use rotate_extrude() with an angle value of say 355 degree to create an overlapping/attachment to the cylinder, the facets of the polygons do not match. rotate_extrude() calculates its step angle by itself. Yes I know this is difficult to solve, because the resulting mesh of rotate_extrude() should use the angle step size of $fn as cylinder() does but then may have to draw an "uncomplete" facet to draw the last degrees.

I vote for fixing it. 
Thank you

kintel, 2024-12-22:
Closing as angle= has been released for years now, and we deem this to be a minor enough issue that we can live with it

closed 2024-12-22 by kintel