openscad/openscad #1887: UX Suggestion: change numeric values from keyboard + auto-refresh on change

iafan, 2016-12-11:
I apologize if something similar has already been discussed somewhere, but couldn't find any relevant issues.

I just started tinkering with OpenSCAD and the first thing that I noticed was that I have to do small repetitive tasks when adjusting the model:

1. Change some numeric value in the code
2. Hit F5 (or Ctrl+S) for preview to be triggered
3. Preview the result
4. Repeat

I believe there's a pretty easy way to significantly increase the model tweaking process by implementing these two features:

1. Whenever the caret is placed to the left, to the right, or inside a number in the source code editor, pressing e.g. <kbd>Alt+Up</kbd> and <kbd>Alt+Down</kbd> could increase/decrease the number by 1. Similarly, <kbd>Shift+Alt+Up</kbd> and <kbd>Shift+Alt+Down</kbd> could change the value with the step of 0.1. Ultimately, these steps could be defined in the settings.

2. Whenever the source code is changed, refresh the preview area (without pressing F5 or saving the project).

These two options coupled together could provide a very easy way to adjust the code on the fly: just go to any number, start changing it using keyboard shortcuts, and see the preview instantly updated.

t-paul, 2016-12-11:
That already exists in the dev snapshots. ([screencast](http://files.openscad.org/video/OpenSCAD-interactive-number-adjustment.webm))

closed 2016-12-11 by t-paul

t-paul, 2016-12-11:
Actually 1) is implemented. Auto-refresh for every change sounds very troublesome for any model that does not preview in fractions of a second.

iafan, 2016-12-11:
> That already exists in the dev snapshots.

Wow, that's cool. Thanks!

MichaelAtOz, 2016-12-11:
>  and see the preview instantly updated.

You'll soon find instant is not the right word...