openscad/openscad #1890: Critical error for unclosed block comment

thehans, 2016-12-12:
When this code is previewed:
`/*`

OpenSCAD pops up this alert:
[image]

The application no longer previews or renders after fixing the error and adding geometry.

Ideally this error would be notified through the console, and handled such that OpenSCAD would continue to function.

kintel, 2016-12-12:
Thx - this is a bug in the customizer comment parser.
@amarjeetkapoor1 Does this look easy to fix?

kintel, 2016-12-12:
@thehans FYI: This bug only triggers if the customizer feature is turned on.

amarjeetkapoor1, 2016-12-12:
I have corrected this by adding more condition checking but I have written comment.cpp with assumption that whole file will be syntacticly correct and It only runs only if parser doesn't produce error and `/*` is something which  should be syntacticly wrong. I think for `/*` parser should produce error message stating "unterminated comment "  but It doesn't produce I any error.  Similar is case for `" ` although their is no crash but it should produce error.

amarjeetkapoor1, 2017-01-23:
I think that remaining two issues can be solved by adding following rule to 
`<cond_string>` and `<cond_lcomment>` states:

```<<EOF>>  { BEGIN(INITIAL); return TOK_ERROR;  }```

But I am not getting the expected result. Any comments?

kintel, 2017-04-14:
@amarjeetkapoor1 Can you take a look at #1992 ?

amarjeetkapoor1, 2017-04-14:
Yes, sure

commit c3ab3069 (2017-04-22): Merge pull request #1992 from openscad/issue1890

Handle unclosed start conditions. Fixes #1890
closed 2017-04-22 by kintel