openscad/openscad #1892: text() - Fontconfig needs to update its cache - always

MichaelAtOz, 2016-12-12:
As I mentioned in the [forum](http://forum.openscad.org/text-Fontconfig-needs-to-update-its-cache-td19536.html), I can now confirm with 2016.11.11 the font cache update happens every time (when text()is used).
This was when no other versions were being used.
Clean start OpenSCAD, F5 a text() model, cache updates.
Exit.
Repeat, cache updates.

romeojulietthotel, 2017-01-30:
Saw this behavior as well while looking into #1896.

hackwin, 2017-04-10:
Deleting the C:\Users\[User]\AppData\Local\fontconfig folder fixed it for me

MichaelAtOz, 2017-04-10:
@hackwin Thank you! Fixed it for me too, that was reallly annoying...

romeojulietthotel, 2017-04-16:
Doesn't address linux.

hackwin, 2017-04-16:
@romeojulietthotel did you try searching for the folder and deleting it in Linux?

MichaelAtOz, 2017-04-17:
It has reoccurred.  Clearing font cache fixed it again.

huyinjie, 2017-04-29:
@MichaelAtOz  Me too

matthew-viglione, 2018-04-26:
Still an issue for me on Linux. Built it all myself on 2018.03.06. Every time text() is used, I get the small popup window. It doesn't even render correctly on Linux. Using the fc tools to clear and rebuild the caches manually through the command line doesn't fix it either. 

[image]

t-paul, 2018-08-29:
There's multiple locations as shown by running `fc-cache -v`, on my Debian it's
```
/var/cache/fontconfig
$HOME/.cache/fontconfig
$HOME/.fontconfig
```

nophead, 2018-08-29:
Thanks for that hint. This is what I got running that command.
```
fc-cache -v
C:/Windows/fonts: caching, new cache contents: 1060 fonts, 0 dirs
C:/msys64/home/ChrisP/.local/share/fonts: skipping, no such directory
C:/msys64/home/ChrisP/.fonts: skipping, no such directory
C:/msys64/mingw64/var/cache/fontconfig: cleaning cache directory
C:/msys64/mingw64/var/cache/fontconfig: invalid cache file: 06b29fc5927124ed8095d9ade59dec79-x86_64.cache-7
C:/msys64/mingw64/var/cache/fontconfig: invalid cache file: 23674ee8fd663d674d81b0a76a8d3411-x86_64.cache-7
C:/msys64/mingw64/var/cache/fontconfig: invalid cache file: 385be20327f148f5876fe4d690f90d01-x86_64.cache-7
C:/msys64/mingw64/var/cache/fontconfig: invalid cache file: 38a0c0c31618b1035335ded521eb7f95-x86_64.cache-7
C:/msys64/mingw64/var/cache/fontconfig: invalid cache file: 3f9c584c5e3c46e1064a2ef6d23315ec-x86_64.cache-7
C:/msys64/mingw64/var/cache/fontconfig: invalid cache file: 8b15b82c4135ec402f2ae99699618564-x86_64.cache-7
C:/msys64/mingw64/var/cache/fontconfig: invalid cache file: 91e442875a2d2b285e417303198d000f-x86_64.cache-7
C:/msys64/mingw64/var/cache/fontconfig: invalid cache file: 96de4d57a4ee995f532dddf45802e580-x86_64.cache-7
C:/msys64/mingw64/var/cache/fontconfig: invalid cache file: 990a638998760a0b73ba384971c2aa4f-x86_64.cache-7
C:/msys64/mingw64/var/cache/fontconfig: invalid cache file: c83b8ce5c72fa6d7cc3a30e63e84b1f6-x86_64.cache-7
C:/msys64/mingw64/var/cache/fontconfig: invalid cache file: d3b02eb10c2d8a2c5a7e3d721a53d071-x86_64.cache-7
C:/msys64/mingw64/var/cache/fontconfig: invalid cache file: e4b856cd46d6586e0907f738fe806f5f-x86_64.cache-7
C:/msys64/mingw64/var/cache/fontconfig: invalid cache file: e720f842f913ec01e41a7a9d8ed03665-x86_64.cache-7
C:/msys64/mingw64/var/cache/fontconfig: invalid cache file: ed3842811c49576ccb0b1bd6600b63e1-x86_64.cache-7
C:/msys64/mingw64/var/cache/fontconfig: invalid cache file: fe6213507baf27a0fc0dfb1e7d91b505-x86_64.cache-7
C:/msys64/mingw64/var/cache/fontconfig: invalid cache file: ff3399f32cbd7b26a186c1e786fd6d24-x86_64.cache-7
C:/msys64/home/ChrisP/.cache/fontconfig: not cleaning non-existent cache directory
C:/msys64/home/ChrisP/.fontconfig: not cleaning non-existent cache directory
C:\msys64\mingw64\bin\fc-cache.exe: failed
C:/Windows/fonts: failed to write cache
```

So I deleted everything in ```C:/msys64/mingw64/var/cache/fontconfig``` and now it says:
```
C:/Windows/fonts: skipping, existing cache is valid: 1060 fonts, 0 dirs
C:/msys64/home/ChrisP/.local/share/fonts: skipping, no such directory
C:/msys64/home/ChrisP/.fonts: skipping, no such directory
C:/msys64/mingw64/var/cache/fontconfig: cleaning cache directory
C:/msys64/home/ChrisP/.cache/fontconfig: not cleaning non-existent cache directory
C:/msys64/home/ChrisP/.fontconfig: not cleaning non-existent cache directory
C:\msys64\mingw64\bin\fc-cache.exe: succeeded
```

Now I can run all 64 tests involving text in 44 seconds whereas each one was taking about that time before. After running the tests fc-cache gives this.
```
$ fc-cache -v
C:/Windows/fonts: skipping, existing cache is valid: 1060 fonts, 0 dirs
C:/msys64/home/ChrisP/.local/share/fonts: skipping, no such directory
C:/msys64/home/ChrisP/.fonts: skipping, no such directory
C:/msys64/mingw64/var/cache/fontconfig: cleaning cache directory
C:/msys64/mingw64/var/cache/fontconfig: invalid cache file: 38a0c0c31618b1035335ded521eb7f95-x86_64.cache-7
C:/msys64/mingw64/var/cache/fontconfig: invalid cache file: ff3399f32cbd7b26a186c1e786fd6d24-x86_64.cache-7
C:/msys64/home/ChrisP/.cache/fontconfig: not cleaning non-existent cache directory
C:/msys64/home/ChrisP/.fontconfig: not cleaning non-existent cache directory
C:\msys64\mingw64\bin\fc-cache.exe: succeeded
```
Running a second time:-
```
$ fc-cache -v
C:/Windows/fonts: skipping, existing cache is valid: 1060 fonts, 0 dirs
C:/msys64/home/ChrisP/.local/share/fonts: skipping, no such directory
C:/msys64/home/ChrisP/.fonts: skipping, no such directory
C:/msys64/mingw64/var/cache/fontconfig: cleaning cache directory
C:/msys64/home/ChrisP/.cache/fontconfig: not cleaning non-existent cache directory
C:/msys64/home/ChrisP/.fontconfig: not cleaning non-existent cache directory
C:\msys64\mingw64\bin\fc-cache.exe: succeeded
```

I don't know why invalid files appear or why it failed to fix them before I cleaned it manually.

MichaelAtOz, 2018-11-10:
This, the rebuild, no longer happens, see version below.
_BUT the small window/dialog-box does pop-up and instantly disappear._

As I don't recall any fixes for the rebuild, as @nophead suggests, perhaps it was fixed in fontconfig.
Unfortunately it isn't listed below.
Has a new library version been included in the build?

```
OpenSCAD Version: 2018.11.04.ci339 (git 29742ad7)
System information: Windows(TM) 6.1 SP 1.0 NTW 1 MSDN 724833 6 CPUs 8.00 GB RAM
Compiler: GCC "7.3.0" 64bit
Boost version: 1_60
Eigen version: 3.2.5
CGAL version, kernels: 4.11, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.11.2
QScintilla version: 2.10.4
InputDrivers: 
MingW build: MingW64
GLib version: 2.50.2
libzip version: 1.5.1
lib3mf version: 1.8.0
Application Path: C:/Program Files/openscad-2018.11.04.ci339
Documents Path: C:\Users\MeB\Documents
Resource Path: C:/Program Files/openscad-2018.11.04.ci339
User Library Path: C:/Users/MeB/Documents/OpenSCAD/libraries
User Config Path: C:\Users\MeB\AppData\Local/OpenSCAD
Backup Path: C:/Users/MeB/Documents/OpenSCAD/backups
OPENSCADPATH: C:\Users\MeB\Documents\3D-REPRAP\OpenSCAD\OpenSCAD Libraries;C:\Users;C:\test;C:\test2
OpenSCAD library path:
C:\Users\MeB\Documents\3D-REPRAP\OpenSCAD\OpenSCAD Libraries
C:\Users
C:\test
C:\test2
C:/Users/MeB/Documents/OpenSCAD/libraries
C:/Program Files/openscad-2018.11.04.ci339\libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.12.0
OpenGL Version: 1.1.0
GL Renderer: GDI Generic
GL Vendor: Microsoft Corporation
RGBA(8888), depth(16), stencil(8)
GL_ARB_framebuffer_object: no
GL_EXT_framebuffer_object: no
GL_EXT_packed_depth_stencil: no

Qt graphics widget: QGLWidgetGL Extensions:GL_EXT_bgra
GL_EXT_paletted_texture
GL_WIN_swap_hint
```

t-paul, 2018-11-10:
Adding version info via #2569.

Yes, fontconfig was updated pretty regularly
* 2018-09-07 - 2.13.1
* 2018-06-17 - 2.13.0
* 2017-09-22 - 2.12.6
* 2017-07-09 - 2.12.4
* 2017-06-14 - 2.12.3

MichaelAtOz, 2026-02-22:
For the record, this stopped happening quite a while ago. (4me)