openscad/openscad #1895: Bad syntax still works

MichaelAtOz, 2016-12-21:
```
%cube(10);
translate([5,5,,,,5]) // any number of extra commas
  #cube(10);
```
It actually uses 5 for z.
I would have thought it would have looked after the second comma for the z value, and either used zero, or got an error, not keep going along the parameter list.

t-paul, 2016-12-21:
Any reference to the "still"? Was this supposed to be changed somewhere in the past? The parser has lots of places where it explicitly ignores multiple commas.

MichaelAtOz, 2016-12-21:
> where it explicitly ignores multiple commas

Wasn't aware of that. This was the first I came across it.

closed 2016-12-21 by MichaelAtOz

t-paul, 2016-12-21:
I'm not sure it was intended to cover this case though, it's common enough to ignore trailing commas. Maybe the definition for that got a bit out of hand. I suspect it's hard to change now, although it might be possible to separate wanted and unwanted cases and warn for the unwanted ones.