openscad/openscad #1896: hangs on startup

romeojulietthotel, 2016-12-23:
Using git master.  On startup sometimes it hangs and sometimes it starts fine.  On the times when it starts fine it will hang when I open another scad file.  It hangs either when choosing new or loading a recent file. Seems to be no pattern with the files.

What you're looking at below is openscad starting and I load a file and there's no problem. Then I open another file and the app hangs for 20 or 30 seconds. Then I quit the app normally.

```
strace -c > /dev/null openscad

% time     seconds  usecs/call     calls    errors syscall
------        ----------- ----------- --------- --------- ----------------
 20.72    0.026667           6      4384       389      stat
 13.42    0.017266           2      8986      6775    recvmsg
 10.19    0.013112           3      5195           poll
  6.75    0.008682           3      3298           clock_gettime
  6.67    0.008579           4      2152           writev
  6.42    0.008265          17       498         1 ioctl
  5.62    0.007230           4      1795       233 open
  5.62    0.007229         145        50           munmap
  5.46    0.007024           3      2430        10 lstat
  3.51    0.004520           2      2451           fcntl
  2.87    0.003690           2      1565           close
  2.43    0.003131           3      1199           write
  2.05    0.002632           3       913        47 read
  1.88    0.002414           2      1489           fstat
  1.83    0.002357           2      1253           fstatfs
  1.04    0.001340           4       352           brk
  0.82    0.001055           4       280           mmap
  0.76    0.000973          18        53           ftruncate
  0.51    0.000654           4       157           mprotect
  0.39    0.000508          85         6           shmdt
  0.28    0.000365           3       141           gettimeofday
  0.17    0.000222           4        53        11 access
  0.14    0.000175           9        20           getdents
  0.09    0.000114           2        57           lseek
  0.05    0.000061           6        11         2 recvfrom
  0.04    0.000048           8         6           readlink
  0.03    0.000041          41         1           unlink
  0.03    0.000038           6         6           shmat
  0.03    0.000037           2        17           fadvise64
  0.03    0.000036           6         6           shmget
  0.02    0.000025           2        16           getcwd
  0.02    0.000022           2        12           shmctl
  0.02    0.000021           3         8           futex
  0.01    0.000017           3         6           uname
  0.01    0.000013           7         2           chdir
  0.01    0.000013           2         6           getrlimit
  0.01    0.000013           7         2           statfs
  0.01    0.000012           2         6           getuid
  0.01    0.000009           2         6           rt_sigaction
  0.01    0.000008           8         1           connect
  0.01    0.000008           2         4           geteuid
  0.01    0.000007           7         1           socket
  0.01    0.000007           7         1           sysinfo
  0.00    0.000006           2         4           getgid
  0.00    0.000005           5         1           eventfd2
  0.00    0.000005           5         1           pipe2
  0.00    0.000004           2         2           getegid
  0.00    0.000003           3         1           shutdown
  0.00    0.000003           3         1           getpeername
  0.00    0.000003           3         1           arch_prctl
  0.00    0.000003           3         1           clock_getres
  0.00    0.000002           2         1           rt_sigprocmask
  0.00    0.000002           2         1           set_tid_address
  0.00    0.000001           1         1           set_robust_list
  0.00    0.000000           0         1           execve
------ ----------- ----------- --------- --------- ----------------
100.00    0.128677                 38911      7468 total
```

Some other observations....

Also I notice that there are about 3,000 open/stat/lstat of OpenSCAD.conf and I also see that it tries to open font cache files that do not exist.  For example if I have my own fonts in then it will try to open a system-wide font cache file using the hashed value filename for one of my cached font names.

```
strace openscad 2>&1|grep OpenSCAD.conf|wc -l

strace -c > /dev/null openscad

```

I don't know if these observations are relevant to the hangs or not.

t-paul, 2016-12-23:
I've never had that issue, neither with the nightly builds nor with self-build version on my Debian/testing system.

I don't think that information is enough to do much about it. Please add at least what exact system (distribution / desktop environment) we are talking about and the output of the Help->Library Info to see what lib versions are used.

The delay might be fontconfig rebuilding the font cache (that should happen only with models actually using `text()`).

kintel, 2016-12-23:
@romeojulietthotel If you built this yourself, could you try to break it in a debugger while it hangs, and share the stack traces here?

romeojulietthotel, 2016-12-27:
I have found, and I am using, a work-around. When openscad hangs I just flip to a vt terminal and back to the desktop and things are working.  It's annoying, but it works.

I don't know much about fontconfig but rebuilding the cache does not make sense unless you have added fonts. But I might be wrong about that. Why would openscad rebuild the font cache? But that doesn't seem like the problem anyway.

I don't think I built with -g but I can try to get a stack trace.

```
Compiler: GCC "5.4.0" 64bit
Boost version: 1_56
Eigen version: 3.2.7
CGAL version, kernels: 4.7, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 4.8.6
```

MichaelAtOz, 2016-12-27:
> rebuilding the cache does not make sense unless you have added fonts. 

That's a bug. Reported #1892

romeojulietthotel, 2016-12-27:
6775 errors  out of 8986 calls to recvmsg. That seems high but perhaps it's meaningless.
Also should I file a separate issue about the 1000's of calls to OpenSCAD.conf?

```

%time     seconds  usecs/call     calls    errors   syscall
------       ----------- -----------     ---------  ---------  ----------------
 20.72    0.026667           6      4384       389      stat
 13.42    0.017266           2      8986      6775    recvmsg
```

kintel, 2016-12-28:
@romeojulietthotel OpenSCAD.conf is managed by QSettings (https://doc.qt.io/qt-5/qsettings.html). Not sure what the errors could be, but it might be worth looking into it.

t-paul, 2016-12-28:
I don't think those `recvmsg` have anything to do with `QSettings`. Every X11 program shows that, it's quite likely just reading the X11 AF_UNIX socket (e.g. /tmp/.X11-unix/X0).

I'm not sure chasing issues based just on a summary output of `strace` will lead to much. Unlocking after switching console also might point to some distribution / desktop-environment / window manager issue (and we still don't know what we are talking about).

The huge amount of open calls to QSettings is probably caused by the sloppy way of using it as local instance everywhere. That could use some improvement, so a separate issue for that might make sense.

romeojulietthotel, 2017-01-19:
@t-paul 
> I don't think those recvmsg have anything to do with QSettings.
No they don't but they may have something to do with this issue. And it's not reading anything most of the time because it's failing. 6775 errors from 8986 attempts.

I haven't had time to do any custom builds or testing. I need to at least test since there have been X11 updates. Sometimes these strange issues just get fixed via updates. I will log an issue about the calls to OpenSCAD.conf.

donbright, 2017-01-25:
just for poops and giggles can you dump your library info here? (help/library info)  the whole issue with the entire  GUI locking up might be a driver issue or something and if we can search on text for the drivers it might help future us find a pattern with other people with similar lockup issues (hello future us!)

romeojulietthotel, 2017-01-30:
Library info above already.

romeojulietthotel, 2017-04-16:
...time flows...

I will try to test this week.  Not sure what happens when switching from a Desktop/X11 env out to the virtual terminal and back, a.k.a. VT switching, but I have noticed other strange things that I think are graphics related (not OpenSCAD). Just making the point that whatever occurs during VT switching is significant in some way. Maybe it's a clue but I have no idea where to look.

romeojulietthotel, 2018-11-12:
...has it really been that long...

I don't see any hangs on startup as I initially reported. I would like to add a corollary to the adage (which is untrue) that "all bugs are shallow with enough eyes".
 `All bugs solve themselves with enough passing time.`

closed 2018-11-12 by romeojulietthotel