openscad/openscad #1898: uni-build-dependencies.sh failed, "Invalid configuration 'neon'"

GrapeJamMan, 2016-12-27:
Trying to build a copy of current openscad for raspbian and my output for check-dependencies looks pretty good (I think):

depname     minimum     found       OKness      
qt          4.4         4.8.6       OK          
qscintilla2 2.7         unknown     NotOK       
cgal        3.6         4.5         OK          
gmp         5.0         6.0.0       OK          
mpfr        3.0         3.1.2       OK          
boost       1.35        1.55        OK          
opencsg     1.3.2       1.3.2       OK          
glew        1.5.4       1.7.0       OK          
eigen       3.0         3.2.2       OK          
glib2       2.0         2.42.1      OK          
fontconfig  2.10        2.11.       OK          
freetype2   2.4         18.0.12     OK          
harfbuzz    0.9.19      unknown     NotOK       
bison       2.4         3.0.2       OK          
flex        2.5.35      2.5.39      OK          
make        3           4.0         OK          
Warning: you have pkgconfig under /usr/local/lib
Please verify these local copies don't conflict with the system

Looks like just qscintilla2 and harfbuzz are needed. I ran setenv-unibuild.sh per the instructions and then uni-build-dependencies.sh and I got a nasty surprise:

Building gmp 6.0.0 ...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 2282k  100 2282k    0     0  70165      0  0:00:33  0:00:33 --:--:--  105k
checking build system type... Invalid configuration `neon': machine `neon' not recognized
configure: error: /bin/bash ../config.sub neon
neon
neon
neon-unknown-linux-gnueabihf failed

Anyone seen this before? eigen is ok in the output but I don't know what gmp 6.0.0 is since it's not listed as a dependency. And "neon" doesn't sound like anything familiar, uname -a is this:

Linux raspberrypi 4.4.38-v7+ #938 SMP Thu Dec 15 15:22:21 GMT 2016 armv7l GNU/Linux

Is there anything I can do?

kintel, 2016-12-27:
I don't think uni-build-dependencies will help you. This has not been tested on RPi and I don't think it supports building scintilla under Qt4 either.
FYI: gmp is a dependency of CGAL. The script you're running is trying to build everything.

I'd recommend trying to find the missing packages in a repository or building it manually.

t-paul, 2016-12-27:
As raspbian/jessie comes with OpenSCAD, it already should have all the dependencies available. So there's a good chance all dev packages are available too.
I do see dev packages for both harfbuzz and qscintilla in the [raspbian repo](http://archive.raspbian.org/raspbian/pool/main/).

GrapeJamMan, 2016-12-29:
Ah, looks like I was using the wrong package names to install manually. Today I learned about "apt-cache search", thanks for the hints! I definitely liked the extra stuff of the developer snapshot on Windows so I didn't want to leave that behind dumping my loud desktop moving in to RPi.

donbright, 2017-01-25:
just curious if you tried the 'get dependencies' script before attempting 'build dependencies? in an ideal world it would have just gotten the deps you needed.

cwuebker2, 2019-11-17:
In the ~3 years since this ticket was filed, Raspberry Pi's Raspbian OS, based on Debian, has moved to Buster & no longer comes with OpenSCAD.  My OS is:

  Linux raspberrypi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux

  Ideally, OpenSCAD would provide an apt-type package with code compiled for the Raspberry Pi's ARM processor in addition to the packages available for Linux, the Mac & Windows, so that's my primary request.

  Until then, I've been trying to build OpenSCAD on my Raspberry Pi 2B and many of the
source files compile.  However, when trying to compile **src/LibraryInfo.cc** with qt5,
I see a number of what I think are #include file conflicts which causes the compile to
error out with messages similar to this one:
In file included from /usr/include/arm-linux-gnueabihf/qt5/QtGui/qopengl.h:105,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtWidgets/qopenglwidget.h:49,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtWidgets/QOpenGLWidget:1,
                 from src/QGLView.h:7,
                 from objects/ui_MainWindow.h:34,
                 from src/MainWindow.h:6,
                 from src/input/InputDriverManager.h:34,
                 from src/LibraryInfo.cc:6:
/usr/include/GLES3/gl32.h:75:25: error: conflicting declaration ‘typedef khronos_ssize_t GLsizeiptr’
 typedef khronos_ssize_t GLsizeiptr;

and other messages similar to this one:
In file included from /usr/include/arm-linux-gnueabihf/qt5/QtGui/qopengl.h:105,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtWidgets/qopenglwidget.h:49,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtWidgets/QOpenGLWidget:1,
                 from src/QGLView.h:7,
                 from objects/ui_MainWindow.h:34,
                 from src/MainWindow.h:6,
                 from src/input/InputDriverManager.h:34,
                 from src/LibraryInfo.cc:6:
/usr/include/GLES3/gl32.h:528:99: error: ‘void __glewBindAttribLocation(GLuint, GLuint, const GLchar*)’ redeclared as different kind of symbol
 GL_APICALL void GL_APIENTRY glBindAttribLocation (GLuint program, GLuint index, const GLchar *name);

  As far as I can tell, my dependencies exist & are up to date.  I'm including a copy of the script I'm using to try to cleanly compile OpenSCAD -- any suggestions would be appreciated:

-- buildOpenSCAD script
#!/bin/bash

if [ -d ~/openscad_deps ]; then
  echo openscad_deps exists
  exit 1
fi

{
set -x

cd ~/openscad

source scripts/setenv-unibuild.sh
make clean

git pull origin master
git submodule update --init

source scripts/setenv-unibuild.sh
qmake -qt=qt5 openscad.pro
make
} 2>&1 | tee ~/buildOpenSCAD.log

t-paul, 2019-11-17:
Raspbian comes with GLES instead of Desktop OpenGL. There's no simple way to solve that. There might be an AppImage at some point, but the huge amount of time needed to get everything working means I don't know if that ever will be released.

t-paul, 2019-11-18:
Closing as there's no official support for `armv7l`.

closed 2019-11-18 by t-paul