openscad/openscad #1899: Preview incorrect when CSG tree >= 255 elements and black background

thehans, 2016-12-27:
This issue only appears when using the "Starnight" theme, which has a pure black background.
```
difference() {
  cube(20,center=true);
  for(j = [0:253])
    cube(20);
}
```
console says "Normalized CSG tree has 255 elements" in this example.  Changing the loop to anything less makes the error go away.

[image]

thehans, 2017-01-03:
If I recall correctly, the opencsg library uses the stencil buffer for some parts of its calculations, which I suspect is somehow causing this issue.

On the computer that has this issue, I know that the stencil buffer is 8bits.  

I tried to reproduce this on another computer of mine, and it did not occur, I suspect it has wider stencil buffer.  I will attempt to verify this later, when I have access to the other computer.

The Library info for the offending computer is here:
```
OpenSCAD Version: 2016.12.26.nightly (git 1fd9f05)
System information: Linux 3.19.0-32-generic #37~14.04.1-Ubuntu SMP Thu Oct 22 09:41:40 UTC 2015 x86_64 Ubuntu 14.04.5 LTS 8 CPUs 11.67 GB RAM
Compiler: GCC "4.8.2" 64bit
Boost version: 1_54
Eigen version: 3.2.1
CGAL version, kernels: 4.4, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.3.2
Qt version: 4.8.6
QScintilla version: 2.8.1
MingW build: No
GLib version: 2.40.0
libzip version: 
Application Path: /usr/bin
Documents Path: /home/peeps/.local/share
Resource Path: /usr/share/openscad-nightly
User Library Path: /home/peeps/.local/share/OpenSCAD/libraries
User Config Path: /home/peeps/.config/OpenSCAD
Backup Path: /home/peeps/.local/share/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/home/peeps/.local/share/OpenSCAD/libraries
/usr/share/openscad-nightly/libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.10.0
OpenGL Version: 3.3.0 NVIDIA 340.98
GL Renderer: Quadro FX 880M/PCIe/SSE2
GL Vendor: NVIDIA Corporation
RGBA(8880), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QGLWidgetGL Extensions:
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_clear_buffer_object
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
[... 198 lines cut ...]
GL_NV_texture_expand_normal
GL_NV_texture_multisample
GL_NV_texture_rectangle
GL_NV_texture_shader
GL_NV_texture_shader2
GL_NV_texture_shader3
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_vdpau_interop
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_video_capture
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
```

kintel, 2017-01-04:
I can reproduce it as well.

Based on my understanding of OpenCSG, this is likely a stencil buffer overflow (more than 255 subtractions in a subtraction sequence, each of which increment the stencil buffer). I thought OpenCSG had added a fix to allow more than 255 negative objects, but we might have run into a limitation.

thehans, 2017-01-04:
Isue does not occur on my windows machine, with the folliwing info:
```
OpenSCAD Version: 2016.04.06
System information: Windows 7 SP1 (v6.1.1.0) 8 CPUs 15.94 GB RAM
Compiler: GCC "5.2.0" 64bit
Boost version: 1_57
Eigen version: 3.2.5
CGAL version, kernels: 4.5, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.0
Qt version: 5.5.1
QScintilla version: 2.8.4
MingW build: MingW64
GLib version: 2.44.1
Application Path: C:/Program Files/OpenSCAD
Documents Path: C:\Users\hans\Documents
Resource Path: C:/Program Files/OpenSCAD
User Library Path: C:/Users/hans/Documents/OpenSCAD/libraries
User Config Path: C:\Users\hans\AppData\Local/OpenSCAD
Backup Path: C:/Users/hans/Documents/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
C:/Users/hans/Documents/OpenSCAD/libraries
C:/Program Files/OpenSCAD\libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.12.0
OpenGL Version: 4.5.0 NVIDIA 368.39
GL Renderer: GeForce GTX 760/PCIe/SSE2
GL Vendor: NVIDIA Corporation
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Using QGLWidget

GL Extensions:
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
[... 268 lines cut ...]
GL_NV_texture_rectangle
GL_NV_texture_shader
GL_NV_texture_shader2
GL_NV_texture_shader3
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
```

kintel, 2017-01-20:
@thehans Looking at this more closely: Could this simply be a bug in OpenCSG 1.3.x? I believe OpenCSG 1.4 added support for optimizing parts of their algorithm using a shader instead of the fixed function pipeline.

thehans, 2017-01-20:
I'm pretty sure I already had 1.4 installed, but I went ahead and installed the latest 1.4.2 from source, and I'm still getting the error.  I'm 99% sure I removed any instances of 1.3.x libs on my hard drive, but the library info dialog in openscad still shows 1.3.2

I vaguely recall this OpenCSG version reporting was a problem that I discussed with you a couple years ago, don't remember the exact reason for it.  OpenSCAD just lists the OpenCSG version it was compiled against, or something like that?  It lacks a mechanism to obtain the version directly from the .so ?

kintel, 2017-01-20:
I don't think OpenCSG supports run-time version queries.
Just changing the binary may also not be sufficient due to potential header implementations and API changes.

Concretely, I believe this is related to the new `SCSChannelManagerFragmentProgram` class, introduced in OpenCSG 1.4.0, which enable more than 255 primitives. See  https://github.com/openscad/OpenCSG/blob/4463fb128f462b42e98bbea94bca54d55d50e547/src/renderSCS.cpp#L139

donbright, 2017-01-25:
it is true. OpenCSG version comes from OpenCSG.h which is baked into a .o file at compile time. if it were me i would run 'locate opencsg' or 'find /usr -name opencsg.h' and then wipe my entire openscad build tree and rebuild entirely with proper opencsg.

kintel, 2017-02-14:
@thehans Did you have a chance to rebuild using the new OpenCSG version and test again?

thehans, 2017-10-14:
I'm actually still confused about how to specify what version of OpenCSG that OpenSCAD builds against.

I've been using the cmake build lately FWIW

kintel, 2017-10-15:
You don't generally specify the version, it just uses the version found on your system. It will look at a few env. variables for hints where to search (`OPENCSGDIR`, `OPENSCAD_LIBRARIES`)
See https://github.com/openscad/openscad/blob/master/cmake/Modules/FindOpenCSG.cmake

thehans, 2017-10-18:
OK, I made sure that I was built against OpenCSG 1.4.2 and its still giivng me the same issue.

Pressing F5 to preview initially looks ok, but as soon as I move the camera, the same negative volume appears as before.

kintel, 2017-10-19:
Odd. Perhaps "Starnight" doesn't have a black background, but no background at all, and this is somehow a failure to clear the screen or smth.?