openscad/openscad #1900: Feature request - When Editor Pane hidden add Export STL Icon to preview icon list

MichaelAtOz, 2016-12-29:
Or just add it regardless.
When working with an external editor I miss the convenience of the export icon, on the now hidden editor pane.

kintel, 2017-01-20:
The preview icon list is a bit long. Perhaps a good opportunity to look into compressing it a bit.
..perhaps by using some sort of drop-down icon selector on the view angle icons? Not sure how annoying that is to implement in Qt..

MichaelAtOz, 2017-05-30:
BTW, I had a narrow window and noticed when the Icon bar is cut short, presumably Qt, makes an icon drop-down menu. Perhaps there is a wraparound option - ie get two lines of icons.
[image]
(couldn't easily capture with the menu showing)

So maybe more icons are OK?
I think the order will be important tho.

Sharma-Hrishabh, 2020-04-01:
Tried to do this. As it is a GUI update, feedback will be very helpful :)

t-paul, 2020-04-05:
I would prefer having the toolbars user customizable instead of dynamic, but I guess that may still take a while. So let's have this solution now and see how it goes :smile:.

@Sharma-Hrishabh you should be able to claim the bounty on bounty source now (see link in the issue description). Although it seems their site has some trouble at the moment.

PR #3265 (2020-04-05): Add Export STL option to Viewer-Toolbar when Editor Pane is hidden 
Fixes #1900
closed 2020-04-05 by t-paul