openscad/openscad #1901: Feature request: Render in background without blocking preview

msrd0, 2016-12-29:
Modern computers are able to do multiple jobs at the same time. As with some complex structures, rendering is not the fastest and can consume some time if the cache is empty. But while rendering, I am not able to do anything else in openscad, not even if I have two files open at the same time (I need to start a second instance of openscad). How about moving the rendering in the background and at least enable previews of other objects while rendering?

If this is not possible, another idea would be to simply start another instance when pressing ctrl+n or ctrl+o. This wouldn't enable preview and render of the same object at the same time but at least enable preview of other objects.

t-paul, 2016-12-29:
Duplicate of #1195.

closed 2016-12-29 by t-paul

t-paul, 2016-12-29:
Right now preview is running in the main thread. This basically blocks the whole application including GUI actions in other windows. I think #1195 is needed to decouple the long running actions so the additional global [GUI lock](https://github.com/openscad/openscad/blob/master/src/mainwin.cc#L134) can be relaxed or even better removed.

msrd0, 2016-12-29:
I see that #1195 is an issue and that it is somehow related to this one but in my case GUI thread is in theory idle (and GUI is responsive during rendering), but preview (which should be running on GUI thread) is not working. To me this seems to be another issue.

t-paul, 2016-12-29:
As just said, there is an additional GUI lock set when starting either rendering or preview, so while the GUI thread is technically idle with F6 render running, it will still skip preview due to [that lock](https://github.com/openscad/openscad/blob/master/src/mainwin.cc#L1852). Decoupling both F5 preview and F6 render should also solve this issue.

msrd0, 2016-12-29:
Ok thank you for clarifying

nophead, 2016-12-29:
Any chance the GUI can be made responsive in Windows during F6, or is that
another issue? It does respond to an event eventually but I think it is
only as the progress bar moves. I certainly can't use the editor.

t-paul, 2016-12-29:
As @MichaelAtOz mentioned in #1195, F6 rendering should work in background on Windows too. If that does not work it's either some more special bug or settings/environment issue. Without developers working on Windows it's going to be hard to track down and fix.

MichaelAtOz, 2016-12-29:
>  It does respond to an event eventually but I think it is only as the progress bar moves. I certainly can't use the editor.

Unless it got to the >3GB .exe process size & starts swapping, I can easily edit while F6 is running.

nophead, 2016-12-29:
> @nophead <https://github.com/nophead> can you share your config (PC) and
> OpenSCAD cache sizes.

Actually I was mistaken. It is during F5 with renders() that the GUI is
unresponsive.

version 2016.10.04 (git 7e0935d) but it has always been the same behaviour
for as long as I can remember.

Win 7 64bit but it was the same on XP 32bit.  Current machine is a Core
I7-3520M @ 2.9Ghz with 8 GB and it certainly isn't swapping. It doesn't
behave like the GUI has its own thread. It only processes mouse and
keyboard each time the progress bar moves.

kintel, 2016-12-29:
FYI: Even if we fix the preview GUI block, we use parts of CGAL which aren't threadsafe. There are rumors that this will be improved in the next version (CGAL 4.10) though..