openscad/openscad #1904: ERROR: CGAL error in CGAL_Nef_polyhedron3()

boelle, 2017-01-03:
I get this error when i try to compile

ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /usr/include/CGAL/Nef_S2/SM_const_decorator.h Line: 326 

script:

rotate([0,0,0])import("Untitled.stl", convexity=3);
rotate([0,0,0])translate([16,423,-232])cube([10,70,2]);

this one does not give an error:

rotate([0,0,0])import("Untitled.stl", convexity=3);
//rotate([0,0,0])translate([16,423,-232])cube([10,70,2]);

the stl has been run through netfabb

t-paul, 2017-01-03:
Netfabb does not fix zero faces by default anymore. Please check the [FAQ entry](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_is_my_imported_STL_file_only_showing_up_with_F5_but_not_F6.3F) for more options.

boelle, 2017-01-03:
"which are currently not handled well in OpenSCAD."

so basicly i'm screwed at the moment unless i'm willing to fiddle with yet another software package?

boelle, 2017-01-03:
blender is out as it will only show things

meshlab seems the only bet until now

should i let this one stay open so stl import can be "fixed" to better handle this?

t-paul, 2017-01-03:
If it's indeed the known zero-faces issue, we have #1864.

boelle, 2017-01-03:
i'm trying the suggestion to use meshlab

is there a 100% sure way to know if its zero face?

i'm not sure if i can share the stl but will ask

t-paul, 2017-01-03:
Blender is usually nice for finding issues using the 3d-printing plugin (needs to be enabled).

boelle, 2017-01-03:
yeah, did the steps in meshlab and that did not help it

will try blender

boelle, 2017-01-03:
i can see the plugin but i cant enable it, the check box is greyed out

need to figure why and how to enable

nophead, 2017-01-03:
NetFabb will remove them. You just have to tell it to.

boelle, 2017-01-03:
also the free version?

boelle, 2017-01-03:
btw i just used https://service.netfabb.com/

boelle, 2017-01-03:
hmm... seems that there is no version for Debian8

am i right or blind?

nophead, 2017-01-03:
Yes the free version I have does it. It is Windows only now.

boelle, 2017-01-03:
bummer

but should meshlab not have done the trick also?

nophead, 2017-01-03:
I don't know, never used it.

The problem is OpenSCAD can only import a small subset of valid STL files,
not even all the ones it made by itself. It can't cope with vertices close
together because it snaps them to a grid and makes triangles degenerate
that didn't have zero area in the file. In Netfabb I can remove degenerate
triangles with a tolerance and I have to increase that from the default to
get a file OpenSCAD will accept.

boelle, 2017-01-03:
opened it in netfabb on my partners pc

everything is green, tried the auto repair and its still the same

are there any good sites like pastebin i could use to share it at

repaired was the same size as the org

boelle, 2017-01-03:
http://www.filedropper.com/1xboxdoornetfabbmm

its from 3d hubs hence the netfabb in the filename, has the customers approval  to share

kintel, 2017-01-03:
@boelle Carsten wrote a tool for detecting and repairing such issues. It's still proprietary, but he's been helpful with testing people's files. See here: http://forum.openscad.org/bug-with-difference-tp19439p19481.html

boelle, 2017-01-03:
will try

the problem with the STL i have is that the text is to deep and there are not enough material to close it off

i print it with the text down

will have a look at link

boelle, 2017-01-03:
tried to post, dont want to gatecrash him

nophead, 2017-01-03:
According to NetFabb it has self intersections in the first S. The free version doesn't fix those but the online version does. I just tried it and it seems that since the last time I used it a few months ago they changed it to produce a 3mf file, which is useless to me.

boelle, 2017-01-03:
that is strange... 

i started out with the online version and still have the issue

time to tripplecheck

nophead, 2017-01-03:
Did you get an STL back or a 3mf?

boelle, 2017-01-03:
stl

https://service.netfabb.com/

boelle, 2017-01-03:
https://tools3d.azurewebsites.net/

gives a 3mf, but that is Microsoft and not Netfabb

boelle, 2017-01-03:
seems that if i just print in 0.3 layers and not 0.2 the text is barely covered

nophead, 2017-01-03:
Yes there is only 0.55mm above the text.

boelle, 2017-01-03:
it will have to make do

its for a 3d scanner based on arduino micro

closing this one as it seems to be a dublicate

kintel, 2017-01-20:
Closing as duplicate - see #1864

closed 2017-01-20 by kintel