openscad/openscad #1905: Add macOS Touch Bar Support

Yona-Appletree, 2017-01-03:
I recently switched to a new Macbook Pro with Touch Bar, and it makes using OpenSCAD a bit of a pain. I would love to see the addition of basic Touch Bar support  -- just Render, Refresh, Reset View would be awesome.

I may try and do the work myself, would folks be open to a pull request for that support?

kintel, 2017-01-03:
This sounds a bit like something which eventually should fit into Qt.
As that will likely take a while, a pull request is probably the way to go.
Admittedly, I haven't looked into what the touch bar really can do, but I would look into ways of making it work well with Qt, e.g. rendering Qt components into the touch bar and using the same action system as the rest of the OpenSCAD GUI.

t-paul, 2017-01-03:
Attaching external stuff to the actions should be easy. Some non-GUI based code is in the InputDriver branch. Maybe in combination with an action editor? That would be nice to have even without the TouchBar suppot :-).

donbright, 2017-01-25:
if you buy me a macbook pro i will write a patch.

donbright, 2017-01-25:
for self reference... workaround maybe alt-d e alt-d r 

but imo really what we need is to provide a good alternative to Function Keys because they are going extinct on personal computers. 

every laptop system that i see, the Function Keys get smaller and more remote. On my machine you have to hit some tiny 'fn' button then you can hit the function key. Normally it is for volume control, brightness, etc. 

so it is needed to pick new keyboard shortcuts for f5 and f6. it is a very tricky thing to get right.

it is necessary to pick some combo which wont conflict with a bunch of other stuff on the system. for example on ubuntu there is all sorts of weird stuff they use for their interface. 

also it is necessary to keep in mind different human languages may have different shortcuts for menu items based on the letters in the words used for a menu item in that language. 

it is also necessary to remember some people are used to editors like vi, vim, nano, wordstar, bank street writer, wordperfect, pages, google docs, even ms word, with various shortcuts they are used to using on different OSes and so you dont want to accidentally have f6 replaced by something they might accidentally hit out of habit (or that... you know... if you ever wanted to provide customizable editor keys like a lot of programs do).

kintel, 2017-01-25:
I think I agree that a better solution would be to use non-function shortcuts for the actions which today use function keys, and let users assign their own additional shortcuts. The same probably goes for toolbar actions.

deckar01, 2020-07-20:
I tested out https://github.com/KDAB/KDMacTouchBar. The existing icons scale poorly and there aren't any styling options yet, but the basic buttons and labels are functional.

<details><summary>Demo</summary>

[image]

</details>

I am working on cleaning this up and sharing it. I had to manually build and copy the KDMacTouchBar framework to the app contents, and I'm not sure how this is supposed to be done. Would it need to be a prerequisite installed in the system library? `make installing it didn't put it in a place where OpenSCAD was looking at runtime.

t-paul, 2020-07-20:
The `make install` target is pretty much Linux only. The MacOS App Bundle is created via `scripts/release-common.sh` which in turn uses `macdeployqt`.

deckar01, 2020-07-21:
Sorry for the confusion, I meant KDMacTouchBar's `make install` doesn't put it in a place that OpenSCAD looks for frameworks. I though it was putting it in `/Library/Frameworks`, but it wasn't. OpenSCAD did find it when I moved it there.

I see the `OBJECTIVE_SOURCES`, but I'm not sure how I would setup an ObjectiveC framework build and get the output to target the app content's framework directory. All of the other frameworks seem to be setup as global prerequisites, so maybe that's how I should leave it for now.

kintel, 2022-01-02:
I'll close this, as Apple has (finally) discontinued the touch bar, so I feel it's probably not worth spending resource chasing this any longer.

closed 2022-01-02 by kintel