openscad/openscad #1906: Make the object vs operator module types more explicit in the language.

zapta, 2017-01-04:
Currently a module can be of two types, an object or an operator. The module type determine it's usage pattern and the kind of arguments it accepts. While this is semantically well define, it hides the conceptual difference between the two and make the concept of 'children' an hard to understand and a somewhat obscure aspect of the language. 

This proposal is for making the distinction between the two more explicit in the language. It will make the code more readable (e.g. by indicating that the module is an operator that expect children) and make the concept of user defined operators more discoverable.

Here is one possible syntax:

> object my_object(...) {
>    // references to children() at al are not allowed here. 
> }
> 
> my_object(...);  // allowed.
> my_object(...) { ... };   // not allowed

> operator my_operator(...) {
>    // references to children() at al are not allowed here. 
> }
> my_operator(...) {... };  // allowed
> my_operator(...);          // not allowed?

The existing common 'module' syntax can stay for backward compatibility.

MichaelAtOz, 2017-01-04:
-1. Why limit flexibility.
I can think of a module() with an if() that may use the children() or ignore them and do something else..

module checkIt(something,thing) 
  if (something && $children!=0)
    do_something() children();
  else
    import( str(thing,".stl") );

t-paul, 2017-01-04:
I'd suggest to discuss this a https://github.com/doug-moen/openscad2.

nophead, 2017-01-04:
I have modules that do both. For example my washer module draws a washer
but if it has a child it translates the child to be on top of it. That can
be, for example, another washer or a screw or a nut.

OpenSCAD could warn if you give a child and there are no children
statements in the module.

zapta, 2017-01-04:
>>  have modules that do both. For example my washer module draws a washer
but if it has a child it translates the child to be on top of it. That can
be, for example, another washer or a screw or a nut.

You can achieve that with:

operator my_washer() {
  ..
}

...

my_washer() { /* empty */ }.
my_washer();
my_washer() cube();
my_washer() my_other_washer();
my_washer() { ... };

As it is now, the signature of your module doesn't indicate that your
module accepts children.

nophead, 2017-01-04:
> issuecomment-270307174
> auth/AAijhaOY1jGesdmzY9Fc_

>As it is now, the signature of your module doesn't indicate that your
module accepts children.

No but I don't see marking it as operator helps because nobody would know
what a washer operator does without reading the code or looking from a
comment. Also it is optional, so OpenSCAD can't give a warning if it
missing. It already gives a warning if you attempt to use child that is not
provided. The only other error is providing a child when one is not
expected. That could also give an error without needing a change to the
language.

zapta, 2017-01-04:
> issuecomment-270316973

The primary goal of my proposal is to make the concept of operators more
discoverable, i.e. in the cheatsheet, and more visible in the syntax. The
concept is already here, hence the two separate explanations of object
module and operator module, but it has virtually no visibility in the
language itself. This is of a subjective call of course.

doug-moen, 2017-01-05:
> openscad2
> issuecomment-270422188
> auth/AAijhWiUr8QCvWFka99-

Zapta said

> The primary goal of my proposal is to make the concept of operators more
> discoverable, i.e. in the cheatsheet, and more visible in the syntax.

...

As it is now, the signature of your module doesn't indicate that your
> module accepts children.

In the openscad2 proposal, an "object" is defined like this:

my_cube(size) = cube(size);

and an "operator" is defined like this:

my_translate(vector) shapes = translate(vector) shapes;

So it achieves the same goal of indicating in the signature whether a
module accepts children as arguments.

zapta, 2017-01-05:
> issuecomment-270426164

>> my_translate(vector) shapes = translate(vector) shapes;
>>
>> So it achieves the same goal of indicating in the signature whether a
>> module accepts children as arguments.

That's very good.

BTW, what is the status and expected timeline of openscad2?

(may be useful to add this information to the README/Overview files).

doug-moen, 2017-01-05:
> auth/ACFWQXLA1lHImUAKe_
> issuecomment-270432192

I'm not currently working on OpenSCAD2. I got far enough along to figure
out that the backward compatibility scheme described in the paper doesn't
work. OpenSCAD is full of weird undocumented bugs (the scoping rules are a
particular sore spot for me), and these bugs cannot be fixed, because we
don't want old scripts to break. So it is effectively impossible for an
outside party to create an independent and fully compatible implementation
of OpenSCAD, using modern implementation techniques, which is what I was
trying to do. At least, not without a lot more resources to pour into the
project. So my project went in a different direction. Now I've got a
working subset of OpenSCAD2, and I'm building a new geometry kernel from
scratch that is focussed on different goals than OpenSCAD: curved surfaces
are represented exactly, it's possible to model extremely complex models
with a lot of detail and render them quickly, and there is continuously
varying volumetric colour. So that's what's consuming my free time.

In order to reboot the OpenSCAD2 project, what's needed is a way to
incrementally evolve the existing code base, adding new features from the
OpenSCAD2 design without breaking backward compatibility. I worked out some
preliminary ideas about this, but they need more development. Meanwhile,
other people are welcome to cherry pick the OpenSCAD2 design and implement
whatever bits they like in OpenSCAD.

doug-moen, 2017-01-05:
Chris said: "I have modules that do both. For example my washer module
draws a washer but if it has a child it translates the child to be on top
of it."

The original Mendel90 code is:

module washer(type) {
    ...
    if($children)
        translate([0, 0, thickness])
            child();
}

In OpenSCAD2, module declarations continue to work without change. But
functions can now take shapes as arguments. So you can also implement
"washer" as a function. Here's a first attempt, which just transliterates
the module code:

washer(type) children = {
    ...
    if (len(children) > 0)
        translate([0,0,thickness])
            children[0];
};

The problem with this definition is that the `children` argument is not
optional. So you'd be writing `washer(t){}` a lot.

So probably you'd implement `washer` like this instead, if you wanted to
use a function:

washer(type, child=undef) = {
    ...
    if (child != undef)
        translate([0,0,thickness])
            child;
};

zapta, 2017-01-05:
>> washer(type, child=undef) = {
>> ...
>>  if (child != undef)
>>
>>  };

How will this module be called? Passing children as an argument within the
() or with {} following it?

The later form using {} is useful and natural for operators.  Forcing the
former one using () may end up with deep parenthesis nesting levels.

doug-moen, 2017-01-05:
>> washer1(type) child = {
>>     ...
>> };
>>
>> washer2(type, child=undef) = {
>>     ...
>> };

> How will this module be called? Passing children as an argument within the
> () or with {} following it?

In OpenSCAD2, these are actually function definitions. Functions can take
shapes as arguments and return shapes as results.

How a function is called depends on how the parameters are declared.

In the first case, washer1 is called as
    washer1(M8_penny_washer) {};
or
    washer1(M8_penny_washer) screw(M8_cap_screw, 30);

In the second case, washer2 is called as
    washer2(M8_penny_washer);
or
    washer2(M8_penny_washer, screw(M8_cap_screw, 30));

Technically, washer1 is a *curried* function with two argument lists
applied in succession. It can also be called like this:
   washer1(M8_penny_washer)({});
   washer1(M8_penny_washer)(screw(M8_cap_screw, 30));

jordanbrown0, 2025-09-01:
As discussed, there is no bright-line distinction between the two - you can have a module that operates on its children but does not generate a shape of its own, a module that generates a shape and does not expect children, or a module that both generates s a shape of its own *and* operate on children.  And you can have modules that do none of the above, that purely echo or assert, or that only conditionally generate (non-empty) shapes.  The two-way distinction is a useful introductory mental model, but it is not correct in general.

If you want your object modules to be "safe", so that you can't accidentally give them children, there's already a mechanism for that:
```
assert($children == 0);
```

closed 2025-09-01 by jordanbrown0