openscad/openscad #1907: Plot mathematical functions

LYRIKpage, 2017-01-05:
In addition to the graphical primitives available in OpenSCAD, it would be very helpful to be able to plot a sine wave, parabola or any other mathematical function. I hacked up some OpenSCAD code that will more or less accomplish what I mean, but doing it in native code would certainly render faster, since my code requires calculations to be repeated several times due to variable behavior.

[plot.scad](https://github.com/openscad/openscad/files/688168/plot.txt)

doug-moen, 2017-01-05:
It's a cool application. However, I think it's the sort of thing that ought
to be written in OpenSCAD, not built in. This is well within the range of
applications that OpenSCAD is designed to support.

Instead of rewriting the application in C++, put that effort into speeding
up the OpenSCAD code. I got a > 3x speedup in plot3d just by replacing the
'sphere' calls with 'cube'. More speedups are possible.

Instead of unioning together a large number of triangles, each constructed
by building a hull around 3 spheres, you could build a triangular mesh as a
data structure and render it using one call to polyhedron(). That would be
much faster. And more work to code, probably, but it's a better use of your
time than recoding in C++, and I'm sure the performance will be okay.

LYRIKpage, 2017-01-06:
I would rather see this included in OpenSCAD, since many users might need this and not everyone will have the ability or ambition to code it themselves.
Whether it should be coded in C++ or included as a built-in scad library, I don't really care too much.

As to the speedup using 'cube' instead of 'sphere', I'm afraid there is a pretty big trade-off concerning the uniform thickness of the plane. In any case, I don't think the 'thin' parameter will be used much, since, in most cases, you will want a solid body.

t-paul, 2017-01-06:
For a `polyhedron()` based example see: https://github.com/openscad/list-comprehension-demos#3d-functionscad

I agree with @doug-moen that it's too specialized to go into the C++ core. There may be some general features that could go there to make generating polyhedrons easier.

@kintel what about creating some empty repo as preparation for a standard library to collect this kind of requests? I think we have a couple of topics that would fit there but those are a bit buried in the issue list here.

LYRIKpage, 2017-01-08:
Many thanks to @t-paul for the polyhedron approach (I wouldn't have thought of that kind of for loops). It speeds up rendering by a factor of OMG.
I have also rewritten two-dimensional plot to create a single polygon.
Unfortunately, I cannot find the motivation to rewrite the "thin" part of 3D plots as well - it was hard enough for 2D.
Here is what I have so far.

[plot2.scad](https://github.com/openscad/openscad/files/692044/plot2.txt)

I think it would be cool to have a standard library containing useful modules - and of course having these documented in the cheat sheet as well.

kintel, 2017-01-10:
@t-paul The scad-utils repo was meant as a starting point for this, but it needs to be way more thought out, or we'll end up building another MCAD.. (https://github.com/openscad/scad-utils)

t-paul, 2019-11-21:
Closing the issue here as function plotting is out of scope for the core application.

closed 2019-11-21 by t-paul