openscad/openscad #1909: Bug in Rotating cube?

kenarab, 2017-01-06:
With ´OpenSCAD version 2015.03-3´

If I run this code:

```
$fs=0.25;$fa=6;
difference(){sphere(r=13);
#rotate([90,90,0])cube([26.26,8,5.6],center=true);
#rotate([90,90,0])cylinder(h=26.26,r=4,center=true);
}
```

rotation on cube and cylinder doesn't work the same way.

By my tests, it seems that X rotation on cube wrongly applies to Z dimension. It could be possible?

t-paul, 2017-01-06:
I can't see what would be wrong, please add images of what's generated vs. what's expected.

thehans, 2017-01-06:
rotation in multiple axes is applied one axis at a time, in x,y,z order so 
```rotate([x,y,z]) ... ```
is the same as 
```rotate([0,0,z]) rotate([0,y,0]) rotate([x,0,0]) ...```

Your example behaves as expected.

kenarab, 2017-01-06:
As I apply the exactly same rotation, both the cube and cylinder should be transformed in the same way... And it doesn't work that way. I have been used cylinder primitive and it works ok. But today I tried with cube... And it doesn't works ok.

[image]

kenarab, 2017-01-06:
I'm using the prepackaged binary. Compiling from source for checking if the (seemly) bug isn't fixed yet

thehans, 2017-01-06:
What's wrong with the picture again?  You expect the cylinder and cube to be in the same position?
Have you seen what it looks like before you apply the rotation?  They aren't in the same place.  Cylinder is length is along the z axis initially.  Your cube is long along the x axis initially...

kenarab, 2017-01-06:
ups. 
Yep, you are right. It's was a little bit confusing. 
Thank you very much.
I'm closing the issue.

closed 2017-01-06 by kenarab

kenarab, 2017-01-07:
I'been keep thinking... The definition of different primitives shoudn't be as consistent as possible for avoiding this kind of issues? 

My solution is add rotate[0,90,0] to cube primitive. Is simple, but not as fancy as should be a so fancy method for 3D modelling as openscad.

Thank you for your help, Guys.