openscad/openscad #1913: New feature: Export .scad file with libraries included

tinkerology, 2017-01-10:
It would be nice if I could export my .scad file with libraries included so there was just one .scad file. Optionally, libraries included with the Thingiverse customizer could be excluded.  This would ideally work recursively. This export option would also be available from the command line.

This would allow me to export a model which uses a non-standard set of libraries so that I can upload it to thingiverse easily.

Thanks, Scott

kintel, 2017-01-10:
This should probably be discussed together with library/dependency management and namespace support.

Note two things:
* This is not possible today as we don't have a way of emulating the `use` keyword due to the lack of namespace functionality.
* The libraries included in Thingiverse are in no way considered a "standard libraries", and Thingiverse is not transparent wrt. which libraries and which versions they include.

Off the top of my head, I think the best way of doing this today would be to recursively package all dependencies to a local folder and upload this folder to Thingiverse. This would effectively bypass the built-in libraries on Thingiverse and you'd be isolated from any future changes on their end.

MichaelAtOz, 2017-01-10:
> I think the best way of doing this today would be to recursively package all dependencies to a local folder and upload this folder to Thingiverse

I suspect the OP was so the .scad file can the be tagged 'Customizer' and run via it. Having files in a folder won't work, unless Customizer is 'customised'.

polkovnikov-ph, 2017-01-10:
@kintel Dependency management would be not that hard to implement. There is `npm` which is agnostic to what is stored into it.

tinkerology, 2017-01-10:
Right, I would like all the files concatenated together in the proper order
so that one file can be uploaded.  This would also make it easier to submit
bug reports and requests for help.

Thanks, Scott

t-paul, 2017-01-10:
This was also discussed in a thread on Thingiverse some time ago. I don't think there is a solution that can work in general (in addition to the topics @kintel mentioned, there's also no simple way to handle all the import style modules like existing `surface()`, `import()`, font `use()` and proposed `trace()`, I may be missing some...). We can't fix Thingiverse and any half-way solution will increase the maintenance effort on our side.

I'd suggest to close this here, or re-target to solve the distribution problem for our version of the customizer (which then could be picked up by Thingiverse if someone cares).

kintel, 2017-01-20:
One thought: We're working on a component which isolate parser+AST+traversal, without any other dependencies, potentially into a separate lib. The above request sounds a bit like something that could be implemented as a stand-alone tool on top of such a library. ..or at least as a relatively clean AST transform.