openscad/openscad #1915: Polyhedron and degenerated faces

RonaldoCMP, 2017-01-15:
Consider the following code of a prism degenerated in a tetrahedron:

 ```
v = [ [0,0,0],  [50,0,0], [0,50,0],
       [0,0,50], [0,0,50], [0,0,50] ];

 f1 = [ [0,1,2],    // base
        [1,0,3,4],  // three degenerated quads - triangles
        [2,1,4,5],
        [0,2,5,3],
        [3,4,5] ];  // triangle degenerated to a point

 f2 = [ [0,1,2],           // base
        [0,4,1], [0,3,4],  // one degenerated triangle
        [1,5,2], [1,4,5],  // one degenerated triangle
        [2,3,0], [2,5,3],  // one degenerated triangle
        [3,4,5]  ];        // triangle degenerated in a point

 // Polyhedron P1
 polyhedron(v,f1);

 // Polyhedron P2
 translate([60,0,0]) polyhedron(v,f2);

 translate([0,70,0]) cube(10);

```
Polyhedron P1 has a triangular face ([3,4,5]) degenerated to a point and 3 quads degenerated in triangles. P2 is a triangulation of P1: it has the same triangular face degenerated to a point and 3 other triangular faces degenerated to line segments. 

The surprising result is: P2 is previewed and rendered correctly even with the presence of the cube and can be boolean operated; but P1 preview and render has 3 missing faces when it is alone. However, P1 render() is as expected when P2 or  the cube are not commented out. In short: degenerated triangles are allowed in polyhedron, degenerated quads may be suppressed.

What is even more striking: if we insert one additional vertex in each edge of the tetrahedron base and change P1 accordingly, the quad faces turns to degenerated pentagons and the model is previewed and rendered correctly even alone!!!
```
v3 = [ [0,0,0],  [50,0,0],  [0,50,0],
       [0,0,50], [0,0,50],  [0,0,50], 
       [25,0,0], [25,25,0], [0,25,0] // three additional vertices inside the base edges
     ];

f3 = [ [0,6,1,7,2,8],  // base
       [1,6,0,3,4],    // 3 degenerated pentagons -> triangle
       [2,7,1,4,5],    
       [0,8,2,5,3], 
       [3,4,5] ];      // degenerated triangle -> point

// Polyhedron P3
polyhedron(v3,f3);
```
I have got the same results in versions  2015.03-2 and 2016.11.11 in a Windows 7 environment.

kintel, 2017-01-20:
This looks like a duplicate of #1867, which was fixed 2016.11.17.

@t-paul Any chance of kicking that Windows box to see if a new binary pops out? :)

donbright, 2017-01-25:
@kintel and @t-paul i am right now upgrading the romilly machine to Dual CPU with 3TB/month transfer, and so it should be OK to get a nightly windows cross-build going.... something as simple as a crontab. we just need somewhere to dump the built files and not worry about filling up someone's disk or trashing their bandwidth limit.

donbright, 2017-01-25:
ok i set up a new user devoted 100% to nightly windows builds. crontab is setup, will see what happens. 
delivery to public will be another issue perhaps to discuss offline

kintel, 2017-01-29:
@RonaldoCMP Could you give the recent dev snapshots a spin?http://www.openscad.org/downloads.html#snapshots

RonaldoCMP, 2017-01-29:
I am sorry, Kintel, but I don't have the necessary libraries and softwares
nor experience to build the snapshot.

2017-01-29 17:32 GMT-02:00 Marius Kintel <notifications@github.com>:

> @RonaldoCMP <https://github.com/RonaldoCMP> Could you give the recent dev
> snapshots a spin?http://www.openscad.org/downloads.html#snapshots
>

kintel, 2017-01-29:
@RonaldoCMP The snapshot is a precompiled binary, no external libs or software needed.

RonaldoCMP, 2017-01-30:
I had downloaded the wrong file.
I have tested the snapshot and it seems that the bug has gone. Nice job. Thank you.
The issue may be closed as solved.

closed 2017-01-30 by kintel