openscad/openscad #1917: On startup openscad does too many open/stat/lstat on OpenSCAD.conf

romeojulietthotel, 2017-01-19:
I notice that there are about 3,000 open/stat/lstat of OpenSCAD.conf on startup.

strace openscad 2>&1|grep OpenSCAD.conf|wc -l

kintel, 2017-01-19:
@romeojulietthotel Two questions:
1) Which version of OpenSCAD and which version of Qt is this?
2) Could you try to run this on another Qt-based app and see how the conf file access pattern is different?

romeojulietthotel, 2017-01-19:
OpenSCAD built from git master about 1 month ago.

Compiler: GCC "5.4.0" 64bit
Boost version: 1_56
Eigen version: 3.2.7
CGAL version, kernels: 4.7, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 4.8.6

I don't know that I have any other Qt-based apps but I will check.

donbright, 2017-01-25:
confirm ubuntu 16.04 openscad 2015.03-1

don@serebryanya:~/src/openscad/bin2$ strace openscad 2>&1|grep OpenSCAD.conf|wc -l
2786

action - open 'logo.scad' file
close

result: 2786 hits lol

1416 are like this
```

stat("/etc/xdg/OpenSCAD/OpenSCAD.conf", 0x7fffe4cfb500) = -1 ENOENT (No such file or directory)
stat("/etc/xdg/OpenSCAD.conf", 0x7fffe4cfb530) = -1 ENOENT (No such file or directory)
stat("/etc/xdg/OpenSCAD.conf", 0x7fffe4cfb530) = -1 ENOENT (No such file or directory)
stat("/etc/xdg/OpenSCAD.conf", 0x7fffe4cfb500) = -1 ENOENT (No such file or directory)

```

a bunch more is like this

```
access("/home/don/.config/OpenSCAD/OpenSCAD.conf", W_OK) = 0
lstat("/home/don/.config/OpenSCAD/OpenSCAD.conf", {st_mode=S_IFREG|0664, st_size
=1315, ...}) = 0
open("/home/don/.config/OpenSCAD/OpenSCAD.conf.T24303", O_RDWR|O_CREAT|O_EXCL|O_
CLOEXEC, 0666) = 22
stat("/home/don/.config/OpenSCAD/OpenSCAD.conf", {st_mode=S_IFREG|0664, st_size=
1315, ...}) = 0
access("/home/don/.config/OpenSCAD/OpenSCAD.conf", R_OK) = 0
access("/home/don/.config/OpenSCAD/OpenSCAD.conf", W_OK) = 0
access("/home/don/.config/OpenSCAD/OpenSCAD.conf", X_OK) = -1 EACCES (Permission denied)
chmod("/home/don/.config/OpenSCAD/OpenSCAD.conf.T24303", 0664) = 0
rename("/home/don/.config/OpenSCAD/OpenSCAD.conf.T24303", "/home/don/.config/OpenSCAD/OpenSCAD.conf") = 0
stat("/home/don/.config/OpenSCAD/OpenSCAD.conf", {st_mode=S_IFREG|0664, st_size=1315, ...}) = 0
unlink("/home/don/.config/OpenSCAD/OpenSCAD.conf.lock") = 0
stat("/home/don/.config/OpenSCAD.conf", 0x7fffe4cfb190) = -1 ENOENT (No such file or directory)
stat("/home/don/.config/OpenSCAD.conf", 0x7fffe4cfb190) = -1 ENOENT (No such file or directory)

```

romeojulietthotel, 2017-01-30:
I found another Qt app but it starts from a shell script and so I can't yet get a clean strace. But it looks like it has been confirmed by @donbright .

kintel, 2017-01-30:
@romeojulietthotel What I'm curious about is if this is a general issue with Qt apps or if it's specific to OpenSCAD.

romeojulietthotel, 2017-01-30:
Right, but at least we know that my openscad setup is not the cause. Will still try to get a strace from the other Qt app.

romeojulietthotel, 2017-01-31:
The other Qt app shows no such behavior.

DrCodeWizzard, 2017-02-13:
Quick update. I believe this behavior is caused by the QSettings class (based on the fact that OpenSCAD.conf contains configuration values that are being read by QSettings). There appear to be 61 references to QSettings through out the project. A bunch of them are creating an instance of the QSettings on the stack. I'll keep digging.

DrCodeWizzard, 2017-02-13:
And here's an excerpt from QT documentation showing that the locations mentioned in this ticket are ones used by QSettings.

> Platform-Specific Notes
> 
> Locations Where Application Settings Are Stored
> 
> As mentioned in the Fallback Mechanism section, QSettings stores settings for an application in up to four locations, depending on whether the settings are user-specific or system-wide and whether the settings are application-specific or organization-wide. For simplicity, we're assuming the organization is called MySoft and the application is called Star Runner.
> 
> On Unix systems, if the file format is NativeFormat, the following files are used by default:
> 
> $HOME/.config/MySoft/Star Runner.conf (Qt for Embedded Linux: $HOME/Settings/MySoft/Star Runner.conf)
> $HOME/.config/MySoft.conf (Qt for Embedded Linux: $HOME/Settings/MySoft.conf)
> /etc/xdg/MySoft/Star Runner.conf
> /etc/xdg/MySoft.conf

http://doc.qt.io/qt-5/qsettings.html

kintel, 2017-02-13:
It's definitely related to QSettings. The questions is if we're using QSettings wrong, is QSettings accessing files too often, or is this normal behavior and the freeze issues are unrelated?

DrCodeWizzard, 2017-02-13:
One question I have about QSettings. Throughout the code base I see comments saying QSettings configured correctly in main.cc ... Since that file does not exist, where is QSettings being set for the first time? Also what file is the entry point for OpenSCAD? Is it OpenSCADApp.cc?

kintel, 2017-02-13:
See https://github.com/openscad/openscad/blob/master/src/openscad.cc#L783

DrCodeWizzard, 2017-02-13:
Pulled QT source code. Each value retrieval from QSettings results in a file traversal. So what we are observing is correct behavior given that we use qsettings in many places. To work around this we would need to cache the qsettings state internally in openscad, and only write to qsettings when app is closing and read from qsettings when app is starting.

nophead, 2017-02-13:
Writes can probably afford to be immediate as you don't want settings to revert when OpenSCAD crashes.

DrCodeWizzard, 2017-05-10:
So I finally had time to start looking into this issue in more detail (yay for online courses). I implemented a cache for QSettings simple values (i.e. set with setValue only). Here are the results of running the code on my machine:

cache miss and cache access refer to just read operations on settings (settings.value() retrievals).
cache miss and cache accesses also contain the expense of preloading the whole config file into std::map the first time we access any settings.

konrad@T530:~/Sources/openscad$ strace ./openscad 2>&1|grep "Cache miss"|wc -l
63
konrad@T530:~/Sources/openscad$ strace ./openscad 2>&1|grep "cache access"|wc -l
91
konrad@T530:~/Sources/openscad$ strace ./openscad 2>&1|grep OpenSCAD.conf|wc -l
3841

As a reference point without the cache I get the following number of entries for opening the conf file:
3911

I also installed ubuntu version of openscad on my machine:

That one generates 2731 entries on my machine.

So it looks like just caching the values in a map is not really going to decrease the number of times we traverse the file system in search of config files.

DrCodeWizzard, 2017-05-10:
Ok did some more digging and discovered another way of accomplishing the same thing that only results in 91 entries. Will clean up my code and post for review probably tomorrow.

kintel, 2017-05-14:
Fixed by #2030

closed 2017-05-14 by kintel

donbright, 2017-08-09:
fyi this broke the nightly windows cross build, details at https://github.com/openscad/openscad/pull/2030