openscad/openscad #1919: Windows 10 msys2 build failure due to libzip AMF zip importer code

donbright, 2017-01-25:
While exploring  #1883 i found the build fails on Windows 10 MSYS2 with a libzip error

[image]
 
The problem appears to be related tо how amf_import uses boost to get a filename for libzip's zip_fopen and conflict between wchar_t and char_t for 'value_type' of c_str() on boost::fs::path filenames

 http://www.boost.org/doc/libs/1_61_0/boost/filesystem/path.hpp

```

# ifdef BOOST_WINDOWS_API
    typedef wchar_t                        value_type;
    BOOST_STATIC_CONSTEXPR value_type      preferred_separator = L'\\';
# else 
    typedef char                           value_type;
    BOOST_STATIC_CONSTEXPR value_type      preferred_separator = '/';
# endif

```

later on

```
   //  -----  observers  -----
  
    //  For operating systems that format file paths differently than directory
    //  paths, return values from observers are formatted as file names unless there
    //  is a trailing separator, in which case returns are formatted as directory
    //  paths. POSIX and Windows make no such distinction.

    //  Implementations are permitted to return const values or const references.

    //  The string or path returned by an observer are specified as being formatted
    //  as "native" or "generic".
    //
    //  For POSIX, these are all the same format; slashes and backslashes are as input and
    //  are not modified.
    //
    //  For Windows,   native:    as input; slashes and backslashes are not modified;
    //                            this is the format of the internally stored string.
    //                 generic:   backslashes are converted to slashes

    //  -----  native format observers  -----

    const string_type&  native() const BOOST_NOEXCEPT  { return m_pathname; }
    const value_type*   c_str() const BOOST_NOEXCEPT   { return m_pathname.c_str(); }
    string_type::size_type size() const BOOST_NOEXCEPT { return m_pathname.size(); }
````

and OpenSCAD amf import

https://github.com/openscad/openscad/blob/master/src/import_amf.cc#L325

```

xmlTextReaderPtr AmfImporterZIP::createXmlReader(const char *filename)
{
	archive = zip_open(filename, 0, NULL);
	if (archive) {
		fs::path f(filename);
                zipfile = zip_fopen(archive, f.filename().c_str(), ZIP_FL_NODIR);
```

i cant remember how to fix this stuff... 

---------

You can install libzip using 'pacman' msys package manager. It will be something like this

     pacman -Ss zip # searches for 'zip' libraries in msys2 pacman package manager
     pacman -S mingw-w64-x86_64-libzip  # installation of 64bit msys2 libzip package

The build itself quits on the call to zip_fopen

fs::filsystem

donbright, 2017-01-25:
Might be worth it to investigate converting to string() then to c_str() or something. Maybe use  openscad/src/PlatformUtils-win.cc .... needs more digging. 
 
http://stackoverflow.com/questions/11352641/boostfilesystempath-and-fopen

kintel, 2017-01-25:
Writing file I/O code on Windows is a bit tricky due to their special handling of unicode filenames.

In https://github.com/openscad/openscad/pull/1296, we tried to alleviate some of the Windows path issues, e.g by using Boost.Nowide (http://cppcms.com/files/nowide/html/). Not sure if that makes the code too readable though, and that PR is still pending.

I did also try to configure the default locale to automatically convert to UTF-8 when calling path::string(), see https://github.com/openscad/openscad/pull/1296/files#diff-dfca9b2232d11047edf4d87631324630R778.

Assuming that libzip supports utf-8 filenames in zip archives, the latter method sounds like the correct solution.

nophead, 2017-04-26:
I note that AMF import dosen't work in the latest Windows snapshot either. Is this issue the reason why?

t-paul, 2018-10-28:
Fixed by @nophead in #2471.

closed 2018-10-28 by t-paul