openscad/openscad #1921: Application hang forever in simple model

mrdeveloperdude, 2017-01-29:
The following hangs nightly build (git 315ebe1):

$fn=100;
```
module base(){
	cylinder(h=1,  r=5,center=true);
}

module limb(){
	cube([5,1,1]);
	sphere([1,1,1]);
}

module leg_part(index){
	translate([1,0,0]){
		rotate([0,10,0]){
			echo (index);
			if( index > 0 ) {
				leg_part(index-1);
			}
			else{
				limb();
			}
		}
	}
}

module leg(){
	color("Red")render(){
		leg_part(5);
	}
}
	
module robot(){
	//base();
	for(i=[0:6]){
		rotate([0,0,(360*i)/6]){
				leg();
		}
	}
}

robot();
```

mrdeveloperdude, 2017-01-29:
I noticed if i reduce $fn to 20 it works. Also commenting out the sphere().

maybe it is just generating too much data?

t-paul, 2017-01-29:
Yes, $fn = 100 a huge amount of vertices for spheres. This renders in 2 minutes for me. Maybe try using $fa / $fs (e.g. set both to 0.5) which adapts the resolution dependent on the actual size of the spheres.

t-paul, 2017-01-29:
Oh, just saw that post from 15 minutes ago: https://twitter.com/caterpillar/status/825541171095048192 :-)

=> http://openhome.cc/eGossip/OpenSCAD/Circle.html

closed 2017-01-29 by t-paul