openscad/openscad #1925: Automatic preview in customizer

RonaldoCMP, 2017-01-31:
The automatic preview in the customizer is a nice feature but some applications need something else. When the code is somewhat lengthy, we should keep the automatic preview off otherwise each field change will take a long time. So I keep it off and after the necessary parameter editions mark it as on to see the results. That is annoying because I have to unmark it again to the next bunch of editions,

To solve this, it would be enough that an ENTER in the customizer panel be a forced preview when the non automatic preview is off.

kintel, 2017-01-31:
Wouldn't ENTER have the exact same function as F5 in this case?

RonaldoCMP, 2017-01-31:
You are right. Sorry about that.

closed 2017-01-31 by RonaldoCMP