openscad/openscad #1927: Error msg when opening file, tries to open .json file

MichaelAtOz, 2017-02-01:
With newish snapshot, every file I open gets:
```
OpenSCAD 2017.01.20
http://www.openscad.org

Copyright (C) 2009-2017 The OpenSCAD Developers

This program is free software; you can...
ERROR: Cannot open Parameter Set 'C:/Users/MeB/Documents/MH Logo - Shapeways.json': C:/Users/MeB/Documents/MH Logo - Shapeways.json: cannot open file 
Loaded design 'C:/Users/MeB/Documents/MH Logo - Shapeways.scad'.
```
I cut the path length above for clarity, this is what it looks like:
[image]

kintel, 2017-02-01:
This is related to the customizer. We should be nicer about trying to open the "parameter set" json file if it's missing.

MichaelAtOz, 2017-02-01:
Note a. I was not using customizer, b. started happening after 2016.11.11

kintel, 2017-02-01:
@MichaelAtOz You didn't even have the experimental customizer feature enabled?

amarjeetkapoor1, 2017-02-01:
Above message is shown as already created .scad files don't have corresponding .json files (Which is required for Parameters set widget) and .json files are automatically created only on SaveAs(). 
Plus I doubt that this will happen without enabling the Customiser.

MichaelAtOz, 2017-02-01:
Sorry, yes it was enabled via previous version while testing it (noting that error didn't happed in previous testing). But it should be usable (nicely) with it enabled but not in use.

kintel, 2017-02-01:
I think that warning should probably be silenced as it's confusing to people not actively using the customizer feature.

amarjeetkapoor1, 2017-02-03:
I am thinking of replacing given code with below one :
https://github.com/openscad/openscad/blob/master/src/parameter/parameterset.cpp#L57
```
    try {
        //check whether file is read-only
        std::fstream file;
        if (boost::filesystem::exists(filename)) {
            pt::read_json(filename, this->root);
            file.open(filename, std::ios::app);
            if (file.is_open() ) {
                file.close();
                return true;
            }
        }else{
            file.open(filename, std::ios::out);
            if (file.is_open()){
                file.close();
                return true;
            }
        }
        return false;
    }
```

This will create the new json file if file didn't exit and if its possible to write in that directory but It will create .json file for all .scad file automatically when customizer is enabled even if not used. 

A better option will be If we are able to check for permission like below without actually creating it.
```
    try {
        //check whether file is read-only
        if (boost::filesystem::exists(filename)) {
            pt::read_json(filename, this->root);
             if (access(filename, W_OK) == 0){
                return true;
            }
        }else{
            //check if we can create json file or not.
            if (access(boost::filesystem::path(filename).parent_path().c_str(), W_OK) == 0){
                return true;
            }
        }
        return false;
    }
```

kintel, 2017-02-03:
I'm thinking that we should probably not automatically create such a JSON file unless it's needed, i.e. if use user has created any presets.
..at least not until this becomes a more core feature of OpenSCAD.

amarjeetkapoor1, 2017-02-03:
Then, We can use the second solution if access() function which belonging to unistd.h library of c/c++ work works on the all OS that openscad is supporting. It will only create new file if user create set of parameter in customizer i.e JSON file will be created only if needed.

amarjeetkapoor1, 2017-02-08:
Will above method be okay?

kintel, 2017-02-08:
Hm, the `access()` man page states:
```
access() is a potential security hole and should never be used.
```

It sounds a bit like the only way to know is to try to create the file on demand, and if it fails, give a warning.

MichaelPFrey, 2018-09-15:
I think this issue should be resolved.

kintel, 2018-09-15:
@MichaelPFrey Not sure I follow; is this no longer an issue, or do you think it needs solving?

MichaelPFrey, 2018-09-16:
This should not longer be an issue, as I think tjat #2311 has fixed it coincidentally 
I think I just close it, as the symptom described in this issue does not exist anymore.

closed 2018-09-16 by MichaelPFrey