openscad/openscad #1930: 3D view not showing difference()

adamjezek98, 2017-02-04:
I have this simple code:
```
difference(){
    cylinder(90,45,45);
    translate([0,0,5])cylinder(90,40,40);
}
```

When I pres F5, the view shows the first cylinder, but it's not cutted. It behaves just like if there were only `cylinder(90,45,45);` To see the things as they are, I have to press F6 and wait for render. I have enough powerfull PC so it took few second even for complex models, but what I need is to see shapes colored to know what I'm doing.
I haven't used OpenSCAD for about a year, and now when I installed it, this happens. Did something changed?

kintel, 2017-02-05:
This is typically due to bad graphics card drivers.
Could you post the version of your OS, the version of OpenSCAD as well as the output of Help->Library Info ?

adamjezek98, 2017-02-05:
Drivers should be fine & up to date (downloaded from Lenovo's website and Radeon settings are regulary checking for updates).

```
OpenSCAD Version: 2015.03-2
System information: Windows 8 (v6.2.0.0) 4 CPUs 7.92 GB RAM //actually I have Win10
Compiler: GCC "5.2.0" 64bit
Boost version: 1_57
Eigen version: 3.2.5
CGAL version, kernels: 4.5, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.0
Qt version: 5.5.1
QScintilla version: 2.8.4
MingW build: MingW64
GLib version: 2.44.1
Application Path: C:/Program Files/OpenSCAD     //the slashes looks a bit strange
Documents Path: D:\OneDrive\Dokumenty
Resource Path: C:/Program Files/OpenSCAD
User Library Path: D:/OneDrive/Dokumenty/OpenSCAD/libraries
User Config Path: C:\Users\jezek\AppData\Local/OpenSCAD
Backup Path: D:/OneDrive/Dokumenty/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
D:/OneDrive/Dokumenty/OpenSCAD/libraries
C:/Program Files/OpenSCAD\libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.12.0
OpenGL Version: 4.4.0 - Build 20.19.15.4531
GL Renderer: Intel(R) HD Graphics 5500
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Using QGLWidget

GL Extensions:
GL_3DFX_texture_compression_FXT1
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
[... 166 lines cut ...]
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_ARB_buffer_region
WGL_ARB_create_context
WGL_ARB_create_context_profile
WGL_ARB_create_context_robustness
WGL_ARB_extensions_string
WGL_ARB_framebuffer_sRGB
WGL_ARB_make_current_read
WGL_ARB_multisample
WGL_ARB_pbuffer
WGL_ARB_pixel_format
WGL_ARB_pixel_format_float
WGL_EXT_create_context_es2_profile
WGL_EXT_create_context_es_profile
WGL_EXT_depth_float
WGL_EXT_extensions_string
WGL_EXT_pixel_format_packed_float
WGL_EXT_swap_control
WGL_EXT_swap_control_tear
WGL_INTEL_cl_sharing
WGL_NV_DX_interop
WGL_NV_DX_interop2

```

MichaelAtOz, 2017-02-06:
>  Intel(R) HD Graphics 

Try Edit/Preferences/Advanced/Force_goldfeather

adamjezek98, 2017-02-07:
@MichaelAtOz 's solution worked, so no need to try snaphots.

kintel, 2017-02-08:
@adamjezek98 The posted solution basically degrades performance by not utilizing your GPU as much. It would still be appreciated if you tested a snapshot, just to know if the original problem was fixed.

adamjezek98, 2017-02-11:
@kintel I downloaded 2017.1.20 and the problem seems to be solved. After disabling force goldfeather, the differences are shown correctly.
Only at some places, there are the blinking faces, but I guess it's because I didn't translated the objects and both cubes has some faces at the same cords.

    difference(){
        translate([-24,-4,0])cube([48,8,height]);
        translate([-12,-4,0])cube([24,8,height]);
     }

It would be great if somebody will do something with this, translating every object by 0.001 so it isn't at the same place as the others is a bit annoying, especially where I have dozens or even hundreds of this in my code.

kintel, 2017-02-11:
@adamjezek98 Thanks!

Solving the flickering is a different topic. We cannot just automatically translate objects as we don't know the design intent, e.g. in which axis to translate. The may be smarter ways around that, but that requires resources to design and implement.

closed 2017-02-11 by kintel