openscad/openscad #1931: Customiser preset name can't contain '.' characters

smartavionics, 2017-02-05:
I entered a customiser preset name that contained a '.' and the name was split at the period. The name was "Rigid.ink 1Kg spool" and this is what got saved in the Json:

        "Rigid": {
            "ink 1Kg spool": {
                "$fn": "64",
                "base_fillet_radius": "5",
                "base_thickness": "5",
                "hub_hole_dia": "6.5",
                "hub_thickness": "18",
                "hub_top_dia": "12",
                "lip_depth": "3",
                "lip_width": "4",
                "spool_hole_dia": "53",
                "spool_width": "75",
                "wall_thickness": "4"
            }
        },

amarjeetkapoor1, 2017-02-07:
This problem is being caused as the `name.name2` is being considered by boost property tree as save `name2` as children of the `name`.
Possible solutions is that:
* Don't allow "." and show the warning. 
* Replace `.` with space or any character during writing in JSON file.
* Use any other library.

smartavionics, 2017-02-07:
Hi, can't you use object["foo.bah"] rather than object.foo.bah ?

amarjeetkapoor1, 2017-02-07:
Basically, Property Tree of Boost library is being used for JSON related stuff and they are using following notation.

t-paul, 2017-02-07:
Maybe there's a different way of constructing the boost property tree that will handle the "." case (= not parse it but just use it verbatim).

amarjeetkapoor1, 2017-02-07:
Yes, There are. I am looking how they can be used like One is `pt.put('/', "example.txt", "10");` where the delimiter is changed to some another symbol but it will create the same problem for another symbol and there is one more method using `ptree::value_type`. which I am looking at.

http://stackoverflow.com/questions/1948745/boost-property-tree-with-filename-as-key

kintel, 2017-02-14:
Fixed by #1932

closed 2017-02-14 by kintel