openscad/openscad #1936: Issue with variables

SparkyCola, 2017-02-10:
Hi,
I found a weird issue in my current [project's first commit](https://github.com/SparkyCola/SparkyScreen/commit/4fa719d7e22af4393d021f548bf598f47a88eacb) when running line 14-16: 
`thread="M4";`
`nutThickness = _get_fam(thread)[_NB_F_NUT_HEIGHT];`
`echo("nutThickness is (from nutThickness variable):",nutThickness, "however it should be", _get_fam(thread)[_NB_F_NUT_HEIGHT]);`

1. works perfectly fine when ran in a new document (with required definitions)
2. In the first 4 previews there is a warning "thread not defined", then on the 5th+ preview update this warning disappears but nutThickness stays undef.

SparkyCola, 2017-02-10:
Found a hint/fix:
if I change the 2nd line to:
`nutThickness = _get_fam("M4")[_NB_F_NUT_HEIGHT];`
I everything work fine. So there has to be some issue with the parameters of functions here (my best guess).

nophead, 2017-02-10:
Remove this line and it will work: https://github.com/SparkyCola/SparkyScreen/commit/4fa719d7e22af4393d021f548bf598f47a88eacb#diff-65dc3134da622e514c395c4b66863a83R12

When you have multiple definitions of the same variable OpenSCAD use the last definition but it places where the first one is. At that point thread is not defined.

SparkyCola, 2017-02-10:
ah OK thanks. I didn't see that double definition and fixed the issue in my code. But I don't quite get how this has an effect on the definition of thread.

SparkyCola, 2017-02-10:
Are redefinitions something that would be OK in OpenSCAD or should you not do it/avoid it? And if yes wouldn't it make sense to show a warning.

nophead, 2017-02-10:
OpenSCAD takes the value of the last definition, in this case ```_get_fam(thread)[_NB_F_NUT_HEIGHT]``` and places it where the first definition is so it has only one value throughout the script. In this case line 12. At line 12 thread isn't defined yet, so get_fam is passed undef and returns undef.

SparkyCola, 2017-02-10:
oh so there should be a warning when you do that, right? I mean when I run into that issue others might too since this is a big difference to other programming languages that i know (haven't had such issues in Python or C++).

nophead, 2017-02-10:
Redefinitions are allowed to allow command line options to override constants and also to allow constants in libraries to be changed to customise the. For that reason there are no warnings as it is expected use.

Probably not the best design choice, but it works once you understand the rules.

nophead, 2017-02-10:
It could probably warn when there are multiple definitions in the same file as that is always a mistake I think.

SparkyCola, 2017-02-10:
I meant there should be warning if a variable is not defined when the definition was moved.

SparkyCola, 2017-02-10:
> It could probably warn when there are multiple definitions in the same file as that is always a mistake I think.
And add a hint if it might be related to another variable not being defined.

nophead, 2017-02-10:
You said it was warning "thread not defined". It would be improved if it showed which line it was undefined at.

SparkyCola, 2017-02-10:
And maybe point out that are are multiple definitions the variable it was assigned to.

nophead, 2017-02-10:
Possibly but you soon find out that error reporting is not one of OpenSCAD's strong points. Because errors don't have line numbers associated with them it can be almost impossible to find where they are in a large project.

SparkyCola, 2017-02-10:
But it would be very useful to improve this, right? But the line numbers would have the priority there.

nophead, 2017-02-10:
Yes line numbers would be a great benefit and a better error message for redefining variables would avoid questions like this. This is an open source project with a small number of people working on it so if you want it anytime soon you probably need to do it yourself.

SparkyCola, 2017-02-10:
I thought about it and since I have some time in the next weeks I might do it.

SparkyCola, 2017-02-10:
I just figured out that the parser does show the line number on errors, so it's only missing for warnings.

nophead, 2017-02-10:
I don't think warnings like these are done at parse time. That is why it is hard to attribute line numbers to them.

doug-moen, 2017-02-10:
If you look at src/AST.h, you'll see that every parse tree node has line
number information.

In expression.h, a Lookup node (which represents an identifier expression
in the parse tree) has a location.

In expr.cc, we have

ValuePtr Lookup::evaluate(const Context *context) const
{
    return context->lookup_variable(this->name);
}

In context.cc, Context::lookup_variable prints "WARNING: Ignoring unknown
variable '%s'.".
The location is known to the caller, but it isn't passed into
lookup_variable, so lookup_variable can't print the location in its warning
message.

This could be fixed by passing a Location& argument to lookup_variable.

nophead, 2017-02-10:
I think the location data in the AST is new, so there are probably a lot of other warnings that can make use of it now.

The really hard ones to find are warnings from rendering the geometry. For example if you pass bad data to polygon. If you have lots of polygon statements it could be any one of them and in a module called from anywhere.

Does the geometry tree have any pointers back to the AST to be able to attribute a line? Or better still a stack trace of modules that generated it. I am guessing not.

kintel, 2017-02-10:
@nophead We keep references to the instantiating AST node from the concrete node tree, but we don't currently keep track of the node-geometry relationship. That's not hard to pull off though (hashmap or node pointer in Geometry) as we have all that information ready when creating geometry for preview, see e.g.  https://github.com/openscad/openscad/blob/master/src/CSGTreeEvaluator.cc#L189

In terms of stack traces: We don't maintain stack information after we have evaluated nodes. I haven't thought about how feasible that would be. It doesn't sound impossible, but it feels unlikely that anyone will have time to do it : /

jordanbrown0, 2025-03-09:
Reassigning a variable in the same file yields a warning:
```
a = 1;
a = 2;
```
```
WARNING: "a" was assigned on line 1 but was overwritten in file test.scad, line 2
```

This warning is also given in "variable not defined at first definition" cases:

```
a = 1;
b = 2;
a = b + 1;
```

```
WARNING: "a" was assigned on line 1 but was overwritten in file test.scad, line 3
Compiling design (CSG Tree generation)...
WARNING: Ignoring unknown variable "b" in file test.scad, line 3
WARNING: undefined operation (undefined + number) in file test.scad, line 3
```

closed 2025-03-09 by jordanbrown0