openscad/openscad #1937: Syntax errors only shown in one window

nophead, 2017-02-10:
If you have multiple widows open, and a file is used by more than one of them, then syntax errors in that file are only shown in the first window that updates the module. Other windows just get undefined modules and functions from that module being missing.

kintel, 2017-02-10:
This is not a well-tested use-case, but thanks for keeping track!
This is a drawback of our ModuleCache.
Some notes:
* We could look into better error reporting on modules missing due to parse errors
* We could look into error message caching. I feel the rabbit hole could be a bit deep, but could be used to build valuable infrastructure for other caches (e.g. CGAL errors)
* Perhaps a global ModuleCache is misguided and it should be per Window?

Question: If a module was already compiled successfully, does future parse errors kill it, or does it survive in the same way we keep references to the AST of removed files?

doug-moen, 2017-02-10:
I think that a cache should be transparent to the user. The system should
behave exactly as if there was no cache, except hopefully it runs faster.

nophead, 2017-02-10:
>Perhaps a global ModuleCache is misguided and it should be per Window?
I don't think so because the windows can be just different views of the same project. It doesn't make sense to have two identical copies of all the modules just to repeat the error messages in each window.

If they view files that have nothing in common it all works as it should.

>If a module was already compiled successfully, does future parse errors kill it

I believe the parser returns NULL but there is no test for that so entry.module becomes NULL.

>I think that a cache should be transparent to the user. The system should behave exactly as if there was no cache, except hopefully it runs faster.

Yes of course, it is just a matter of working out how to make it work like that.

I think the parser errors need to be stored in the module cache and regurgitated once per window. That requires the cache to know which window is calling it. We could pass the window's this pointer to a function before calling the top level handleDependencies().  It can just be a void pointer to avoid dependencies on QT. 

The module cache entry would need to store a set of windows that had seen the errors.

What are print_messages_push() and print_messages_pop() around the parse call for?

kintel, 2017-02-10:
The print push/pop used to be management of error message caching, but it has fallen out of maintenance.

In terms of how often to display parser messages, this needs a bit of thinking:
1. Is the user expecting that modules are recompiled every time (but only once per window?) the main file is recompiled?
2. Should subsequent refreshes (e.g. due to animation) display all warnings, or should we silence warnings in some cases?

In terms of displaying something once per window: I think it should be the Window's (or its document's) responsibility to enforce this. If the parser could return a string containing all errors and warnings, we could probably "just" print this string whenever the module pointer changes.
As often is the case, some green refactoring of internals might be very helpful to establish the necessary infrastructure.

nophead, 2017-02-10:
It probably isn't a priority as multiple document mode with overlapping file sets was completely broken before my fix and nobody seemed to notice. I never used multiple windows from the same OpenSCAD instance before as the performance was too bad before I added the stat cache, so I never noticed this bug. 

I also wonder how many people work on big multi-file projects like me. The GUI performance with Auto Reload and compile turned on was unworkable when it was stat()ing 17000 files every 200ms per window but I didn't see many complaints.

>Is the user expecting that modules are recompiled every time (but only once per window?) the main file is recompiled?

I don't think so when they are successfully compiled as that would be a waste of time but it is confusing when warnings happen once and go away again without being fixed. As Doug said the fact that the modules are cached should be transparent but it currently has the side effect of making messages only happen once when you expect them every time you compile until they are fixed. That is what would happen without a cache.

>Should subsequent refreshes (e.g. due to animation) display all warnings, or should we silence warnings in some cases?

I think they should happen repeatedly during animations until they are fixed. Why would you want to run an animation before your code was correct anyway? If you are upset by a stream of warnings then fix them!