openscad/openscad #1939: Customizer shows values from included libraries but not from used ones

SparkyCola, 2017-02-12:
Hi, on my [current Project](https://github.com/SparkyCola/SparkyScreen) that uses and includes [@JohnK](https://github.com/JohK/)'s [nutsnbolts library](https://github.com/JohK/nutsnbolts). The Customizer has been a huge help there but it shows the variables from the _<include>d_ libraries too in the first tab (Backside) even though they included before that and adds comments of the Backside tab (like "height of 2nd...") to them:
[image]
I'm using the newest nightly version from the Ubuntu-repo (version 2017.02.11.nightly (git e990ac4)). I don't know if this behavior is planned and if it is it should be mentioned in the WikiBook (I'd like to add it, just tell me)

amarjeetkapoor1, 2017-02-12:
This is a issue and customiser is not supposed to do that.
#1840

kintel, 2017-02-12:
Closing as duplicate

closed 2017-02-12 by kintel

SparkyCola, 2017-02-12:
sorry, didn't find that one

t-paul, 2017-02-12:
As workaround you could try moving the includes to the end.

SparkyCola, 2017-02-12:
> As workaround you could try moving the includes to the end.

Good idea but does unfortunately not work, changes neither the customizer nor includes them so the model won't work correctly. What works is to use `<use>` rather then `<include>` but their not the same and `<use>` does not work for all libraries in my case.