openscad/openscad #1941: No Version number in ABOUT dialog

MarcusWolschon, 2017-02-13:
The help->about dialog shows no obvious  version number/build date/commit .
There is a tiny and easy to miss version number in the title of the window.
However that is hard to see above the giant OPENSCAD - THE PROGRAMMERS SOLID.... text.

t-paul, 2017-02-13:
Please add some details... like which OS, compiled how, downloaded where? Screenshot?

nophead, 2017-02-13:
[image: Inline images 1]

t-paul, 2017-02-13:
I'm getting this for the nightly build on Debian/testing...

[image]

t-paul, 2017-02-13:
@nophead are you having that issue too? Github decided to drop the attached image...

Ma-XX-oN, 2017-02-13:
Um, what about the "OpenSCAD version 2017.02.11.nightly (git e990ac4)"?

nophead, 2017-02-13:
@t-paul,
   No it works fine for me on Win7 and looks like the picture you posted.

MarcusWolschon, 2017-02-14:
You are right. I completely missed that year+month while looking for something like "0.2" or something.
The text was already scrolled down a bit when it apeared on screen. Probably happened while I was trying to enlarge it to fill more then 10% of the screen (17" 4K. Probably used pixe-counts for initial dimensions instead of physical sizes and DPI reported by OS) to real the text.

temoto, 2017-02-19:
@MarcusWolschon would you close the issue then?

closed 2017-02-19 by t-paul