openscad/openscad #1942: Help->about placement

MarcusWolschon, 2017-02-13:
"about" should be the LAST entry in the HELP menu.
Currently the last entry is "FONT LIST".

t-paul, 2017-02-13:
Why? This seems to be OS dependent, e.g. macOS has it usually as first entry (although not necessarily in the Help menu). I'm not sure there's a simple solution that fits everything.

MarcusWolschon, 2017-02-14:
Win+Linux have it as the last entry in the help-menu and the help-menu is the only one that is right-aligned.
MacOS has it in the application menu as the first entry.

Just following the style guides of the platforms it is build for.
Menus are platform dependent resources after all.

Nobody is asking for a simple solution or for a solution now. It's just something that should be done at some point.

t-paul, 2017-02-14:
We can have a look if Qt has some support for this, creating and maintaining platform specific UI descriptions is probably not going to be worthwhile.