openscad/openscad #1945: segmentation fault "0x00005555555e677b in Eigen::internal::Assignment<Eigen::Block<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 3, 1, true>, Eigen::Product<Eigen::Block<Eigen::Matrix<double, 4, 4, 0, 4, 4> const, 3, 3, false>, Eigen::Matrix<double, 3, 1, 0, 3, 1>, 1>, Eigen::internal::add_assign_op<double>, Eigen::internal::Dense2Dense, double>::run (func=..., src=..., dst=...)"

krichter722, 2017-02-14:
After starting `openscad` openscad-2015.03-1-882-ge990ac49e in `gdb` and opening

    wall_distance = 2;
    length = 600-wall_distance;
    width = 300-wall_distance;
    thickness = 3;
    chamber_height = 9;
    hole_diameter = 0.8;
    hole_distance = 2*hole_diameter;
    /* Distance between border to plate and holes.
    */
    hole_offset = 10;

    hole_count_length = length-2*hole_offset/(hole_diameter+hole_distance);
    hole_count_width = width-2*hole_offset/(hole_diameter+hole_distance);
    difference() {
        cube([length, width, thickness]);
        for(i=[0:hole_count_length]) {
            for(j=[0:hole_count_width]) {
                translate([hole_offset+hole_diameter*i*2, hole_offset+hole_diameter*j*2,0]) {
                    cylinder(h=thickness,r1=hole_diameter/2,r2=hole_diameter/2,center=false);
                }
            }
        }
    }

`openscad` crashes due to segmentation fault and backtrace

    > gdb openscad
    GNU gdb (Ubuntu 7.11.90.20161005-0ubuntu1) 7.11.90.20161005-git
    Copyright (C) 2016 Free Software Foundation, Inc.
    License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
    This is free software: you are free to change and redistribute it.
    There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
    and "show warranty" for details.
    This GDB was configured as "x86_64-linux-gnu".
    Type "show configuration" for configuration details.
    For bug reporting instructions, please see:
    <http://www.gnu.org/software/gdb/bugs/>.
    Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.
    For help, type "help".
    Type "apropos word" to search for commands related to "word"...
    Reading symbols from openscad...Reading symbols from /usr/lib/debug/.build-id/35/54370a416d6a75f9aeea9ccdc95fbbb9069ebb.debug...done.
    done.
    (gdb) run
    Starting program: /usr/bin/openscad 
    warning: the debug information found in "/lib64/ld-2.24.so" does not match "/lib64/ld-linux-x86-64.so.2" (CRC mismatch).

    [Thread debugging using libthread_db enabled]
    Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
    [New Thread 0x7fffe9e84700 (LWP 10660)]
    [New Thread 0x7fffdfc14700 (LWP 10661)]
    [New Thread 0x7fffdf413700 (LWP 10662)]
    [New Thread 0x7fffdec12700 (LWP 10663)]
    [New Thread 0x7fffde411700 (LWP 10664)]
    [New Thread 0x7fffcd2e7700 (LWP 10666)]
    [New Thread 0x7fffccae6700 (LWP 10667)]
    [Thread 0x7fffcd2e7700 (LWP 10666) exited]
    [Thread 0x7fffccae6700 (LWP 10667) exited]

    Thread 1 "openscad" received signal SIGSEGV, Segmentation fault.
    0x00005555555e677b in Eigen::internal::Assignment<Eigen::Block<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 3, 1, true>, Eigen::Product<Eigen::Block<Eigen::Matrix<double, 4, 4, 0, 4, 4> const, 3, 3, false>, Eigen::Matrix<double, 3, 1, 0, 3, 1>, 1>, Eigen::internal::add_assign_op<double>, Eigen::internal::Dense2Dense, double>::run (func=..., src=..., dst=...)
        at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:790
    warning: Source file is more recent than executable.
    790	{
    (gdb) bt
    #0  0x00005555555e677b in Eigen::internal::Assignment<Eigen::Block<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 3, 1, true>, Eigen::Product<Eigen::Block<Eigen::Matrix<double, 4, 4, 0, 4, 4> const, 3, 3, false>, Eigen::Matrix<double, 3, 1, 0, 3, 1>, 1>, Eigen::internal::add_assign_op<double>, Eigen::internal::Dense2Dense, double>::run(Eigen::Block<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 3, 1, true>&, Eigen::Product<Eigen::Block<Eigen::Matrix<double, 4, 4, 0, 4, 4> const, 3, 3, false>, Eigen::Matrix<double, 3, 1, 0, 3, 1>, 1> const&, Eigen::internal::add_assign_op<double> const&) (func=..., src=..., dst=...) at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:790
    #1  0x00005555555e677b in Eigen::internal::call_assignment_no_alias<Eigen::Block<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 3, 1, true>, Eigen::Product<Eigen::Block<Eigen::Matrix<double, 4, 4, 0, 4, 4> const, 3, 3, false>, Eigen::Matrix<double, 3, 1, 0, 3, 1>, 1>, Eigen::internal::add_assign_op<double> >(Eigen::Block<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 3, 1, true>&, Eigen::Product<Eigen::Block<Eigen::Matrix<double, 4, 4, 0, 4, 4> const, 3, 3, false>, Eigen::Matrix<double, 3, 1, 0, 3, 1>, 1> const&, Eigen::internal::add_assign_op<double> const&) (func=..., src=..., dst=...) at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:747
    #2  0x00005555555e677b in Eigen::internal::generic_product_impl<Eigen::Block<Eigen::Matrix<double, 4, 4, 0, 4, 4> const, 3, 3, false>, Eigen::Matrix<double, 3, 1, 0, 3, 1>, Eigen::DenseShape, Eigen::DenseShape, 3>::addTo<Eigen::Block<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 3, 1, true> >(Eigen::Block<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 3, 1, true>&, Eigen::Block<Eigen::Matrix<double, 4, 4, 0, 4, 4> const, 3, 3, false> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&) (rhs=..., lhs=..., dst=...)
[... 39 lines cut ...]
    #19401 0x00007ffff482c0e8 in QTimer::timerEvent(QTimerEvent*) (this=0x555556b921b0, e=<optimized out>) at kernel/qtimer.cpp:247
    #19402 0x00007ffff4820653 in QObject::event(QEvent*) (this=0x555556b921b0, e=<optimized out>) at kernel/qobject.cpp:1278
    ---Type <return> to continue, or q <return> to quit---
    #19403 0x00007ffff512d8ac in QApplicationPrivate::notify_helper(QObject*, QEvent*) (this=<optimized out>, receiver=0x555556b921b0, e=0x7fffffffc120)
        at kernel/qapplication.cpp:3804
    #19404 0x00007ffff5132d4f in QApplication::notify(QObject*, QEvent*) (this=0x7fffffffc550, receiver=0x555556b921b0, e=0x7fffffffc120) at kernel/qapplication.cpp:3561
    #19405 0x00007ffff47f23b0 in QCoreApplication::notifyInternal2(QObject*, QEvent*) (receiver=0x555556b921b0, event=event@entry=0x7fffffffc120)
        at kernel/qcoreapplication.cpp:1015
    #19406 0x00007ffff48473ae in QCoreApplication::sendEvent(QObject*, QEvent*) (event=0x7fffffffc120, receiver=<optimized out>)
        at ../../include/QtCore/../../src/corelib/kernel/qcoreapplication.h:225
    #19407 0x00007ffff48473ae in QTimerInfoList::activateTimers() (this=0x555555dde620) at kernel/qtimerinfo_unix.cpp:637
    #19408 0x00007ffff48478d1 in timerSourceDispatch(GSource*, GSourceFunc, gpointer) (source=<optimized out>) at kernel/qeventdispatcher_glib.cpp:176
    #19409 0x00007ffff63917d7 in g_main_dispatch (context=0x7fffe40016f0) at ././glib/gmain.c:3203
    #19410 0x00007ffff63917d7 in g_main_context_dispatch (context=context@entry=0x7fffe40016f0) at ././glib/gmain.c:3856
    #19411 0x00007ffff6391a40 in g_main_context_iterate (context=context@entry=0x7fffe40016f0, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>)
        at ././glib/gmain.c:3929
    #19412 0x00007ffff6391aec in g_main_context_iteration (context=0x7fffe40016f0, may_block=may_block@entry=1) at ././glib/gmain.c:3990
    #19413 0x00007ffff484848f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) (this=0x555555de3ac0, flags=...) at kernel/qeventdispatcher_glib.cpp:417
    #19414 0x00007ffff47f00fa in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) (this=this@entry=0x7fffffffc370, flags=..., flags@entry=...) at kernel/qeventloop.cpp:204
    #19415 0x00007ffff47f890c in QCoreApplication::exec() () at kernel/qcoreapplication.cpp:1285
    #19416 0x000055555573c7c6 in gui(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, boost::filesystem::path const&, int, char**) (inputFiles=..., original_path=..., argc=1, argv=<optimized out>)
        at src/openscad.cc:746
    #19417 0x00005555555d05f5 in main(int, char**) (argc=1, argv=0x7fffffffd5c8) at src/openscad.cc:919

I edited the `#18-19381 [...]` lines manually (and didn't check whether every line contains exactly the same, but I'm pretty sure).

kintel, 2017-02-14:
It looks like you're punching 170K holes in a plate. This is most likely exhausting resources on your computer. Admittedly we could detect this and provide a better error message.

It looks like our check for exploding CSG tree sizes somehow got sidestepped with this particular design.

krichter722, 2017-02-14:
> It looks like you're punching 170K holes in a plate. This is most likely exhausting resources on your computer. Admittedly we could detect this and provide a better error message.

That's right. I'm just getting started with OpenSCAD and CAD, so I can't expect this to work. Prevention of crashes and better error messages are always good :)

t-paul, 2020-02-16:
This does not seem to crash anymore. Original model renders for me, increasing the number of holes eventually results in "aborting CSG tree normalization".

closed 2020-02-16 by t-paul

MichaelAtOz, 2020-02-17:
This, OP code, worked with nightly in Debian, but is not completing* with 2019.05 on Windows 7.

\* preview has been running for hours, the window is 'frosted' & title shows 'not responding', progress bar 928/1000, it is using ~70% CPU, ~360MB memory - pretty static, Microsoft process explorer shows a number of threads#
[image]
 active.
\# that VM has 6 CPUs. (2019.05 standard release, NOT multithread version)

I just tried with 2020.02.06, similar behaviour, but blank window, no progress bar.
So somethings wrong.

t-paul, 2020-02-17:
It's quite a huge number of vertices to chew through and OpenSCAD is not very good at that, so there could be other issues. Still, the ticket is about a crash in CSG normalization which does not seem to happen anymore.
It is indeed a bit strange that it behaves that differently on my Linux notebook. The preview did not take very long to show up. It's not really responsive though, but that's a known issue with the old display code.

thehans, 2020-02-17:
Yeah the normalization code was fixed up to tolerate really large number of objects without crashing. 

And, while I know this issue is quite old and the original reporter has likely moved on from this model long ago... just in case anyone reads this and wanted to do something similar, it is **tons** more efficient and responsive for a design like this to do all the differences in 2d, _then_ linear_extrude that:
```
wall_distance = 2;
length = 600-wall_distance;
width = 300-wall_distance;
thickness = 3;
chamber_height = 9;
hole_diameter = 0.8;
hole_distance = 2*hole_diameter;
/* Distance between border to plate and holes.
*/
hole_offset = 10;

hole_count_length = length-2*hole_offset/(hole_diameter+hole_distance);
hole_count_width = width-2*hole_offset/(hole_diameter+hole_distance);
linear_extrude(thickness)
difference() {
    square([length, width]);
    for(i=[0:hole_count_length]) {
        for(j=[0:hole_count_width]) {
            translate([hole_offset+hole_diameter*i*2, hole_offset+hole_diameter*j*2]) {
                circle(d=hole_diameter);
            }
        }
    }
}
```