openscad/openscad #1946: Cannot launch program

Atomsk102, 2017-02-14:
On my office PC, whenever I try to open a project file, or create a new one in OpenSCAD, I get an error pop-up, with the message:
"LoadLibrary failed with error 87: The parameter is incorrect."
I've tried v2015.3-2 and v2017.1.20.  Both produce the same error.
I run OpenSCAD on other PCs, without this problem.

kintel, 2017-02-14:
Googling this error indicates that it's due to a bad driver from ATI: https://answers.microsoft.com/en-us/windows/forum/windows8_1_pr-winapps/loadlibrary-failed-with-error-87-the-parameter-is/c2cfe87c-c7d7-4371-bb79-0da75d05cd6d

Could you update the drivers and try again?

Atomsk102, 2017-02-14:
Thanks, sorry for the false alarm.  I looked for a fix online before, and never found anything.  One of your links lead me to a solution.

You can close this issue.

closed 2017-02-14 by t-paul