openscad/openscad #1951: Allow resize of editor/console undocked windows OSX

robinhilliard, 2017-02-18:
When the editor and console windows are undocked resizing controls in top left are disabled and you can't drag the window edges to resize them.

t-paul, 2017-02-18:
Looks like that's an issue in the Qt library - https://bugreports.qt.io/browse/QTBUG-46882 / https://bugreports.qt.io/browse/QTBUG-48618

Could you please check which version is used in the OpenSCAD you have installed (see Help -> Library Info)?

robinhilliard, 2017-02-20:
Hi Paul,

It’s OpenSCAD 2015.03-3 on OSX 10.10.5 with Qt version 5.5.1.

Cheers,
Robin

kintel, 2017-02-20:
@robinhilliard Could you try the development snapshot (http://www.openscad.org/downloads.html#snapshots) and see if it's fixed there?

robinhilliard, 2017-02-20:
Hi Marius,

Thanks, I can resize the windows by grabbing any edge. The minimise/maximise buttons in the top left corners of the editor and console are still disabled as per the attached screenshot (min/max work on the preview window and max works on the about window)  I can easily work around that now but Mac users will notice.

Cheers,
Robin

_________________________________________________

Robin Hilliard
Chief Technology Officer
RocketBoots Pty Limited

mobile:	+61 (0)418 414 341
office:	+61 (0)2 9323 2507
email:	robin@rocketboots.com
street:	Level 11, 189 Kent St Sydney NSW 2000
post:	GPO Box 4647 Sydney NSW 2001

kintel, 2024-12-03:
Qt bug was fixed a long time ago. The greyed out minimize/maximize icons seem to be a default setting across OS'es, not just macOS. Fixable by setting the correct window flags for all floating windows.