openscad/openscad #1952: OpenSCAD should report the missing library files when it fails to load.

ghpenguin, 2017-02-19:
I run OpenSCAD in Devuan 64 bit Linux. 
Kernel version is 4.8.0-0.bpo.2-amd64 #1 SMP Debian 4.8.15-2~bpo8+2 (2017-01-17) x86_64 GNU/Linux.

My attempts to install AMD Catalyst™ Proprietary Linux Graphics Driver failed. I got back to Open Source radeon driver (Radeon 3000).  But the rollback lost critical OpenGL library files I couldn't trace. I managed to reinstall the missing OpenGL library files by using 
'update-alternatives --config glx' command. 

OpenSCAD 2017.02.11.nightly showed the empty preview window complaining assertion error. If it tells the file names upon failure, it simplify troubleshooting of the end user. It prevents few hours hair scratching frustration.

t-paul, 2017-02-19:
What exactly was missing? It sounds more like an environment / distribution issue. It's not really a common thing for applications to explicitly check for specific OpenGL driver files. For OpenSCAD most of that is hidden via Qt anyway.

t-paul, 2018-03-28:
Closing due to lack of feedback.

closed 2018-03-28 by t-paul