openscad/openscad #1954: [Windows] Error saving design on Dropbox folder

kevinta893, 2017-02-20:
There's an issue with OpenSCAD when saving a file that is located on a Dropbox folder. It claims that the "file is being used by another process" which is clearly Dropbox. 

This happens whenever you add sufficient code to add a new object to be rendered/previewed (e.g. cube) and then saving the file. I've repeated the issue on both Windows 8 and 10, on version 2015.03-2.

This is a mild inconvenience but it would be great to have this fixed.

t-paul, 2017-02-20:
I think that's https://bugreports.qt.io/browse/QTBUG-57299 - probably not much OpenSCAD can do here.

kevinta893, 2017-02-20:
Alright sounds good. Thanks.

t-paul, 2020-02-16:
Closing as upstream Qt issue is closed. Should be fixed in Qt5.12 (version used in OpenSCAD is shown in Help -> Library Info).

closed 2020-02-16 by t-paul