openscad/openscad #1955: Gears not producing involute gears

trayracing, 2017-02-21:
MCAD/gear.scad is not producing involute gears as claimed - the teeth look very boxy.
[image]

For comparison: gears.scad quotes [a cartertools web page](http://www.cartertools.com/involute.html) on gear construction. That page works through an example that corresponds to 
gear(number_of_teeth=20, diametral_pitch=16, pressure_angle=14.5, clearance=0.009860625);
That cartertools gets this:
[image]
which looks a credible tooth profile, when compared to [wikipedia](https://en.wikipedia.org/wiki/Involute_gear)

I tried this in 2015.3 and 2016.09.18 (git a5e4def) on windows 10, using both F5 and F6 and both appear to have the problem. I have not dug around to figure out if it's gears.scad or openscad at fault.

bond4u, 2017-03-22:
use MCAD/involute_gears.scad

trayracing, 2017-03-22:
Thanks.That would be a great addition to the top of MCAD/gears.scad:      "//Non-involute gears. For involute gears, use MCAD/involute_gears.scad".As gears.scad currently reads, it's misleading. The first line after the copyright notice is:      "//test_involute_curve();" Then it references an article on involute gear design.

kintel, 2017-03-23:
This should be moved to an MCAD issue (https://github.com/openscad/MCAD/issues)