openscad/openscad #1956: Parse error caused by an included file does not recompile when fixed

nophead, 2017-02-21:
FileModule::parse() returns null when there is a syntax error. This causes lib_mod to be set to null here: https://github.com/openscad/openscad/blob/master/src/ModuleCache.cc#L117 and so this line: https://github.com/openscad/openscad/blob/master/src/ModuleCache.cc#L74 never checks the includes again.

I think this has always been the case but I have only noticed it after my changes.

nophead, 2017-02-21:
The same problem occurs in winmain.cc. Parse errors cause the root module to be deleted but the is where its include files are stored and they can cause parse errors.

I think the solution is to store the dependency graph in its own data structure.

nophead, 2017-02-22:
I think I can fix it by storing the last good module pointer in the cache entry and the window and use that for the checking includes. Whenever there is a good parse delete the old module and set it to the new one.

kintel, 2017-02-22:
@nophead To clarify that I understand this correctly: Is this only a problem with automatic reload, or are you also seeing this when manually triggering a preview using F4 or F5 ?

nophead, 2017-02-22:
The problem in mainwin.cc only affects auto reload because and you can force recompilation with F5 or F4. The effect of the same problem in the ModuleCache.cc is worse because you can't force a used module to be recompiled, unless you touch it, or clear the module cache, because it stops looking at the include times.

oysteinkrog, 2017-02-26:
I can confirm this problem. I have always just had to manually reload the window.

nophead, 2019-02-24:
Fixed by #1958

closed 2019-02-24 by nophead